OSSystems/meta-browser #790: Chromium 120.0.6099.224 fails to build with mickledore  branch

rjanani-p, 2024-02-07:
Hi ,
I am facing build issue for mickledore  branch for Chromium 120.0.6099.224 version , I have added the chromium package to the build (IMAGE_INSTALL:append = " chromium-x11") in local.conf . 
I am attaching the build log for reference. I would like to know what steps I need to take to build works,

It is failing at do_copy_target_rustlibs task,

0: chromium-x11-120.0.6099.224-r0 **do_copy_target_rustlibs** - 0s (pid 4010289)
ERROR: chromium-x11-120.0.6099.224-r0 do_copy_target_rustlibs: ExecutionError('build/tmp-glibc/work/core2-64-wrs-linux/chromium-x11/120.0.6099.224-r0/temp/run.do_copy_target_rustlibs.4010289', 1, None, None)
ERROR: Logfile of failure stored in: build/lts23base_build/tmp-glibc/work/core2-64-wrs-linux/chromium-x11/120.0.6099.224-r0/temp/log.do_copy_target_rustlibs.4010289
Log data follows:
| DEBUG: Executing shell function do_copy_target_rustlibs
| cp: cannot stat 'build/lts23base_build/tmp-glibc/work/core2-64-wrs-linux/chromium-x11/120.0.6099.224-r0/recipe-sysroot/usr/lib/rustlib/x86_64*': **No such file or directory**
| WARNING: build/tmp-glibc/work/core2-64-wrs-linux/chromium-x11/120.0.6099.224-r0/temp/run.do_copy_target_rustlibs.4010289:150 exit 1 from 'cp -r $rustlib_src_dir "/build/lts23base_build/tmp-glibc/work/core2-64-wrs-linux/chromium-x11/120.0.6099.224-r0/recipe-sysroot-native/usr/lib/rustlib"'
| WARNING: Backtrace (BB generated script):
|       #1: do_copy_target_rustlibs, /build/lts23base_build/tmp-glibc/work/core2-64-wrs-linux/chromium-x11/120.0.6099.224-r0/temp/run.do_copy_target_rustlibs.4010289, line 150
|       #2: main,/build/lts23base_build/tmp-glibc/work/core2-64-wrs-linux/chromium-x11/120.0.6099.224-r0/temp/run.do_copy_target_rustlibs.4010289, line 153

MaxIhlenfeldt, 2024-02-07:
Does `build/lts23base_build/tmp-glibc/work/core2-64-wrs-linux/chromium-x11/120.0.6099.224-r0/recipe-sysroot/usr/lib/rustlib` exist? If yes, what do its contents look like?

rjanani-p, 2024-02-07:
rustlib is not present in following path
build/lts23base_build/tmp-glibc/work/core2-64-wrs-linux/chromium-x11/120.0.6099.224-r0/recipe-sysroot/usr/lib/rustlib, to get rust lib any steps I need to perform? could you please share your input.

MaxIhlenfeldt, 2024-02-07:
I'll do a build using mickledore myself and will check, I think that's easier.

rjanani-p, 2024-02-07:
sure, @MaxIhlenfeldt ,  thanks for the sharing the inputs.

MaxIhlenfeldt, 2024-02-07:
Hm, my build is now happily compiling, that means the `do_copy_target_rustlibs` task finished successfully. Can you please share the build configuration `bitbake` prints when you start a build?

This is what I get:
```
Build Configuration:
BB_VERSION           = "2.4.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "debian-12"
TARGET_SYS           = "aarch64-oe-linux"
MACHINE              = "qemuarm64"
DISTRO               = "nodistro"
DISTRO_VERSION       = "nodistro.0"
TUNE_FEATURES        = "aarch64 armv8a crc cortexa57"
TARGET_FPU           = ""
meta
meta-poky            = "HEAD:7235399a86b134e57d5eb783d7f1f57ca0439ae5"
meta-clang           = "HEAD:ca6c714d4a807f199db009bf4be293640623d7af"
meta-chromium        = "master:0fcfadca55fa36c48ef09ff85c82d5bcde39540b"
meta-oe              = "HEAD:8e1f0fa6bfac0e96fedc666fe9066f92c85afb27"
```

MaxIhlenfeldt, 2024-02-08:
How do you set up your sources? The commit hashes listed don't seem to map to commits in the repositories (e.g. https://github.com/kraj/meta-clang/commit/de872f7b72923cd2474b39d444500ce3af929720 doesn't exist, and neither does https://github.com/OSSystems/meta-browser/commit/23843b037e853696e35d14773d28f8b9f83198c0).

rjanani-p, 2024-02-09:
We are maintaining mickledore branch internally that's the reason difference in commit id's and I have cross checked the meta-clang and remaining layers those are having the mickledore latest commit changes, could you please share your input how to resolve this issue.

rwmacleod, 2024-02-12:
@rjanani-p please try to reproduce the issue without using WR Linux layers.

rjanani-p, 2024-02-12:
@rwmacleod I have tried to reproduce the issue using poky mickledore branch (without using WR Linux layers) but I am unable to reproduce the issue. But why it is reproducing for WR linux branch not understanding. I am sharing the build configuration to reproduce the issue for reference.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "2.4.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "x86_64-poky-linux"
MACHINE              = "qemux86-64"
DISTRO               = "poky"
DISTRO_VERSION       = "4.2.4"
TUNE_FEATURES        = "m64 core2"
TARGET_FPU           = ""
meta
meta-poky
meta-yocto-bsp       = "mickledore:7235399a86b134e57d5eb783d7f1f57ca0439ae5"
meta-filesystems
meta-gnome
meta-initramfs
meta-multimedia
meta-networking
meta-oe
meta-perl
meta-python
meta-webserver
meta-xfce            = "mickledore:8e1f0fa6bfac0e96fedc666fe9066f92c85afb27"
meta-clang.git       = "mickledore:ca6c714d4a807f199db009bf4be293640623d7af"
meta-chromium        = "master:0fcfadca55fa36c48ef09ff85c82d5bcde39540b"
meta-raspberrypi.git = "mickledore:9c81413d0b6f41f14f0f9c504d19e312c5b9958f"

could you please share your inputs

rwmacleod, 2024-02-12:
@rjanani-p let's deal with the problem ourselves, it's nothing for@MaxIhlenfeldt to work on or worry about. For now, please resolve this issue as not reproducible or something like that.

rjanani-p, 2024-02-13:
okay sure @rwmacleod , I will close the issue

rjanani-p, 2024-02-13:
unable to reproducing the issue, hence closing the issue. Thank you.

closed 2024-02-13 by MaxIhlenfeldt