OSSystems/meta-browser #792: chromium-x11 build error with Rust enabled

MaxIhlenfeldt, 2024-02-28:
Original: https://lists.openembedded.org/g/openembedded-devel/message/109009

---

Hi,

since commit 0438fba4e773824fd2b41ded2728076deeeedeb1
("chromium: Enable Rust") in meta-browser, I get this error
on several different chromium-x11 builds (i.e. Fedora 33 and
Fedora 39 hosts, mickledore and nanbield Yocto versions):

```
| FAILED: obj/build/rust/std/target.json obj/build/rust/std/lib/libstd.rlib obj/build/rust/std/lib/liballoc.rlib obj/build/rust/std/lib/libcfg_if.rlib obj/build/rust/std/lib/libcompiler_builtins.rlib obj/build/rust/std/lib/libcore.rlib obj/build/rust/std/lib/libgetopts.rlib obj/build/rust/std/lib/libhashbrown.rlib obj/build/rust/std/lib/liblibc.rlib obj/build/rust/std/lib/libpanic_abort.rlib obj/build/rust/std/lib/libpanic_unwind.rlib obj/build/rust/std/lib/librustc_demangle.rlib obj/build/rust/std/lib/libstd_detect.rlib obj/build/rust/std/lib/libtest.rlib obj/build/rust/std/lib/libunicode_width.rlib obj/build/rust/std/lib/libunwind.rlib obj/build/rust/std/lib/libaddr2line.rlib obj/build/rust/std/lib/libadler.rlib obj/build/rust/std/lib/libgimli.rlib obj/build/rust/std/lib/libmemchr.rlib obj/build/rust/std/lib/libminiz_oxide.rlib obj/build/rust/std/lib/libobject.rlib obj/build/rust/std/lib/libprofiler_builtins.rlib obj/build/rust/std/lib/librustc_std_workspace_alloc.rlib obj/build/rust/std/lib/librustc_std_workspace_core.rlib obj/build/rust/std/lib/librustc_std_workspace_std.rlib
| python3 ../../build/rust/std/find_std_rlibs.py --rust-bin-dir ../../../recipe-sysroot-native/usr/bin --output obj/build/rust/std --depfile obj/build/rust/std/stdlib.d --depfile-target std --rustc-revision custom --target x86_64-oe-linux-gnu
| Traceback (most recent call last):
|   File "/home/zozo/test-yocto-4.2-gh/tmp-sicom-glibc/work/corei7-64-oe-linux/chromium-x11/120.0.6099.224-r0/chromium-120.0.6099.224/out/Release/../../build/rust/std/find_std_rlibs.py", line 139, in <module>
|     sys.exit(main())
|              ^^^^^^
|   File "/home/zozo/test-yocto-4.2-gh/tmp-sicom-glibc/work/corei7-64-oe-linux/chromium-x11/120.0.6099.224-r0/chromium-120.0.6099.224/out/Release/../../build/rust/std/find_std_rlibs.py", line 104, in main
|     (crate_name, metadata) = RLIB_NAME_REGEX.match(f).group(1, 2)
|                              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
| AttributeError: 'NoneType' object has no attribute 'group'
```

Obviously, using the previous commit works.

Best regards,
Zoltán Böszörményi

MaxIhlenfeldt, 2024-02-28:
Hm, this should be avoided by [this part](https://github.com/OSSystems/meta-browser/blob/0438fba/meta-chromium/recipes-browser/chromium/rust_%25.bbappend#L9-L22) of our changes to the Rust recipe.

Can you please run `recipe-sysroot-native/usr/bin/rustc --print target-libdir --target x86_64-oe-linux-gnu` and paste the output here? It should be a directory path. Please also paste the contents of that directory, it should contain some `.rlib` files (maybe other files as well).

zboszor, 2024-03-08:
Sorry for the belated response.

Here's what you asked for:
```
$ bitbake -c configure chromium-x11
$ bitbake -c devshell chromium-x11
# cd ..
#
# recipe-sysroot-native/usr/bin/rustc --print target-libdir --target x86_64-oe-linux-gnu | sed "s:$(pwd)/::"
couldn't allocate absolute path for 'null'.
recipe-sysroot-native/usr/lib/rustlib/x86_64-oe-linux-gnu/lib
#
# ls -l recipe-sysroot-native/usr/lib/rustlib/x86_64-oe-linux-gnu/lib
total 344884
-rw-r--r--. 1 1000 1001   485554 Feb 23 07:51 libaddr2line-be0c663ce75d21b3.rlib
-rw-r--r--. 1 1000 1001   420614 Feb 23 07:51 libaddr2line-be0c663ce75d21b3.rmeta
-rw-r--r--. 1 1000 1001   482642 Feb 23 07:51 libaddr2line-da7edd4b0fb2f538.rlib
-rw-r--r--. 1 1000 1001    64110 Feb 23 07:51 libadler-2193210a0e3dc701.rlib
-rw-r--r--. 1 1000 1001    63294 Feb 23 07:51 libadler-fa3352091a24c57c.rlib
-rw-r--r--. 1 1000 1001    25667 Feb 23 07:51 libadler-fa3352091a24c57c.rmeta
-rw-r--r--. 1 1000 1001  6853834 Feb 23 07:51 liballoc-a17efbbf320010bd.rlib
-rw-r--r--. 1 1000 1001  6451658 Feb 23 07:51 liballoc-a17efbbf320010bd.rmeta
-rw-r--r--. 1 1000 1001  6854214 Feb 23 07:51 liballoc-a608e601203ab652.rlib
-rw-r--r--. 1 1000 1001     9358 Feb 23 07:51 libcfg_if-4843d3f6775c0073.rlib
-rw-r--r--. 1 1000 1001     6333 Feb 23 07:51 libcfg_if-4843d3f6775c0073.rmeta
-rw-r--r--. 1 1000 1001     9582 Feb 23 07:51 libcfg_if-4e6b07cdbeb0df31.rlib
-rw-r--r--. 1 1000 1001  1863820 Feb 23 07:51 libcompiler_builtins-a02735b1346e2f24.rlib
-rw-r--r--. 1 1000 1001  1549882 Feb 23 07:51 libcompiler_builtins-d102db5c0fcf1e8b.rlib
-rw-r--r--. 1 1000 1001   813239 Feb 23 07:51 libcompiler_builtins-d102db5c0fcf1e8b.rmeta
-rw-r--r--. 1 1000 1001 62227814 Feb 23 07:51 libcore-b7401c14f096a3aa.rlib
-rw-r--r--. 1 1000 1001 59980565 Feb 23 07:51 libcore-b7401c14f096a3aa.rmeta
-rw-r--r--. 1 1000 1001 62225718 Feb 23 07:51 libcore-f5f014003c697bc2.rlib
-rw-r--r--. 1 1000 1001   665932 Feb 23 07:51 libgetopts-606aca4c2c7df2df.rlib
-rw-r--r--. 1 1000 1001   662438 Feb 23 07:51 libgetopts-b00b6b8ac91e0c9d.rlib
-rw-r--r--. 1 1000 1001   221573 Feb 23 07:51 libgetopts-b00b6b8ac91e0c9d.rmeta
-rw-r--r--. 1 1000 1001  7924538 Feb 23 07:51 libgimli-76b1b03e119df45f.rlib
-rw-r--r--. 1 1000 1001  7823948 Feb 23 07:51 libgimli-daac806e2961b0e6.rlib
-rw-r--r--. 1 1000 1001  6505843 Feb 23 07:51 libgimli-daac806e2961b0e6.rmeta
-rw-r--r--. 1 1000 1001  1596512 Feb 23 07:51 libhashbrown-228fb2f117043106.rlib
-rw-r--r--. 1 1000 1001  1578590 Feb 23 07:51 libhashbrown-228fb2f117043106.rmeta
-rw-r--r--. 1 1000 1001  1617576 Feb 23 07:51 libhashbrown-6a082f942cc46845.rlib
-rw-r--r--. 1 1000 1001  3437658 Feb 23 07:51 liblibc-420faeb0913adfda.rlib
-rw-r--r--. 1 1000 1001  3329618 Feb 23 07:51 liblibc-db9329a966947adb.rlib
-rw-r--r--. 1 1000 1001  3156727 Feb 23 07:51 liblibc-db9329a966947adb.rmeta
-rw-r--r--. 1 1000 1001  1380728 Feb 23 07:51 libmemchr-36971b5e7d224c20.rlib
-rw-r--r--. 1 1000 1001  1371082 Feb 23 07:51 libmemchr-c7f9a3b1fcf752be.rlib
-rw-r--r--. 1 1000 1001  1061153 Feb 23 07:51 libmemchr-c7f9a3b1fcf752be.rmeta
-rw-r--r--. 1 1000 1001  1026184 Feb 23 07:51 libminiz_oxide-0a5886c379b738c4.rlib
-rw-r--r--. 1 1000 1001   552365 Feb 23 07:51 libminiz_oxide-0a5886c379b738c4.rmeta
-rw-r--r--. 1 1000 1001  1038398 Feb 23 07:51 libminiz_oxide-50cc4c2c2f6a52d7.rlib
-rw-r--r--. 1 1000 1001  9325320 Feb 23 07:51 libobject-ab0a9ca01b7647bc.rlib
-rw-r--r--. 1 1000 1001  8325350 Feb 23 07:51 libobject-ab0a9ca01b7647bc.rmeta
-rw-r--r--. 1 1000 1001  9497756 Feb 23 07:51 libobject-daa305bc52ad7031.rlib
-rw-r--r--. 1 1000 1001     9924 Feb 23 07:51 libpanic_abort-0f6f463943ee2ff4.rlib
-rw-r--r--. 1 1000 1001    11164 Feb 23 07:51 libpanic_abort-ea4ad6e88ebebb32.rlib
-rw-r--r--. 1 1000 1001     3806 Feb 23 07:51 libpanic_abort-ea4ad6e88ebebb32.rmeta
-rw-r--r--. 1 1000 1001    33306 Feb 23 07:51 libpanic_unwind-8d3e1ae6c5d7411d.rlib
-rw-r--r--. 1 1000 1001    33094 Feb 23 07:51 libpanic_unwind-fda7bb45ddc631c4.rlib
-rw-r--r--. 1 1000 1001    10540 Feb 23 07:51 libpanic_unwind-fda7bb45ddc631c4.rmeta
-rw-r--r--. 1 1000 1001  4194712 Feb 23 07:51 libproc_macro-18e01331508bf76e.rlib
-rw-r--r--. 1 1000 1001  2035619 Feb 23 07:51 libproc_macro-18e01331508bf76e.rmeta
-rw-r--r--. 1 1000 1001  4196046 Feb 23 07:51 libproc_macro-fb0cd9df737071e6.rlib
-rw-r--r--. 1 1000 1001   412312 Feb 23 07:51 libprofiler_builtins-2343b69abf0721ce.rlib
-rw-r--r--. 1 1000 1001   582734 Feb 23 07:51 librustc_demangle-1aa2102130837de6.rlib
-rw-r--r--. 1 1000 1001   252427 Feb 23 07:51 librustc_demangle-1aa2102130837de6.rmeta
-rw-r--r--. 1 1000 1001   585434 Feb 23 07:51 librustc_demangle-fde8bc8ef3271329.rlib
-rw-r--r--. 1 1000 1001     4924 Feb 23 07:51 librustc_std_workspace_alloc-2585aa3810aa2c06.rlib
-rw-r--r--. 1 1000 1001     1799 Feb 23 07:51 librustc_std_workspace_alloc-2585aa3810aa2c06.rmeta
-rw-r--r--. 1 1000 1001     5132 Feb 23 07:51 librustc_std_workspace_alloc-ebedea0f3fa3631e.rlib
-rw-r--r--. 1 1000 1001     6458 Feb 23 07:51 librustc_std_workspace_core-81b145b7a9d5f231.rlib
-rw-r--r--. 1 1000 1001     3346 Feb 23 07:51 librustc_std_workspace_core-81b145b7a9d5f231.rmeta
-rw-r--r--. 1 1000 1001     6642 Feb 23 07:51 librustc_std_workspace_core-98a56c211caacfca.rlib
-rw-r--r--. 1 1000 1001     8288 Feb 23 07:51 librustc_std_workspace_std-21e332a12682ae31.rlib
-rw-r--r--. 1 1000 1001     8080 Feb 23 07:51 librustc_std_workspace_std-2cd73d2033dc872a.rlib
[... cut ...]
-rw-r--r--. 1 1000 1001   146520 Feb 23 07:51 libunicode_width-14fa0c191df006c7.rlib
-rw-r--r--. 1 1000 1001   145944 Feb 23 07:51 libunicode_width-ce20be7323b234fd.rlib
-rw-r--r--. 1 1000 1001   128614 Feb 23 07:51 libunicode_width-ce20be7323b234fd.rmeta
-rw-r--r--. 1 1000 1001    46672 Feb 23 07:51 libunwind-26f0ecf2ae276e3d.rlib
-rw-r--r--. 1 1000 1001    46496 Feb 23 07:51 libunwind-6d9f4c74674870f7.rlib
-rw-r--r--. 1 1000 1001    37212 Feb 23 07:51 libunwind-6d9f4c74674870f7.rmeta
drwxr-xr-x. 2 1000 1001     4096 Feb 23 07:51 self-contained
```

MaxIhlenfeldt, 2024-03-13:
Hm, I think I know what the problem might be. Are you building for x64 on an x64 machine, i.e. not cross-compiling?

zboszor, 2024-03-13:
> Hm, I think I know what the problem might be. Are you building for x64 on an x64 machine, i.e. not cross-compiling?

Correct.

MaxIhlenfeldt, 2024-03-14:
How is it cross-compiling when host and target architecture are the same?

In any case, can you please check if applying the following diff fixes the error:
```diff
diff --git a/meta-chromium/recipes-browser/chromium/rust_%.bbappend b/meta-chromium/recipes-browser/chromium/rust_%.bbappend
index d103e69..1e139f3 100644
--- a/meta-chromium/recipes-browser/chromium/rust_%.bbappend
+++ b/meta-chromium/recipes-browser/chromium/rust_%.bbappend
@@ -16,6 +16,12 @@ rust_do_install:class-target:append() {
     tar -C _dist -xf build/dist/rust-std-${PV}-${RUST_TARGET_SYS}.tar.xz $rlib_path

     target_dir=${D}${libdir}/rustlib
+
+    # If we're not cross-compiling, target_dir might already be populated, but
+    # not with the contents we want. See
+    # https://github.com/OSSystems/meta-browser/issues/792.
+    rm -rf $target_dir
+
     install -d $target_dir
     cp -r _dist/$rlib_path/${RUST_TARGET_SYS} $target_dir
     rm -rf _dist
```

zboszor, 2024-03-14:
I will try a clean build then.

zboszor, 2024-03-14:
It didn't help. I tried this:
```
bitbake -c cleanall rust-native rust-llvm-native rust rust-llvm gn-native chromium-x11
bitbake -c compile chromium-x11
```
Same error.

MaxIhlenfeldt, 2024-03-20:
Does `ls -l recipe-sysroot-native/usr/lib/rustlib/x86_64-oe-linux-gnu/lib` still show the same directory contents?

zboszor, 2024-03-22:
FWIW, I have run `repo sync` for a fresh test, so it's `chromium-x11` version **121.0.6167.184** now.

The directory contents are not the same.

Also, for some reason, there is no `--target x86_64-oe-linux-gnu` now. Instead, there is `--target x86_64-unknown-linux-gnu`.

Here's the directory contents for mickledore:
```
# ls -l recipe-sysroot-native/usr/lib/rustlib/x86_64-unknown-linux-gnu/lib
total 126832
-rw-r--r--. 5 1000 1001   483058 Mar 21 10:35 libaddr2line-b3ba246e740afe80.rlib
-rw-r--r--. 5 1000 1001    64974 Mar 21 10:35 libadler-bbf256bbe8778f92.rlib
-rw-r--r--. 5 1000 1001  6860366 Mar 21 10:35 liballoc-936aae53ec78ce79.rlib
-rw-r--r--. 5 1000 1001     9622 Mar 21 10:35 libcfg_if-849248fcbfb57d4a.rlib
-rw-r--r--. 5 1000 1001  1665468 Mar 21 10:35 libcompiler_builtins-2ec5c0ed3ef2fbac.rlib
-rw-r--r--. 5 1000 1001 62217482 Mar 21 10:35 libcore-4901270800bec6ad.rlib
-rw-r--r--. 5 1000 1001   668370 Mar 21 10:35 libgetopts-68c9df900475ed2b.rlib
-rw-r--r--. 5 1000 1001  7928192 Mar 21 10:35 libgimli-8b9ee3164736aa23.rlib
-rw-r--r--. 5 1000 1001  1617760 Mar 21 10:35 libhashbrown-e2d51c07b00b8a6b.rlib
-rw-r--r--. 5 1000 1001  3438354 Mar 21 10:35 liblibc-6ab72c92dff0db9b.rlib
-rw-r--r--. 5 1000 1001  1380362 Mar 21 10:35 libmemchr-49babf59ca6c80c4.rlib
-rw-r--r--. 5 1000 1001  1040632 Mar 21 10:35 libminiz_oxide-3496e818e531c81a.rlib
-rw-r--r--. 5 1000 1001  9497586 Mar 21 10:35 libobject-baf4671979453700.rlib
-rw-r--r--. 5 1000 1001     9980 Mar 21 10:35 libpanic_abort-85ccf61f7c5639c2.rlib
-rw-r--r--. 5 1000 1001    33562 Mar 21 10:35 libpanic_unwind-be905a9fdaf2d1ca.rlib
-rw-r--r--. 5 1000 1001  4189476 Mar 21 10:35 libproc_macro-a0f26b1220a72e6c.rlib
-rw-r--r--. 5 1000 1001   170232 Mar 21 10:35 libprofiler_builtins-245de444f85f5d6e.rlib
-rw-r--r--. 5 1000 1001   586512 Mar 21 10:35 librustc_demangle-3a59bbe48597eb4b.rlib
-rw-r--r--. 5 1000 1001     5164 Mar 21 10:35 librustc_std_workspace_alloc-b8efbf379069eaee.rlib
-rw-r--r--. 5 1000 1001     6690 Mar 21 10:35 librustc_std_workspace_core-5ff24a720ddda7b7.rlib
-rw-r--r--. 5 1000 1001     8328 Mar 21 10:35 librustc_std_workspace_std-fb9b2a6c67b674cf.rlib
-rw-r--r--. 5 1000 1001 14955032 Mar 21 10:35 libstd-38f48d758f4bd9f6.rlib
-rwxr-xr-x. 4 1000 1001  6243944 Mar 21 10:38 libstd-38f48d758f4bd9f6.so
-rw-r--r--. 5 1000 1001   493224 Mar 21 10:35 libstd_detect-010a824ab5df37a9.rlib
-rw-r--r--. 5 1000 1001  4787096 Mar 21 10:35 libtest-9b338978aecf41d8.rlib
-rwxr-xr-x. 4 1000 1001  1219584 Mar 21 10:38 libtest-9b338978aecf41d8.so
-rw-r--r--. 5 1000 1001   146560 Mar 21 10:35 libunicode_width-40b981f8b3d8094b.rlib
-rw-r--r--. 5 1000 1001    46736 Mar 21 10:35 libunwind-a0cbc63432f5c459.rlib
```

and for nanbield:
```
# ls -l recipe-sysroot-native/usr/lib/rustlib/x86_64-unknown-linux-gnu/lib
total 119932
-rw-r--r--. 10 1000 1001   450960 Mar 21 13:54 libaddr2line-a2daade9081ae32a.rlib
-rw-r--r--. 10 1000 1001    64310 Mar 21 13:54 libadler-5b40c1d3762bf105.rlib
-rw-r--r--. 10 1000 1001  6680610 Mar 21 13:54 liballoc-d94e076697df6707.rlib
-rw-r--r--. 10 1000 1001     9598 Mar 21 13:54 libcfg_if-a13ba189ceabc5aa.rlib
-rw-r--r--. 10 1000 1001  1624834 Mar 21 13:54 libcompiler_builtins-a37d447ff9f43d61.rlib
-rw-r--r--. 10 1000 1001 58499200 Mar 21 13:54 libcore-08ada002d7296ae8.rlib
-rw-r--r--. 10 1000 1001   699320 Mar 21 13:54 libgetopts-89cd8d18b706659a.rlib
-rw-r--r--. 10 1000 1001  7122206 Mar 21 13:54 libgimli-0512ccbd2a7499d2.rlib
-rw-r--r--. 10 1000 1001  1615848 Mar 21 13:54 libhashbrown-6c124f38fcef1b87.rlib
-rw-r--r--. 10 1000 1001  3685490 Mar 21 13:54 liblibc-56e8a7f73f069ebe.rlib
-rw-r--r--. 10 1000 1001  1336470 Mar 21 13:54 libmemchr-66f5f463b5c23ed1.rlib
-rw-r--r--. 10 1000 1001   989898 Mar 21 13:54 libminiz_oxide-fbd292a712fa5921.rlib
-rw-r--r--. 10 1000 1001  8680044 Mar 21 13:54 libobject-9e334401310bf334.rlib
-rw-r--r--. 10 1000 1001     9684 Mar 21 13:54 libpanic_abort-b771eec9b0feb126.rlib
-rw-r--r--. 10 1000 1001    32882 Mar 21 13:54 libpanic_unwind-d1e19a1d9f4389d9.rlib
-rw-r--r--. 10 1000 1001  4073672 Mar 21 13:54 libproc_macro-1e61973162cf383f.rlib
-rw-r--r--. 10 1000 1001   171042 Mar 21 13:54 libprofiler_builtins-78773e035e3dc74d.rlib
-rw-r--r--. 10 1000 1001   559138 Mar 21 13:54 librustc_demangle-45e5578ce5ccb358.rlib
-rw-r--r--. 10 1000 1001     5204 Mar 21 13:54 librustc_std_workspace_alloc-d8b204e0bc283f8d.rlib
-rw-r--r--. 10 1000 1001     7026 Mar 21 13:54 librustc_std_workspace_core-c8c41820f8669080.rlib
-rw-r--r--. 10 1000 1001     8592 Mar 21 13:54 librustc_std_workspace_std-4b2a06f87e622cae.rlib
-rw-r--r--. 10 1000 1001   512464 Mar 21 13:54 libstd_detect-9960905b690242e4.rlib
-rw-r--r--. 10 1000 1001 13806534 Mar 21 13:54 libstd-f1077750b2cf6857.rlib
-rwxr-xr-x.  9 1000 1001  5768264 Mar 21 14:01 libstd-f1077750b2cf6857.so
-rw-r--r--. 10 1000 1001  4868416 Mar 21 13:54 libtest-314aab89245a6e5d.rlib
-rwxr-xr-x.  9 1000 1001  1219832 Mar 21 14:01 libtest-314aab89245a6e5d.so
[... 80 lines cut ...]
-rw-r--r--. 7 1000 1001     6794 Mar 21 14:14 librustc_std_workspace_core-3c6e23a841933792.rlib
-rw-r--r--. 7 1000 1001     3621 Mar 21 14:14 librustc_std_workspace_core-3c6e23a841933792.rmeta
-rw-r--r--. 4 1000 1001     6986 Mar 21 15:04 librustc_std_workspace_core-8a8ed05c8a58baf0.rlib
-rw-r--r--. 4 1000 1001     8568 Mar 21 15:04 librustc_std_workspace_std-afcb3739af0fbd90.rlib
-rw-r--r--. 7 1000 1001     8360 Mar 21 14:14 librustc_std_workspace_std-d61a050d2873a6cf.rlib
[... cut ...]
-rw-r--r--. 7 1000 1001   143720 Mar 21 14:14 libunicode_width-94c2df26c937531a.rlib
-rw-r--r--. 7 1000 1001   126325 Mar 21 14:14 libunicode_width-94c2df26c937531a.rmeta
-rw-r--r--. 4 1000 1001   144296 Mar 21 15:04 libunicode_width-a6178fb168c0a0d7.rlib
-rw-r--r--. 4 1000 1001    45208 Mar 21 15:04 libunwind-74d673a682db49a1.rlib
-rw-r--r--. 7 1000 1001    45040 Mar 21 14:14 libunwind-78f011020e29e916.rlib
-rw-r--r--. 7 1000 1001    35695 Mar 21 14:14 libunwind-78f011020e29e916.rmeta
drwxr-xr-x. 2 1000 1001     4096 Mar 21 16:34 self-contained
```

rjanani-p, 2024-03-26:
@MaxIhlenfeldt , I am getting build issue for wrlinux master branch (branch having the latest meta-browser master branch all the commits), sharing the build log for reference 

ERROR: chromium-x11-121.0.6167.184-r0 do_copy_target_rustlibs: ExecutionError('/buildarea/eng1/rpandi/wrkspc/ltsrealses/lincd_26032024/lincd_chromium_121/tmp-glibc/work/core2-64-wrs-linux/chromium-x11/121.0.6167.184/temp/run.do_copy_target_rustlibs.710741', 1, None, None)
ERROR: Logfile of failure stored in: /buildarea/eng1/rpandi/wrkspc/ltsrealses/lincd_26032024/lincd_chromium_121/tmp-glibc/work/core2-64-wrs-linux/chromium-x11/121.0.6167.184/temp/log.do_copy_target_rustlibs.710741
Log data follows:
| DEBUG: Executing **shell function do_copy_target_rustlibs**
| cp: **cannot stat '/buildarea/eng1/rpandi/wrkspc/ltsrealses/lincd_26032024/lincd_chromium_121/tmp-glibc/work/core2-64-wrs-linux/chromium-x11/121.0.6167.184/recipe-sysroot/usr/lib/rustlib/x86_64*': No such file or directory**
| WARNING: /buildarea/eng1/rpandi/wrkspc/ltsrealses/lincd_26032024/lincd_chromium_121/tmp-glibc/work/core2-64-wrs-linux/chromium-x11/121.0.6167.184/temp/run.do_copy_target_rustlibs.710741:149 exit 1 from 'cp -r $rustlib_src_dir "/buildarea/eng1/rpandi/wrkspc/ltsrealses/lincd_26032024/lincd_chromium_121/tmp-glibc/work/core2-64-wrs-linux/chromium-x11/121.0.6167.184/recipe-sysroot-native/usr/lib/rustlib"'
| WARNING: Backtrace (BB generated script):
|       #1: do_copy_target_rustlibs, /buildarea/eng1/rpandi/wrkspc/ltsrealses/lincd_26032024/lincd_chromium_121/tmp-glibc/work/core2-64-wrs-linux/chromium-x11/121.0.6167.184/temp/run.do_copy_target_rustlibs.710741, line 149
|       #2: main, /buildarea/eng1/rpandi/wrkspc/ltsrealses/lincd_26032024/lincd_chromium_121/tmp-glibc/work/core2-64-wrs-linux/chromium-x11/121.0.6167.184/temp/run.do_copy_target_rustlibs.710741, line 152

sharing the recipe-sysroot-native/usr/lib/rustlib/x86_64-unknown-linux-gnu/lib/ contents too, could you please suggest how to resolve the issue
**ls -l recipe-sysroot-native/usr/lib/rustlib/x86_64-unknown-linux-gnu/lib/**
total 105604
-rw-r--r-- 5  users   457828 Mar 26 13:01 libaddr2line-13b90ddabcfcdff7.rlib
-rw-r--r-- 5  users    34612 Mar 26 13:01 libadler-c41e85cdd0e0dfff.rlib
-rw-r--r-- 5  users  6289116 Mar 26 13:01 liballoc-bc9174b398261284.rlib
-rw-r--r-- 5  users    10478 Mar 26 13:01 libcfg_if-6771b1ae9cbfc2c9.rlib
-rw-r--r-- 5  users  1394422 Mar 26 13:01 libcompiler_builtins-e8cdafc9faa29ecd.rlib
-rw-r--r-- 5  users 47318932 Mar 26 13:01 libcore-baa5449bbf3e5ae7.rlib
-rw-r--r-- 5  users   474364 Mar 26 13:01 libgetopts-b24eeb95e8a36a72.rlib
-rw-r--r-- 5  users  6120184 Mar 26 13:01 libgimli-d24d3e21c4b6d183.rlib
-rw-r--r-- 5  users  1614906 Mar 26 13:01 libhashbrown-67ad1ad36ffec836.rlib
-rw-r--r-- 5  users  3409146 Mar 26 13:01 liblibc-84992ee57e15ddc0.rlib
-rw-r--r-- 5  users   872422 Mar 26 13:01 libmemchr-4ff2a73349a27351.rlib
-rw-r--r-- 5  users   327174 Mar 26 13:01 libminiz_oxide-0f5ce74ab0128e5b.rlib
-rw-r--r-- 5  users  7999318 Mar 26 13:01 libobject-45b585256bbdad6f.rlib
-rw-r--r-- 5  users    10020 Mar 26 13:01 libpanic_abort-a818357d2c99a5c1.rlib
-rw-r--r-- 5  users    23150 Mar 26 13:01 libpanic_unwind-56ffc7344a3fa9ec.rlib
-rw-r--r-- 5  users  3393222 Mar 26 13:01 libproc_macro-897be83c186b3211.rlib
-rw-r--r-- 5  users   379940 Mar 26 13:01 librustc_demangle-09fdf503f250d6a7.rlib
-rw-r--r-- 5  users     5900 Mar 26 13:01 librustc_std_workspace_alloc-ab6e54dadf25bfd1.rlib
-rw-r--r-- 5  users     8658 Mar 26 13:01 librustc_std_workspace_core-ce683bfa6346b7ff.rlib
-rw-r--r-- 5  users    10432 Mar 26 13:01 librustc_std_workspace_std-e165636351474423.rlib
-rw-r--r-- 5  users 12502812 Mar 26 13:01 libstd-64729a5a222c378b.rlib
-rwxr-xr-x 4  users  9211704 Mar 26 13:01 libstd-64729a5a222c378b.so
-rw-r--r-- 5  users   433536 Mar 26 13:01 libstd_detect-4c6d792e86d76f74.rlib
-rw-r--r-- 5  users     5808 Mar 26 13:01 libsysroot-5e30f112053f646e.rlib
-rw-r--r-- 5  users  4093024 Mar 26 13:01 libtest-2571735223127832.rlib
-rwxr-xr-x 4  users  1493224 Mar 26 13:01 libtest-2571735223127832.so
-rw-r--r-- 5  users   150344 Mar 26 13:01 libunicode_width-a4de7578f009194a.rlib
-rw-r--r-- 5  users    41702 Mar 26 13:01 libunwind-b9b23c28f438e60f.rlib

MaxIhlenfeldt, 2024-03-27:
Could both of you please upload the output of `bitbake -e chromium-x11` to e.g. https://pastebin.com/ and paste the link here?

zboszor, 2024-03-27:
Pastebin limit is 512KB, the `bitbake -e` output is over 2MB in my case. Attached as gzipped here instead.
[chromium-x11-vars.txt.gz](https://github.com/OSSystems/meta-browser/files/14772791/chromium-x11-vars.txt.gz)

MaxIhlenfeldt, 2024-03-28:
Thanks! That's helpful, but I'm still not fully sure what's going on. Could you please also attach the output of `bitbake -e rust` as well as the contents of `temp/run.do_install` for the `rust` recipe?

rjanani-p, 2024-04-04:
> Could both of you please upload the output of `bitbake -e chromium-x11` to e.g. https://pastebin.com/ and paste the link here?

[chromium-x11.txt](https://github.com/OSSystems/meta-browser/files/14861010/chromium-x11.txt)
could you please share your inputs, I have observed issue with qemux86-64,qemuarm64  targets , not facing with qemuarm.

MaxIhlenfeldt, 2024-04-04:
Thank you both! I'm still not really sure what the problem is. Do you maybe have a `repo` manifest you can share with me, together with all your local `.conf` files, so I can replicate your setup as close as possible?

zboszor, 2024-04-11:
Not sure if this influences anything, but I also have this bbappend:
```
$ cat chromium-x11_%.bbappend 
PACKAGECONFIG:remove = "use-egl"
PACKAGECONFIG:append:class-target = " gtk4 proprietary-codecs cups use-vaapi "

DEPENDS:append = " wayland-protocols wayland"

RDEPENDS:${PN}:append = " gtk4-printbackend-cups "
```

zboszor, 2024-04-15:
FWIW, the same issue occurs with Yocto 5.0 Scarthgap for us.

I can't share the whole set of layers from our manifest, as some of them are internal layers. Here's the public part of `default.xml`:
```
<manifest>
  <remote fetch="https://github.com" name="github"/>
  <remote fetch="http://git.linaro.org" name="linaro"/>
  <remote fetch="git://git.openembedded.org" name="oe"/>
  <remote fetch="http://git.yoctoproject.org/git" name="yocto"/>
  
  <default revision="scarthgap" sync-j="4"/>
  
  <project name="OSSystems/meta-browser" path="layers/meta-browser" remote="github" revision="master" />
  <project name="zboszor/meta-parallel-php" path="layers/meta-parallel-php" remote="github" />
  <project name="agherzan/meta-raspberrypi" path="layers/meta-raspberrypi" remote="github" revision="master" />
  <project name="kraj/meta-clang" path="layers/meta-clang" remote="github" revision="master" />
  <project name="meta-intel" path="layers/meta-intel" remote="yocto" />
  <project name="meta-security" path="layers/meta-security" remote="yocto" revision="master" />
  <project name="meta-yocto" path="layers/meta-yocto" remote="yocto"  />
  <project name="lucimber/meta-openjdk-temurin" path="layers/meta-openjdk-temurin" remote="github" revision="main" />
  <project name="meta-virtualization" path="layers/meta-virtualization" remote="yocto" />
  <project name="meta-cloud-services" path="layers/meta-cloud-services" remote="yocto" />
  <project name="openembedded/bitbake" path="bitbake" remote="github" revision="2.8"/>
  <project name="openembedded/meta-openembedded" path="layers/meta-openembedded" remote="github"/>
  <project name="openembedded/openembedded-core" path="layers/openembedded-core" remote="github"/>
  <project name="DynamicDevices/meta-mono" path="layers/meta-mono" remote="github" revision="master" />
  <project name="dimtass/meta-elastic-beats" path="layers/meta-elastic-beats" remote="github" revision="master"/>
  <project name="zboszor/meta-python-ai" path="layers/meta-python-ai" remote="github" />
</manifest>
```
Some of these layers are not officially scarthgap compatible yet. For them, the layer compatibility setting in `conf/layer.conf` is manually changed and nothing else.

MaxIhlenfeldt, 2024-04-16:
Thanks, I've kicked off a build that should be as close as possible as yours. Let's see whether I'll now get the error as well :crossed_fingers: 

fyi, I'll be ooo for the rest of the week, but I'll get back to this next week.

zboszor, 2024-04-17:
> Hm, I get this error:
> 
> ```
> ERROR: chromium-x11-122.0.6261.128-r0 do_prepare_recipe_sysroot: The file /usr/bin/llvm-split is installed by both llvm-native and clang-native, aborting
> ```
> 
> Any idea what the difference between your setup and mine could be that leads to this, @zboszor?

It's [documented](https://github.com/kraj/meta-clang?tab=readme-ov-file#providing-llvm) in `meta-clang`. To replace `llvm` with `clang`, you must have these in your local configuration:
```
PREFERRED_PROVIDER_llvm = "clang"
PREFERRED_PROVIDER_llvm-native = "clang-native"
PREFERRED_PROVIDER_nativesdk-llvm = "nativesdk-clang"
PROVIDES:pn-clang = "llvm"
PROVIDES:pn-clang-native = "llvm-native"
PROVIDES:pn-nativesdk-clang = "nativesdk-llvm"
```

MaxIhlenfeldt, 2024-04-22:
Ok, I've finally been able to reproduce it. I suspect that due to the `gtk4` config in your `chromium-x11_%.bbappend`:
```
PACKAGECONFIG:append:class-target = " gtk4 proprietary-codecs cups use-vaapi "
```
Chromium gets a new (transitive) dependency on `librsvg`, which uses Rust and depends on `libstd-rs`. Thus, we get the files from that in our sysroot even though we try to avoid it in our [`rust_%.bbappend`](https://github.com/OSSystems/meta-browser/blob/master/meta-chromium/recipes-browser/chromium/rust_%25.bbappend#L26).

i.e. our approach breaks as soon as we have anything inheriting the `rust` class (which inherits the `rust-common` class, which contains `RUSTLIB_DEP ?= "libstd-rs"`). Didn't think of that when writing that PR :sweat_smile: 

I think we might be able to drop our custom `rust_do_install` logic, since we've decided in #791 to not require `libprofiler_builtins`. We could instead just patch Chromium to accept `.rlib` files without a metadata hash in the filename and should then be able to use the regular `libstd-rs` dependency.

I'll give the approach outlined above a try and will report back here.

rjanani-p, 2024-04-23:
hi @MaxIhlenfeldt , above changes will fix my issue?
DEBUG: Executing shell function do_copy_target_rustlibs
| cp: cannot stat '/lincd_chromium_121/tmp-glibc/work/core2-64-wrs-linux/chromium-x11/121.0.6167.184/recipe-sysroot/usr/lib/rustlib/x86_64': No such file or directory*

MaxIhlenfeldt, 2024-04-23:
@rjanani-p I'm still working on the fix, the `PACKAGECONFIG` changes I mentioned in my last comment were only needed for me to reproduce the issue.

rjanani-p, 2024-04-25:
@MaxIhlenfeldt , regarding 
DEBUG: Executing shell function do_copy_target_rustlibs
| cp: cannot stat '/lincd_chromium_121/tmp-glibc/work/core2-64-wrs-linux/chromium-x11/121.0.6167.184/recipe-sysroot/usr/lib/rustlib/x86_64': No such file or directory*

I am getting above issue for 122.0.6261.128 version also for qemux86-64, intel-x86-64, qemuarm64 machines.
Here I am sharing the config files, could you please share your input to reproduce the issue files are sufficient? need to provide any files?
[bblayers.txt](https://github.com/OSSystems/meta-browser/files/15105179/bblayers.txt)
[local.txt](https://github.com/OSSystems/meta-browser/files/15105184/local.txt)

MaxIhlenfeldt, 2024-05-07:
Also, depending on how recent your checkout is, you might see a ninja error on Yocto master. If that occurs, backporting https://chromium-review.googlesource.com/c/chromium/src/+/5522398 should fix it. I'll upload a separate PR for that tomorrow.

MaxIhlenfeldt, 2024-05-07:
PR for the master fix is up already at https://github.com/OSSystems/meta-browser/pull/808.

zboszor, 2024-05-08:
Thank you very much, I will try your patch soon.

zboszor, 2024-05-08:
I have tried it under mickledore first (after re-enabling mickledore support in meta-browser master).

An unrelated build error occurred:
```
| FAILED: obj/base/allocator/partition_allocator/src/partition_alloc/allocator_core/lightweight_quarantine.o
| x86_64-oe-linux-clang++ -target x86_64-oe-linux  -m64 -march=nehalem -mtune=generic -mfpmath=sse -msse4.2 -mlittle-endian --dyld-prefix=/usr -Qunused-arguments -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/zozo/test-yocto-4.2-gh/tmp-sicom-glibc/work/corei7-64-oe-linux/chromium-x11/122.0.6261.128-r0/recipe-sysroot -MMD -MF obj/base/allocator/partition_allocator/src/partition_alloc/allocator_core/lightweight_quarantine.o.d -DUSE_UDEV -DUSE_AURA=1 -DUSE_GLIB=1 -DUSE_OZONE=1 -DOFFICIAL_BUILD -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D_FORTIFY_SOURCE=2 -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -DNO_UNWIND_TABLES -D_GNU_SOURCE -D_LIBCPP_HARDENING_MODE=_LIBCPP_HARDENING_MODE_NONE -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -DIS_PARTITION_ALLOC_IMPL -I../.. -Igen -I../../base/allocator/partition_allocator/src -Igen/base/allocator/partition_allocator/src -Wall -Wextra -Wimplicit-fallthrough -Wextra-semi -Wunreachable-code-aggressive -Wthread-safety -Wno-missing-field-initializers -Wno-unused-parameter -Wno-psabi -Wloop-analysis -Wno-unneeded-internal-declaration -Wenum-compare-conditional -Wno-unknown-warning-option -Wno-ignored-pragma-optimize -Wno-deprecated-builtins -Wno-bitfield-constant-conversion -Wno-deprecated-this-capture -Wno-invalid-offsetof -Wno-vla-extension -Wno-thread-safety-reference-return -Wshadow -fno-delete-null-pointer-checks -fno-ident -fno-strict-aliasing -fstack-protector -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -pthread -fcolor-diagnostics -fmerge-all-constants -fcrash-diagnostics-dir=../../tools/clang/crashreports -mllvm -instcombine-lower-dbg-declare=0 -ffp-contract=off -flto=thin -fsplit-lto-unit -fwhole-program-vtables -m64 -msse3 -ffile-compilation-dir=. -ftrivial-auto-var-init=pattern -fno-omit-frame-pointer -g0 -fvisibility=hidden -Wheader-hygiene -Wstring-conversion -Wtautological-overlap-compare -Wexit-time-destructors -O3 -fdata-sections -ffunction-sections -fno-unique-section-names -fno-math-errno -Wc++11-narrowing -Wconditional-uninitialized -Wcstring-format-directive -Wctad-maybe-unsupported -Wdeprecated-copy -Wdeprecated-copy-dtor -Wduplicate-enum -Wextra-semi -Wextra-semi-stmt -Wimplicit-fallthrough -Winconsistent-missing-destructor-override -Winvalid-offsetof -Wmissing-field-initializers -Wnon-c-typedef-for-linkage -Wpessimizing-move -Wrange-loop-analysis -Wredundant-move -Wshadow-field -Wstrict-prototypes -Wsuggest-destructor-override -Wsuggest-override -Wtautological-unsigned-zero-compare -Wunreachable-code-aggressive -Wunused-but-set-variable -Wunused-macros -pedantic -Wno-c++11-narrowing-const-reference -std=c++20 -Wno-trigraphs -gsimple-template-names -fno-exceptions -fno-rtti -fvisibility-inlines-hidden  -O2 -pipe  -feliminate-unused-debug-types -fmacro-prefix-map=/home/zozo/test-yocto-4.2-gh/tmp-sicom-glibc/work/corei7-64-oe-linux/chromium-x11/122.0.6261.128-r0/chromium-122.0.6261.128=/usr/src/debug/chromium-x11/122.0.6261.128-r0  -fdebug-prefix-map=/home/zozo/test-yocto-4.2-gh/tmp-sicom-glibc/work/corei7-64-oe-linux/chromium-x11/122.0.6261.128-r0/chromium-122.0.6261.128=/usr/src/debug/chromium-x11/122.0.6261.128-r0  -fmacro-prefix-map=/home/zozo/test-yocto-4.2-gh/tmp-sicom-glibc/work/corei7-64-oe-linux/chromium-x11/122.0.6261.128-r0/chromium-122.0.6261.128/out/Release=/usr/src/debug/chromium-x11/122.0.6261.128-r0  -fdebug-prefix-map=/home/zozo/test-yocto-4.2-gh/tmp-sicom-glibc/work/corei7-64-oe-linux/chromium-x11/122.0.6261.128-r0/chromium-122.0.6261.128/out/Release=/usr/src/debug/chromium-x11/122.0.6261.128-r0  -fdebug-prefix-map=/home/zozo/test-yocto-4.2-gh/tmp-sicom-glibc/work/corei7-64-oe-linux/chromium-x11/122.0.6261.128-r0/recipe-sysroot=  -fmacro-prefix-map=/home/zozo/test-yocto-4.2-gh/tmp-sicom-glibc/work/corei7-64-oe-linux/chromium-x11/122.0.6261.128-r0/recipe-sysroot=  -fdebug-prefix-map=/home/zozo/test-yocto-4.2-gh/tmp-sicom-glibc/work/corei7-64-oe-linux/chromium-x11/122.0.6261.128-r0/recipe-sysroot-native= -g1 -fvisibility-inlines-hidden -stdlib=libc++ -c ../../base/allocator/partition_allocator/src/partition_alloc/lightweight_quarantine.cc -o obj/base/allocator/partition_allocator/src/partition_alloc/allocator_core/lightweight_quarantine.o
| In file included from ../../base/allocator/partition_allocator/src/partition_alloc/lightweight_quarantine.cc:7:
| ../../base/allocator/partition_allocator/src/partition_alloc/internal_allocator.h:29:1: error: missing 'typename' prior to dependent type name 'InternalAllocator<T>::value_type'
| InternalAllocator<T>::value_type* InternalAllocator<T>::allocate(
| ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
| typename
| 1 error generated.
```

Since meta-browser master does not officially support mickledore, I will skip this and test it on nanbield and scarthgap.

MaxIhlenfeldt, 2024-05-08:
> Since meta-browser master does not officially support mickledore, I will skip this and test it on nanbield and scarthgap.

Yes, we had to drop support for mickledore as its version of clang was too old. It should work on both nanbield and scarthgap.

rjanani-p, 2024-05-08:
@MaxIhlenfeldt ,regarding
DEBUG: Executing shell function do_copy_target_rustlibs
| cp: cannot stat '/lincd_chromium_121/tmp-glibc/work/core2-64-wrs-linux/chromium-x11/121.0.6167.184/recipe-sysroot/usr/lib/rustlib/x86_64': No such file or directory*

with below change my issue got resolved ,
do_copy_target_rustlibs () {
     # Chromium needs a single Rust sysroot that contains the rustlibs for both
     # the host and target, so we copy the target rustlibs to the native sysroot.
**-    rustlib_src_dir="${RECIPE_SYSROOT}/usr/lib/rustlib/${TARGET_ARCH}"*
-    cp -r $rustlib_src_dir "${RECIPE_SYSROOT_NATIVE}/usr/lib/rustlib"
+    rustlib_src_dir="${STAGING_LIBDIR}/rustlib/${TARGET_ARCH}"*
+    cp -r $rustlib_src_dir "${STAGING_LIBDIR_NATIVE}/rustlib"**
 }

the error will happen when multilib is enabled.

MaxIhlenfeldt, 2024-05-08:
@rjanani-p thanks for pointing that out. I'll fix it in a separate PR as we should also use the `STAGING_*` variables in the `do_add_clang_symlink` and `do_copy_clang_library` tasks.

zboszor, 2024-05-08:
@MaxIhlenfeldt I will have the test result for nanbield in about 2-3 hours. Please wait until my build finishes and if possible, merge #809 before #806 so our nanbield builds can get chromium 122. Thanks in advance.

zboszor, 2024-05-08:
@MaxIhlenfeldt My chromium build passed the point where the above quoted errors occurred. With a bit of luck, it finishes. FWIW, my Yocto setup does not use `multilib`.

zboszor, 2024-05-09:
@MaxIhlenfeldt `chromium-x11` has been built successfully with your patch from https://github.com/OSSystems/meta-browser/issues/792#issuecomment-2098778290

zboszor, 2024-05-09:
@MaxIhlenfeldt The scarthgap build failed with:
```
Sstate summary: Wanted 1227 Local 112 Mirrors 0 Missed 1115 Current 2395 (9% match, 69% complete)########################################################################################################                    | ETA:  0:00:00
Initialising tasks: 100% |###################################################################################################################################################################################################| Time: 0:00:05
NOTE: Executing Tasks
WARNING: systemd-1_255.4-r0 do_install: Using /home/root as root user's home directory is not fully supported by systemd
ERROR: chromium-x11-122.0.6261.128-r0 do_compile: ExecutionError('/home/zozo/test-yocto-5.0/tmp-sicom-glibc/work/corei7-64-oe-linux/chromium-x11/122.0.6261.128/temp/run.do_compile.121841', 1, None, None)
ERROR: Logfile of failure stored in: /home/zozo/test-yocto-5.0/tmp-sicom-glibc/work/corei7-64-oe-linux/chromium-x11/122.0.6261.128/temp/log.do_compile.121841
Log data follows:
| DEBUG: Executing shell function do_compile
| ninja: error: '../../../recipe-sysroot-native/usr/lib/clang/latest/lib/linux/libclang_rt.builtins-x86_64.a', needed by 'yocto_native/obj/third_party/protobuf/libprotoc_lib.a', missing and no known rule to make it
| WARNING: /home/zozo/test-yocto-5.0/tmp-sicom-glibc/work/corei7-64-oe-linux/chromium-x11/122.0.6261.128/temp/run.do_compile.121841:147 exit 1 from 'ninja -v -j 16 -l 32 chrome chrome_sandbox chromedriver.unstripped'
| WARNING: Backtrace (BB generated script):
| 	#1: do_compile, /home/zozo/test-yocto-5.0/tmp-sicom-glibc/work/corei7-64-oe-linux/chromium-x11/122.0.6261.128/temp/run.do_compile.121841, line 147
| 	#2: main, /home/zozo/test-yocto-5.0/tmp-sicom-glibc/work/corei7-64-oe-linux/chromium-x11/122.0.6261.128/temp/run.do_compile.121841, line 151
ERROR: Task (/home/zozo/test-yocto-5.0/conf/../layers/meta-browser/meta-chromium/recipes-browser/chromium/chromium-x11_122.0.6261.128.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 7747 tasks of which 5608 didn't need to be rerun and 1 failed.
NOTE: Writing buildhistory
NOTE: Writing buildhistory took: 10 seconds
```

rjanani-p, 2024-05-09:
> @rjanani-p thanks for pointing that out. I'll fix it in a separate PR as we should also use the `STAGING_*` variables in the `do_add_clang_symlink` and `do_copy_clang_library` tasks.

sure @MaxIhlenfeldt , thank you

MaxIhlenfeldt, 2024-05-09:
> @MaxIhlenfeldt I will have the test result for nanbield in about 2-3 hours. Please wait until my build finishes and if possible, merge #809 before #806 so our nanbield builds can get chromium 122. Thanks in advance.

As usual, we'll create a `nanbield` branch once #806 has been merged. Should that happen before #809 is merged, we'll of course cherry-pick it to that branch. (Feel free to remind me or submit a PR yourself should I forget to do so.)

MaxIhlenfeldt, 2024-05-09:
> @MaxIhlenfeldt The scarthgap build failed with:
> ...

Hm, I didn't touch anything related to the clang setup... maybe it's related to the same issue @rjanani-p reported. Can you please try if the error still happens if you also apply the changes from #810?

If the error persists, please paste the output of  `ls -al recipe-sysroot-native/usr/lib/clang` and `tree recipe-sysroot-native/usr/lib/clang`.

zboszor, 2024-05-09:
#810 fails the same way. Here's is the info you asked for:
```
chromium-x11/122.0.6261.128$ ls -l recipe-sysroot-native/usr/lib/clang
összesen 12
drwxr-xr-x. 3 zozo zozo 4096 ápr   22 12.38 18
drwxr-xr-x. 4 zozo zozo 4096 máj    9 12.23 18.1.3
drwxr-xr-x. 4 zozo zozo 4096 máj    9 09.53 18.1.5
lrwxrwxrwx. 1 zozo zozo    6 ápr   22 12.38 latest -> 18.1.3
```
and
```
chromium-x11/122.0.6261.128$ tree recipe-sysroot-native/usr/lib/clang
recipe-sysroot-native/usr/lib/clang
├── 18
│   └── include
│       ├── adcintrin.h
│       ├── adxintrin.h
│       ├── altivec.h
│       ├── ammintrin.h
│       ├── amxcomplexintrin.h
│       ├── amxfp16intrin.h
│       ├── amxintrin.h
│       ├── arm64intr.h
│       ├── arm_acle.h
│       ├── arm_bf16.h
│       ├── arm_cde.h
│       ├── arm_cmse.h
│       ├── arm_fp16.h
│       ├── armintr.h
│       ├── arm_mve.h
│       ├── arm_neon.h
│       ├── arm_neon_sve_bridge.h
│       ├── arm_sme.h
│       ├── arm_sve.h
│       ├── arm_vector_types.h
│       ├── avx2intrin.h
│       ├── avx512bf16intrin.h
│       ├── avx512bitalgintrin.h
│       ├── avx512bwintrin.h
│       ├── avx512cdintrin.h
│       ├── avx512dqintrin.h
│       ├── avx512erintrin.h
│       ├── avx512fintrin.h
│       ├── avx512fp16intrin.h
│       ├── avx512ifmaintrin.h
│       ├── avx512ifmavlintrin.h
│       ├── avx512pfintrin.h
│       ├── avx512vbmi2intrin.h
│       ├── avx512vbmiintrin.h
│       ├── avx512vbmivlintrin.h
│       ├── avx512vlbf16intrin.h
│       ├── avx512vlbitalgintrin.h
│       ├── avx512vlbwintrin.h
│       ├── avx512vlcdintrin.h
│       ├── avx512vldqintrin.h
│       ├── avx512vlfp16intrin.h
│       ├── avx512vlintrin.h
│       ├── avx512vlvbmi2intrin.h
│       ├── avx512vlvnniintrin.h
│       ├── avx512vlvp2intersectintrin.h
│       ├── avx512vnniintrin.h
│       ├── avx512vp2intersectintrin.h
│       ├── avx512vpopcntdqintrin.h
│       ├── avx512vpopcntdqvlintrin.h
│       ├── avxifmaintrin.h
│       ├── avxintrin.h
│       ├── avxneconvertintrin.h
│       ├── avxvnniint16intrin.h
│       ├── avxvnniint8intrin.h
│       ├── avxvnniintrin.h
[... 175 lines cut ...]
│       ├── xopintrin.h
│       ├── xsavecintrin.h
│       ├── xsaveintrin.h
│       ├── xsaveoptintrin.h
│       ├── xsavesintrin.h
│       └── xtestintrin.h
├── 18.1.3
│   ├── 18.1.3 -> 18.1.3
│   ├── 18.1.5 -> 18.1.5
│   ├── include
│   │   └── orc
│   └── lib
│       └── linux
├── 18.1.5
│   ├── include
│   │   └── orc
│   │       └── c_api.h
│   └── lib
│       └── linux
│           ├── libclang_rt.builtins-x86_64.a
│           └── liborc_rt-x86_64.a
└── latest -> 18.1.3

19 directories, 236 files
```

MaxIhlenfeldt, 2024-05-09:
Thanks. I'm pretty sure this is #794. `bitbake -c cleanall clang chromium-x11` and then rebuilding should fix the issue, sorry that I haven't gotten around to fix that yet.

zboszor, 2024-05-09:
Thanks, I'll try, with cleaning up `clang-native` as well, just to make sure.

zboszor, 2024-05-09:
The scarthgap build has passed the point of the previously quoted errors, it will succeed. It's almost at 20% now.

zboszor, 2024-05-10:
And indeed, it finished succesfully. #809 can be merged as far as I am concerned.

MaxIhlenfeldt, 2024-05-10:
@zboszor thanks for confirming! I've just updated #809 to get rid of all of our bbappends. If it's not too much to ask, it would be great if you could verify that it still works for you.

zboszor, 2024-05-10:
I will run the build both for nanbield and scarthgap during the weekend with the changes.

zboszor, 2024-05-11:
With the new contents of #809 , the scarthgap build finished successfully. On to nanbield.

zboszor, 2024-05-11:
The nanbield build also succeeded.

MaxIhlenfeldt, 2024-05-13:
Thanks for testing!

PR #809 (2024-05-13): chromium: Depend on libstd-rs instead of rust
Fixes #792.

Build and patch changes:
------------------------

In #782, we decided to depend on rust instead of libstd-rs, because the latter didn't include libprofiler_builtins and also used a naming scheme that trips up Chromium.

However, in #791 we decided to patch Chromium so that it doesn't need libprofiler_builtins any more, because the master version of the rust recipe also doesn't include it.

Finally, while investigating #792 it turned out that our approach breaks as soon as we have something that depends on libstd-rs in our dependency g [... cut]
commit d8bcd04a (2024-05-13): chromium: Depend on libstd-rs instead of rust (#809)

chromium: Depend on libstd-rs instead of rust

Fixes #792.

Build and patch changes:
------------------------

In #782, we decided to depend on rust instead of libstd-rs, because the
latter didn't include libprofiler_builtins and also used a naming scheme
that trips up Chromium.

However, in #791 we decided to patch Chromium so that it doesn't need
libprofiler_builtins any more, because the master version of the rust
recipe also doesn't include it.

Finally, while investigating #792 it turned out that our approach breaks
as soon as we have  [... cut]
closed 2024-05-13 by MaxIhlenfeldt