OSSystems/meta-browser #794: 18.1.1 clang compilation issue

VaginAY, 2024-03-18:
```
**for chromium-ozone-wayland & qemux86-64 target:** getting compilation issue, sharing the log below for reference
   **NOTE: Executing Tasks
   ERROR: chromium-ozone-wayland-121.0.6167.184-r0 do_compile: ExecutionError('/buildarea/eng1/rpandi/wrkspc/chromium/buildchromium/build_poky_master/tmp/work/core2-64-poky-linux/chromium-ozone-wayland/121.0.6167.184/temp/run.do_compile.878508', 1, None, None)
   ERROR: Logfile of failure stored in: /buildarea/eng1/rpandi/wrkspc/chromium/buildchromium/build_poky_master/tmp/work/core2-64-poky-linux/chromium-ozone-wayland/121.0.6167.184/temp/log.do_compile.878508
   Log data follows:
   | DEBUG: Executing shell function do_compile
   | ninja: error: '../../../recipe-sysroot-native/usr/lib/clang/latest/lib/linux/libclang_rt.builtins-x86_64.a', needed by 'yocto_native/obj/third_party/protobuf/libprotoc_lib.a', missing and no known rule to make it
   | WARNING: /buildarea/eng1/rpandi/wrkspc/chromium/buildchromium/build_poky_master/tmp/work/core2-64-poky-linux/chromium-ozone-wayland/121.0.6167.184/temp/run.do_compile.878508:146 exit 1 from 'ninja -v -j 64 chrome chrome_sandbox chromedriver.unstripped'
   | WARNING: Backtrace (BB generated script):**
```

I got this issue when i add master branch to my manifest without revision and meta-clang automatically takes from latest master branch commit. When i set revision in manifest to eebe4ff2e539f3ffb01c5060cc4ca8b226ea8b52 commit (lksctp-tools: Reset SECURITY_STACK_PROTECTOR with clang) , last before bump to 18.1.1 version, all is good.

MaxIhlenfeldt, 2024-03-20:
Thanks! I probably won't be able take a look before the pending updates (first to m122, then to m123) are done, as these are more urgent.

SunboX, 2024-03-21:
I'm getting a similar issue. Is there anyway to work around it for now? Before sync'ing my repo it worked fine.

```
| [18013/63163] clang++ -MMD -MF yocto_native/obj/third_party/flatbuffers/compiler_files/binary_annotator.o.d -DUSE_UDEV -DUSE_AURA=1 -DUSE_GLIB=1 -DUSE_OZONE=1 -DOFFICIAL_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -DNO_UNWIND_TABLES -D_GNU_SOURCE -D_LIBCPP_HARDENING_MODE=_LIBCPP_HARDENING_MODE_NONE -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -DFLATBUFFERS_LOCALE_INDEPENDENT=0 -I../../third_party/flatbuffers/src/grpc -I../.. -Iyocto_native/gen -I../../third_party/flatbuffers/src/include -I../../third_party/flatbuffers/src/src -I../../third_party/flatbuffers/src -I../../third_party/flatbuffers/generated -fno-delete-null-pointer-checks -fno-ident -fno-strict-aliasing -fstack-protector -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -pthread -fcolor-diagnostics -fmerge-all-constants -fcrash-diagnostics-dir=../../tools/clang/crashreports -mllvm -instcombine-lower-dbg-declare=0 -ffp-contract=off -mno-outline -ffile-compilation-dir=. -ftrivial-auto-var-init=pattern -O2 -fdata-sections -ffunction-sections -fno-unique-section-names -fno-math-errno -fno-omit-frame-pointer -g0 -fvisibility=hidden -Wheader-hygiene -Wstring-conversion -Wtautological-overlap-compare -Wall -Wno-unused-variable -Wno-c++11-narrowing -Wno-unused-but-set-variable -Wno-misleading-indentation -Wno-missing-field-initializers -Wno-unused-parameter -Wno-psabi -Wloop-analysis -Wno-unneeded-internal-declaration -Wenum-compare-conditional -Wno-unknown-warning-option -Wno-ignored-pragma-optimize -Wno-deprecated-builtins -Wno-bitfield-constant-conversion -Wno-deprecated-this-capture -Wno-invalid-offsetof -Wno-vla-extension -Wno-thread-safety-reference-return -Wno-constant-conversion -Wno-shorten-64-to-32 -std=c++20 -Wno-trigraphs -gsimple-template-names -fno-exceptions -fno-rtti -fvisibility-inlines-hidden -isystem/home/parallels/gewa-platform/build/tmp/work/cortexa9hf-neon-imx-gel-linux-gnueabi/chromium-ozone-wayland/121.0.6167.184-r0/recipe-sysroot-native/usr/include -isysroot=/home/parallels/gewa-platform/build/tmp/work/cortexa9hf-neon-imx-gel-linux-gnueabi/chromium-ozone-wayland/121.0.6167.184-r0/recipe-sysroot-native -stdlib=libc++ -isystem/home/parallels/gewa-platform/build/tmp/work/cortexa9hf-neon-imx-gel-linux-gnueabi/chromium-ozone-wayland/121.0.6167.184-r0/recipe-sysroot-native/usr/include -isysroot=/home/parallels/gewa-platform/build/tmp/work/cortexa9hf-neon-imx-gel-linux-gnueabi/chromium-ozone-wayland/121.0.6167.184-r0/recipe-sysroot-native -stdlib=libc++ -O2 -pipe -c ../../third_party/flatbuffers/src/src/binary_annotator.cpp -o yocto_native/obj/third_party/flatbuffers/compiler_files/binary_annotator.o
| [18014/63163] python3 ../../ui/webui/resources/tools/bundle_js.py --host new-tab-page --input gen/chrome/browser/resources/new_tab_page/tsc --out_folder gen/chrome/browser/resources/new_tab_page/bundled --depfile gen/chrome/browser/resources/new_tab_page/build_bundle.d --target_name build_bundle --exclude chrome://resources/mojo/mojo/public/js/bindings.js //resources/mojo/mojo/public/js/bindings.js chrome://resources/mojo/mojo/public/mojom/base/time.mojom-lite.js //resources/mojo/mojo/public/mojom/base/time.mojom-lite.js chrome://resources/polymer/v3_0/polymer/polymer_bundled.min.js //resources/polymer/v3_0/polymer/polymer_bundled.min.js chrome://resources/js/cr.js //resources/js/cr.js chrome://resources/js/load_time_data.js //resources/js/load_time_data.js chrome://resources/ash/common/load_time_data.m.js //resources/ash/common/load_time_data.m.js chrome://resources/mwc/lit/index.js //resources/mwc/lit/index.js chrome://resources/cr_components/color_change_listener/color_change_listener.mojom-webui.js chrome://resources/cr_components/customize_themes/customize_themes.mojom-webui.js chrome://resources/cr_components/page_image_service/page_image_service.mojom-webui.js chrome://resources/cr_components/most_visited/most_visited.mojom-webui.js chrome://resources/cr_components/omnibox/omnibox.mojom-webui.js chrome://resources/js/browser_command/browser_command.mojom-webui.js chrome://resources/js/metrics_reporter/metrics_reporter.mojom-webui.js chrome://resources/mojo/mojo/public/mojom/base/string16.mojom-webui.js chrome://resources/mojo/mojo/public/mojom/base/text_direction.mojom-webui.js chrome://resources/mojo/mojo/public/mojom/base/time.mojom-webui.js chrome://resources/mojo/skia/public/mojom/skcolor.mojom-webui.js chrome://resources/mojo/url/mojom/url.mojom-webui.js cart.mojom-webui.js discount.mojom-webui.js chrome_cart.mojom-webui.js drive.mojom-webui.js feed.mojom-webui.js history_cluster_types.mojom-webui.js history_clusters.mojom-webui.js history_clusters_v2.mojom-webui.js history_clusters_layout_type.mojom-webui.js history_types.mojom-webui.js new_tab_page.mojom-webui.js photos.mojom-webui.js recipes.mojom-webui.js tab_resumption.mojom-webui.js --external_paths chrome://resources/polymer/v3_0/\|../../third_party/polymer/v3_0/components-chromium/ //resources/polymer/v3_0/\|../../third_party/polymer/v3_0/components-chromium/ chrome://resources/\|gen/ui/webui/resources/tsc/ //resources/\|gen/ui/webui/resources/tsc/ --js_module_in_files new_tab_page.js lazy_load.js
| FAILED: gen/chrome/browser/resources/new_tab_page/bundled/new_tab_page.rollup.js gen/chrome/browser/resources/new_tab_page/bundled/new_tab_page.rollup.js.map gen/chrome/browser/resources/new_tab_page/bund [... cut ...]
ERROR: Task (/home/parallels/gewa-platform/sources/meta-browser/meta-chromium/recipes-browser/chromium/chromium-ozone-wayland_121.0.6167.184.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 5036 tasks of which 5023 didn't need to be rerun and 1 failed.
NOTE: Writing buildhistory
NOTE: Writing buildhistory took: 2 seconds

Summary: 1 task failed:
  /home/parallels/gewa-platform/sources/meta-browser/meta-chromium/recipes-browser/chromium/chromium-ozone-wayland_121.0.6167.184.bb:do_compile
Summary: There were 2 WARNING messages.
Summary: There was 1 ERROR message, returning a non-zero exit code.
```

MaxIhlenfeldt, 2024-03-21:
That looks like a different issue to me, are you sure this was introduced by #782?

SunboX, 2024-03-21:
Hm, seems so. I'm not sure. How can I check this? We are building using Yocto kirkstone.

```xml
```xml
<?xml version="1.0" encoding="UTF-8"?>
<manifest>
  <remote fetch="git://git.yoctoproject.org" name="yocto"/>
  <remote fetch="git://git.openembedded.org" name="oe"/>
  <remote fetch="ssh://git@github.com" name="github"/>
  <remote fetch="ssh://git@xxx" name="xxx"/>

  <!-- ### Yocto Project release used as default for all included layers and components ### -->
  <default revision="kirkstone"/>

```
```

```
Build Configuration:
BB_VERSION           = "2.0.0"
BUILD_SYS            = "aarch64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-gel-linux-gnueabi"
MACHINE              = "gewa-imx6-r1"
DISTRO               = "gel"
DISTRO_VERSION       = "22.04"
TUNE_FEATURES        = "arm vfp cortexa9 neon thumb callconvention-hard"
TARGET_FPU           = "hard"
```

SunboX, 2024-03-22:
Might #791 break my build? How can I check this?

SunboX, 2024-03-22:
Ok, I changed the revision from `master` to `0fcfadca55fa36c48ef09ff85c82d5bcde39540b` in my `default.xml` and now it builds again.

```xml
`<project remote="github" revision="0fcfadca55fa36c48ef09ff85c82d5bcde39540b" name="OSSystems/meta-browser" path="sources/meta-browser"/>`
```

MaxIhlenfeldt, 2024-03-22:
For kirkstone we unfortunately only support up to version 120, so it's expected that #791 breaks the build. We have a `kirkstone` branch that you can use.

SunboX, 2024-03-22:
Ah, ok. That makes sense. Thanks a lot!

SunboX, 2024-03-24:
@MaxIhlenfeldt If I set the branch to "kirkstone" I'm getting this error message:

```
ERROR: Layer 'chromium-browser-layer' depends on layer 'lts-rust-mixin', but this layer is not enabled in your configuration
```

MaxIhlenfeldt, 2024-03-25:
Right, on kirkstone we depend on [meta-lts-mixins' `rust-1.68` branch](https://git.yoctoproject.org/meta-lts-mixins/about/?h=kirkstone/rust-1.68) to provide a newer version of Rust.

SunboX, 2024-04-04:
if someone comes across this, this works for me:
```xml
<?xml version="1.0" encoding="UTF-8"?>
<manifest>
  <remote fetch="https://git.yoctoproject.org/git/" name="yocto" />
  <remote fetch="ssh://git@github.com" name="github"/>

  <default revision="kirkstone"/>
  
  <project name="meta-lts-mixins" path="sources/meta-lts-mixins/rust" remote="yocto" revision="1a6746a81da4d5bf2e5640fd2fa8f3ea453d40bf" upstream="kirkstone/rust" />

  <project remote="github" revision="kirkstone" name="OSSystems/meta-browser" path="sources/meta-browser"/>
```

MaxIhlenfeldt, 2024-05-13:
I finally have time to look into this. I've managed to reproduce the issue using the following steps:

1. Do the usual setup using `meta-clang` revision `66414c17ae60e489bef9a97031323e92c247fbe1` (version 18.1.4).
2. Run `bitbake -c configure chromium-x11; bitbake -c add_clang_symlink chromium-x11; bitbake -c copy_clang_library chromium-x11`.
3. Update `meta-clang` to revision `e7dceb1c92caf7f21ef1d7b49c85328c30cffd90` (version 18.1.5).
4. Run `bitbake chromium-x11`.

I suspect I know why this is happening, I'm working on a fix by improving how we implement the `add_clang_symlink` and `copy_clang_library` tasks.

closed 2024-05-14 by MaxIhlenfeldt

dsemkowicz-tt, 2024-05-17:
@SunboX I highly suspect that your first issue with build error was caused by #815, because I experienced build failure with the same error message. But this is just for note. you would still fail to build Chromium 121 with Yocto kirkstone with another error.

PR #814 (2024-05-14): chromium: Fix incremental build error after meta-clang update
Fixes #794.

Build and patch changes:
------------------------

We previously created a symlink pointing to the subdirectory of the latest version under $STAGING_LIBDIR_NATIVE/clang, and then copied the libclang_rt.builtins library from $STAGING_LIBDIR/clang to the same directory as the native library. This is necessary because we need to point Chromium to a single clang directory for building both native and target code, and we need its path to be independent of the specific clang version used.

However, the chosen approach leads t [... cut]
commit 68d56d56 (2024-05-14): chromium: Fix incremental build error after meta-clang update (#814)

Fixes #794.

Build and patch changes:
------------------------

We previously created a symlink pointing to the subdirectory of the
latest version under $STAGING_LIBDIR_NATIVE/clang, and then copied the
libclang_rt.builtins library from $STAGING_LIBDIR/clang to the same
directory as the native library. This is necessary because we need to
point Chromium to a single clang directory for building both native and
target code, and we need its path to be independent of the specific
clang version used.

However, the chosen approach [... cut]