OSSystems/meta-browser #798: [yocto-check-layer] Nothing PROVIDES 'bcc' error on meta-chromium layer(yocto compliance error for meta-chromium layer)

rjanani-p, 2024-03-21:
Getting yocto compliance error for meta-chromium layer for poky master branch.
**Steps to reproduce:** 
1. Create project for Poky master
2. source oe-init-build-env
3. yocto-check-layer-wrapper ../../../repos/meta-browser/meta-chromium --dependency ../../../repos/meta-browser ../../../repos/meta-oe.git/ ../../../repos/meta-clang.git/ --no-auto-dependency

**Error log:**

**INFO: Adding layer meta-clang.git
INFO: Adding layer meta-oe
INFO: Getting initial bitbake variables ...
INFO: Getting initial signatures ...
INFO: Generating signatures failed. This might be due to some parse error and/or general layer incompatibilities.
Command: BB_ENV_PASSTHROUGH_ADDITIONS="$BB_ENV_PASSTHROUGH_ADDITIONS BB_SIGNATURE_HANDLER" BB_SIGNATURE_HANDLER="OEBasicHash" bitbake -S lockedsigs world
Output:
Loading cache...done.
Loaded 0 entries from dependency cache.
Parsing recipes...done.
Parsing of 1879 .bb files complete (0 cached, 1879 parsed). 3226 targets, 91 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies
ERROR: Nothing PROVIDES 'bcc' (but /wrkspc/repos/meta-clang.git/dynamic-layers/openembedded-layer/recipes-devtools/bpftrace/bpftrace_0.20.1.bb DEPENDS on or otherwise requires it). Close matches:
  bc
  byacc
ERROR: Required build target 'meta-world-pkgdata' has no buildable providers.
Missing or unbuildable dependency chain was: ['meta-world-pkgdata', 'bpftrace', 'bcc']

Summary: There were 2 ERROR messages, returning a non-zero exit code.

INFO:
INFO: Summary of results:
INFO:
INFO: meta-chromium ... FAIL (Generating world signatures)**

kraj, 2024-03-21:
thats because `bcc` needs recipes from `meta-python` and hence it under meta-python dynamic layer.
so either we need to add meta-python as dependency or perhaps explore moving `bpftrace` under meta-python dynamic layer inside meta-clang

rjanani-p, 2024-03-21:
thanks for the input, I have tried 2 ways to resolve the issue
1. I have added 2 changes in bpftrace_0.20.1.bb file
changes are :
**+inherit features_check
+REQUIRED_DISTRO_FEATURES = "bcc"**
changes are fine ?
2. I have tried which you suggested, moved the bpftrace recipe to meta-clang.git/dynamic-layers/meta-python/recipes-devtools/ 

in 2 ways , bcc issue is resolved but I am getting another issue due to rust. Here I am sharing the log, could you please share your inputs @kraj  @MaxIhlenfeldt 

INFO: Starting to analyze: meta-chromium
INFO: ----------------------------------------------------------------------
INFO: skipped "BSPCheckLayer: Layer meta-chromium isn't BSP one."
INFO: test_layerseries_compat (common.CommonCheckLayer)
INFO:  ... ok
INFO: test_parse (common.CommonCheckLayer)
INFO:  ... ok
INFO: test_patches_upstream_status (common.CommonCheckLayer)
INFO:  ... unexpected success
INFO: test_readme (common.CommonCheckLayer)
INFO:  ... ok
INFO: test_show_environment (common.CommonCheckLayer)
INFO:  ... ok
**INFO: test_signatures (common.CommonCheckLayer)
INFO:  ... FAIL**

Stdout:
Loading cache...done.
Loaded 3228 entries from dependency cache.
INFO: Traceback (most recent call last):
  File "/buildarea/eng1/rpandi/wrkspc/chromium/poky_master/scripts/lib/checklayer/cases/common.py", line 99, in **test_signatures**
    self.fail('Adding layer %s changed signatures.\n%s' % (self.tc.layer['name'], msg))
AssertionError: Adding layer meta-chromium changed signatures.
526 signatures changed, initial differences (first hash before, second after):
   **rust-native:do_install:** 085f25ebfdade0a6ac9dc6d9ba5923ff62cae949b018ddfbdc24c39d169892c5 -> 1ec4d76568cba0c52cfb840c50307e92b3ff5eb48bbde9680857849affce6e86
      bitbake-diffsigs --task rust-native do_install --signature 085f25ebfdade0a6ac9dc6d9ba5923ff62cae949b018ddfbdc24c39d169892c5 1ec4d76568cba0c52cfb840c50307e92b3ff5eb48bbde9680857849affce6e86
      NOTE: Starting bitbake server...
      basehash changed from 42c20dcb3c48db45bce595f47acc9479b24dec0190126a0ce5382710605ca39a to dec5e6f90090b1fcf03c51399d0f0e923a829f993ba474ffdc68eb290832357c
      List of dependencies for variable rust_do_install changed from 'frozenset({'CPP', 'base_bindir', 'CPPFLAGS', 'STRINGS', 'CONFIG_SITE', 'STAGING_LIBDIR', 'AR', 'STRIP', 'BUILD_CPPFLAGS', 'STAGING_INCDIR', 'BUILD_LDFLAGS', 'BUILD_FC', 'libexecdir', 'LC_ALL', 'READELF', 'LD', 'RANLIB', 'base_prefix', 'PYTHON', 'PERL_HASH_SEED', 'PSEUDO_DISABLED', 'nonarch_base_libdir', 'systemd_user_unitdir', 'OBJDUMP', 'sysconfdir', 'TZ', 'BUILD_AS', 'BUILD_CC', 'BUILD_LD', 'BUILD_CXXFLAGS', 'rust_runx', 'includedir', 'systemd_unitdir', 'LDFLAGS', 'BUILD_AR', 'systemd_system_unitdir', 'libdir', 'OBJCOPY', 'BUILD_NM', 'BUILD_CFLAGS', 'mandir', 'nonarch_libdir', 'BUILD_STRIP', 'infodir', 'prefix', 'localstatedir', 'RUST_TARGET_PATH', 'PYTHONNOUSERSITE', 'PKG_CONFIG_DIR', 'rustlibdir', 'BUILD_CCLD', 'sbindir', 'MAKE', 'CARGO_HOME', 'CC', 'PKG_CONFIG_PATH', 'FC', 'PYTHON_LIBRARY', 'PKG_CONFIG_LIBDIR', 'BUILD_CXX', 'AS', 'CFLAGS', 'base_libdir', 'lt_cv_sys_lib_dlsearch_path_spec', 'PKG_CONFIG_DISABLE_UNINSTALLED', 'FORCE_CRATE_HASH', 'PYTHONHASHSEED', 'bindir', 'CXX', 'NM', 'exec_prefix', 'PKG_CONFIG_ALLOW_CROSS', 'PYTHON_INCLUDE_DIR', 'BUILD_CPP', 'base_sbindir', 'servicedir', 'CXXFLAGS', 'BUILD_RANLIB', 'CCLD', 'docdir', 'PKG_CONFIG_SYSROOT_DIR', 'datadir', 'sharedstatedir'})' to 'frozenset({'CPP', 'base_bindir', 'CPPFLAGS', 'STRINGS', 'CONFIG_SITE', 'STAGING_LIBDIR', 'AR', 'STRIP', 'BUILD_CPPFLAGS', 'STAGING_INCDIR', 'BUILD_LDFLAGS', 'BUILD_FC', 'libexecdir', 'LC_ALL', 'READELF', 'LD', 'RANLIB', 'base_prefix', 'PYTHON', 'PERL_HASH_SEED', 'PSEUDO_DISABLED', 'nonarch_base_libdir', 'systemd_user_unitdir', 'OBJDUMP', 'sysconfdir', 'TZ', 'BUILD_AS', 'BUILD_CC', 'BUILD_LD', 'BUILD_CXXFLAGS', 'rust_runx', 'includedir', 'systemd_unitdir', 'LDFLAGS', 'BUILD_AR', 'systemd_system_unitdir', 'libdir', 'OBJCOPY', 'BUILD_NM', 'BUILD_CFLAGS', 'mandir', 'nonarch_libdir', 'BUILD_STRIP', 'D', 'infodir', 'prefix', 'localstatedir', 'RUST_TARGET_PATH', 'PYTHONNOUSERSITE', 'PKG_CONFIG_DIR', 'rustlibdir', 'BUILD_CCLD', 'sbindir', 'MAKE', 'CARGO_HOME', 'CC', 'PKG_CONFIG_PATH', 'FC', 'PYTHON_LIBRARY', 'PKG_CONFIG_LIBDIR', 'BUILD_CXX', 'AS', 'CFLAGS', 'base_libdir', 'lt_cv_sys_lib_dlsearch_path_spec', 'PKG_CONFIG_DISABLE_UNINSTALLED', 'FORCE_CRATE_HASH', 'PYTHONHASHSEED', 'bindir', 'CXX', 'NM', 'exec_prefix', 'PKG_CONFIG_ALLOW_CROSS', 'PYTHON_INCLUDE_DIR', 'BUILD_CPP', 'base_sbindir', 'servicedir', 'CXXFLAGS', 'BUILD_RANLIB', 'CCLD', 'docdir', 'PKG_CONFIG_SYSROOT_DIR', 'datadir', 'sharedstatedir'})'
      changed items: frozenset({'D'})
      Variable rust_do_install value changed:
      @@ -1 +1,2 @@
           rust_runx install
      +    install -m 0644 ${WORKDIR}/rust-targets/${RUST_TARGET_SYS}.json ${D}${libdir}/rustlib/${RUST_TARGET_SYS}/target.json

   rust:do_collect_spdx_deps: 7e659b1003095e17ec20b8ea311a7746031b988ac4e8652a554ad19069556935 -> 99ed5eb150ed40e26314c6a0fa87d85d07d3417121d62754984de876f6e78953
      bitbake-diffsigs --task rust do_collect_spdx_deps --signature 7e659b1003095e17ec20b8ea311a7746031b988ac4e8652a554ad19069556935 99ed5eb150ed40e26314c6a0fa87d85d07d3417121d62754984de876f6e78953
      NOTE: Starting bitbake server...
      basehash changed from 5d9d1b6f261f35a6968aa6613f5da31b2f5f53006d9541b742c4a6813ebe54a0 to a37fb7caeb2bee5b25a16dffb496c82d902c52f36f06374c3b4185f53493f880
[... cut ...] do_unpack']
      Number of task dependencies changed
      Dependency on task libstd-rs:do_create_spdx was removed with hash aa0a92dca2d3e7f3c8d3c12c0f8102e5175f961354167dc1d71a99ed8890edff

Stdout:
Loading cache...done.
Loaded 3228 entries from dependency cache.
----------------------------------------------------------------------
INFO: Ran 8 tests in 199.545s
INFO: FAILED
INFO:  (failures=1, skipped=2, unexpected successes=1)
INFO:
INFO: Summary of results:
INFO:
INFO: meta-chromium ... FAIL