OSSystems/meta-browser #805: cross dependency with meta-rust and meta-lts-mixins

davidcodesido, 2024-05-03:
I'm currently building a project that requires azure-iotedge which depends on meta-rust and chromium. The project is based on kirkstone thus it needs meta-lts-mixins to provide rust according to the documentation.

Is there any way I can combine both dependecies? I'm quite new to yocto/rust/cargo etc. Any help is welcome.

kraj, 2024-05-03:
we only support master branch across releases so best option is to try master branch

davidcodesido, 2024-05-03:
I tried master branch but it doesn't support kirkstone. Only scarthgap and nanbield

davidcodesido, 2024-05-03:
I added kirkstone to the layer.conf file but the compilation of rust fails, my bet is that it can't build rust, only rust-native

```
diff --git a/meta-chromium/conf/layer.conf b/meta-chromium/conf/layer.conf
index ce6372d..98eaa66 100644
--- a/meta-chromium/conf/layer.conf
+++ b/meta-chromium/conf/layer.conf
@@ -9,6 +9,6 @@ BBFILE_PATTERN_chromium-browser-layer := "^${LAYERDIR}/"
 BBFILE_PRIORITY_chromium-browser-layer = "7"

 LAYERVERSION_chromium-browser-layer = "1"
-LAYERSERIES_COMPAT_chromium-browser-layer = "nanbield scarthgap"
+LAYERSERIES_COMPAT_chromium-browser-layer = "nanbield scarthgap kirkstone"

 LAYERDEPENDS_chromium-browser-layer = "clang-layer core openembedded-layer"
```

kraj, 2024-05-03:
there is `kirkstone` branch now so you should try that first for meta-chromium, also go through README in meta-chromium for dependency needs.

davidcodesido, 2024-05-03:
That's my point. I change to kirkstone branch in meta-chromium and it depends of lts-rust-mixins. This is what creates the conflict with meta-rust.

davidcodesido, 2024-05-03:
I see you also maintain lts-rust-mixins @kraj Is there a way for this two projects meta-rust and meta-lts-mixins to coexist? The moment I add the layer for meta-lts-mixins the builds of iotedge fail and with meta-rust only the build for chromium fail. Any help will be welcome since I've spent already weeks trying to figure this one out.

petergerten, 2024-10-01:
same problem here - any workaround?

Asadeds, 2024-11-04:
same problem here - any update for this one ?

MaxIhlenfeldt, 2024-11-05:
> The moment I add the layer for meta-lts-mixins the builds of iotedge fail 

What exactly is the error here?

Asadeds, 2024-11-05:
Hi @MaxIhlenfeldt , thanks for your quick reply ,
Actually i am using the distro package of OpenSTlinux , and i used Kirkstone layers, i wanted to install chromium browser on my yocto, i tried to add the meta-browser layers with meta-chromium layers for wayland support, i successfully add and  i added the dependencies like , meta-clang, and lts-rust also, 
but whenever i try to build it fails on chromum-ozone-wayland : i went through t this page but  also failed. "https://github.com/OSSystems/meta-browser/issues/528 " 
 what-ever you need to check i will give you the required logs.

ERROR at //build/config/linux/dri/BUILD.gn:11:20: Script returned non-zero exit code.
  dri_driver_dir = exec_script(pkg_config_script,
                   ^----------
Current dir: /wrkdirs/usr/ports/www/chromium/work/chromium-125.0.6422.76/out/Release/
Command: python3 /wrkdirs/usr/ports/www/chromium/work/chromium-125.0.6422.76/build/config/linux/pkg-config.py --dridriverdir dri
Returned 1 and printed out:

Error from pkg-config.

See //content/gpu/BUILD.gn:132:18: which caused the file to be included.
    configs += [ "//build/config/linux/dri" ]

MaxIhlenfeldt, 2024-11-05:
Thanks for the info! That error is specific to #528, so let's keep the discussion there. Unfortunately I can't really help with that issue.

MaxIhlenfeldt, 2024-11-21:
Note that as described in https://github.com/OSSystems/meta-browser/pull/852#issuecomment-2490746158 this will soon be an issue for Scarthgap as well.

It would still be useful if someone could paste the error message of the cross dependency with meta-rust and meta-lts-mixins.

EdoardoMolinatti, 2024-12-03:
Hi all, same problem here.
I will paste the error that will popup using Yocto **kirkstone**, when adding the "**meta-browser**" layer:

`ERROR: Layer 'chromium-browser-layer' depends on layer 'lts-rust-mixin', but this layer is not enabled in your configuration`

MaxIhlenfeldt, 2024-12-03:
> `ERROR: Layer 'chromium-browser-layer' depends on layer 'lts-rust-mixin', but this layer is not enabled in your configuration`

Do you have meta-lts-mixins checked out and added to your `BBLAYERS`?

MaxIhlenfeldt, 2024-12-04:
> Could you let me know if you're aware of this new dependency for "kirkstone"?

Yes, we first enabled Rust in #782, and made the dependency explicit in #800 following the discussion in #789.

> I think the repository URL is https://git.yoctoproject.org/git/meta-lts-mixins, right?

Yes, that's the correct URL.

EdoardoMolinatti, 2024-12-04:
FYI I still have the exact same problem, even after removing completely the previous build in the "build/**tmp**/" subfolder.
Of course I added the new layer (I checked out the branch `kirkstone/rust`) and added it to the `BBLAYERS` in the main recipe.
But I get the same error, so bitbake doesn't even start.

---
```
ERROR: Layer 'chromium-browser-layer' depends on layer 'lts-rust-mixin', but this layer is not enabled in your configuration

Summary: There was 1 ERROR message, returning a non-zero exit code.
```