OSSystems/meta-browser #811: chromium-x11-122.0.6261.128 build failed in yocto for raspberry pi5

gitgonewithwind, 2024-05-10:
Hi all,
sorry to bother you all,
I am building raspberry pi5 image with meta-raspberrypi commit 1879cb831f4ea3e532cb5ce9fa0f32be917e8fa3 (HEAD -> master, origin/master, origin/HEAD)
Author: Max Stepanov <mast@imast.ru>
Date:   Thu Mar 21 13:50:55 2024 -0400

in docker container ubuntu 22.04

poky  commit f4885e97a6b631e0b9b4ec9f17443cb92794866e (HEAD -> master, origin/master, origin/HEAD)
Author: Richard Purdie <richard.purdie@linuxfoundation.org>
Date:   Wed May 8 14:52:36 2024 +0100

meta-openembedded commit 5778e32eae201072c5dc37c9db67dc1848ffb9de (HEAD -> master, origin/master, origin/HEAD)
Author: Martin Jansa <martin.jansa@gmail.com>
Date:   Mon May 6 01:59:47 2024 +0200

..
meta-browser  commit ab764118ea0d9fb4a0f01fe69190e47244708018 (HEAD -> master, origin/master, origin/HEAD)
Author: Max Ihlenfeldt <max@igalia.com>
Date:   Wed May 8 11:03:07 2024 +0200

but chromium-x11-122.0.6261.128 build failed!

what's the issue?
yesterday, I build  on docker container ubuntu18.04, the same fatal error output.

Thank you!

build output from the first fatal error: 

| ../../chrome/browser/file_select_helper.h:16:10: fatal error: 'components/enterprise/buildflags/buildflags.h' file not found
|    16 | #include "components/enterprise/buildflags/buildflags.h"

....

| In file included from ../../chrome/browser/devtools/devtools_window.cc:28:
| ../../chrome/browser/file_select_helper.h:16:10: fatal error: 'components/enterprise/buildflags/buildflags.h' file not found
|    16 | #include "components/enterprise/buildflags/buildflags.h"
|       |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
| 1 error generated.

....

I/home/yocto/yocto/source/build/tmp/work/cortexa76-poky-linux/chromium-x11/122.0.6261.128/recipe-sysroot/usr/include/nss3 -I/home/yocto/yocto/source/build/tmp/work/cortexa76-poky-linux/chromium-x11/122.0.6261.128/recipe-sysroot/usr/include/nspr -I/home/yocto/yocto/source/build/tmp/work/cortexa76-poky-linux/chromium-x11/122.0.6261.128/recipe-sysroot/usr/include/dbus-1.0 -I/home/yocto/yocto/source/build/tmp/work/cortexa76-poky-linux/chromium-x11/122.0.6261.128/recipe-sysroot/usr/lib/dbus-1.0/include -I/home/yocto/yocto/source/build/tmp/work/cortexa76-poky-linux/chromium-x11/122.0.6261.128/recipe-sysroot/usr/include/at-spi2-atk/2.0 -I/home/yocto/yocto/source/build/tmp/work/cortexa76-poky-linux/chromium-x11/122.0.6261.128/recipe-sysroot/usr/include/at-spi-2.0 -I/home/yocto/yocto/source/build/tmp/work/cortexa76-poky-linux/chromium-x11/122.0.6261.128/recipe-sysroot/usr/include/libmount -I/home/yocto/yocto/source/build/tmp/work/cortexa76-poky-linux/chromium-x11/122.0.6261.128/recipe-sysroot/usr/include/blkid -I/home/yocto/yocto/source/build/tmp/work/cortexa76-poky-linux/chromium-x11/122.0.6261.128/recipe-sysroot/usr/include/atk-1.0 -I/home/yocto/yocto/source/build/tmp/work/cortexa76-poky-linux/chromium-x11/122.0.6261.128/recipe-sysroot/usr/include/dbus-1.0 -I/home/yocto/yocto/source/build/tmp/work/cortexa76-poky-linux/chromium-x11/122.0.6261.128/recipe-sysroot/usr/lib/dbus-1.0/include -I/home/yocto/yocto/source/build/tmp/work/cortexa76-poky-linux/chromium-x11/122.0.6261.128/recipe-sysroot/usr/include/glib-2.0 -I/home/yocto/yocto/source/build/tmp/work/cortexa76-poky-linux/chromium-x11/122.0.6261.128/recipe-sysroot/usr/lib/glib-2.0/include -Wno-c++11-narrowing-const-reference -std=c++20 -Wno-trigraphs -gsimple-template-names -fno-exceptions -fno-rtti -fvisibility-inlines-hidden  -O2 -pipe  -feliminate-unused-debug-types   -fmacro-prefix-map=/home/yocto/yocto/source/build/tmp/work/cortexa76-poky-linux/chromium-x11/122.0.6261.128/chromium-122.0.6261.128=/usr/src/debug/chromium-x11/122.0.6261.128  -fdebug-prefix-map=/home/yocto/yocto/source/build/tmp/work/cortexa76-poky-linux/chromium-x11/122.0.6261.128/chromium-122.0.6261.128=/usr/src/debug/chromium-x11/122.0.6261.128  -fmacro-prefix-map=/home/yocto/yocto/source/build/tmp/work/cortexa76-poky-linux/chromium-x11/122.0.6261.128/chromium-122.0.6261.128/out/Release=/usr/src/debug/chromium-x11/122.0.6261.128  -fdebug-prefix-map=/home/yocto/yocto/source/build/tmp/work/cortexa76-poky-linux/chromium-x11/122.0.6261.128/chromium-122.0.6261.128/out/Release=/usr/src/debug/chromium-x11/122.0.6261.128  -fdebug-prefix-map=/home/yocto/yocto/source/build/tmp/work/cortexa76-poky-linux/chromium-x11/122.0.6261.128/recipe-sysroot=  -fmacro-prefix-map=/home/yocto/yocto/source/build/tmp/work/cortexa76-poky-linux/chromium-x11/122.0.6261.128/recipe-sysroot=  -fdebug-prefix-map=/home/yocto/yocto/source/build/tmp/work/cortexa76-poky-linux/chromium-x11/122.0.6261.128/recipe-sysroot-native= -g1 -fvisibility-inlines-hidden -stdlib=libc++ -c ../../chrome/browser/media/router/providers/cast/mirroring_activity.cc -o obj/chrome/browser/media/router/router/mirroring_activity.o
| ninja: build stopped: subcommand failed.
| WARNING: /home/yocto/yocto/source/build/tmp/work/cortexa76-poky-linux/chromium-x11/122.0.6261.128/temp/run.do_compile.392:144 exit 1 from 'ninja -v -j 15 chrome chrome_sandbox chromedriver.unstripped'
| WARNING: Backtrace (BB generated script):
| 	#1: do_compile, /home/yocto/yocto/source/build/tmp/work/cortexa76-poky-linux/chromium-x11/122.0.6261.128/temp/run.do_compile.392, line 144
| 	#2: main, /home/yocto/yocto/source/build/tmp/work/cortexa76-poky-linux/chromium-x11/122.0.6261.128/temp/run.do_compile.392, line 148
NOTE: recipe chromium-x11-122.0.6261.128-r0: task do_compile: Failed
ERROR: Task (/home/yocto/yocto/source/layers/meta-browser/meta-chromium/recipes-browser/chromium/chromium-x11_122.0.6261.128.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 9712 tasks of which 9705 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/yocto/yocto/source/layers/meta-browser/meta-chromium/recipes-browser/chromium/chromium-x11_122.0.6261.128.bb:do_compile
Summary: There was 1 ERROR message, returning a non-zero exit code.
Build step 'Execute shell' marked build as failure

MaxIhlenfeldt, 2024-05-10:
This is an upstream problem I've also run into a few times. It's a race condition in the build due to some components not declaring a dependency, meaning that they can sometimes be compiled before the needed header is generated. I've filed https://issues.chromium.org/issues/339401874 but haven't heard back yet, in the meantime you can fix this issue using the following patch (for m122, I hope to be able to include a backport in #806):

```diff
diff --git a/meta-chromium/recipes-browser/chromium/chromium-gn.inc b/meta-chromium/recipes-browser/chromium/chromium-gn.inc
index aeb2e084..bea3ae61 100644
--- a/meta-chromium/recipes-browser/chromium/chromium-gn.inc
+++ b/meta-chromium/recipes-browser/chromium/chromium-gn.inc
@@ -26,6 +26,7 @@ SRC_URI += "\
     file://0008-Use-the-correct-path-to-libclang_rt.builtins.a.patch \
     file://0009-Adjust-the-Rust-build-to-our-needs.patch \
     file://0010-Don-t-require-profiler_builtins.rlib.patch \
+    file://0011-Add-missing-dependencies-on-chrome-browser-browser.patch \
 "
 # ARM/AArch64-specific patches.
 SRC_URI:append:arm = "\
diff --git a/meta-chromium/recipes-browser/chromium/files/0011-Add-missing-dependencies-on-chrome-browser-browser.patch b/meta-chromium/recipes-browser/chromium/files/0011-Add-missing-dependencies-on-chrome-browser-browser.patch
new file mode 100644
index 00000000..323fd6aa
--- /dev/null
+++ b/meta-chromium/recipes-browser/chromium/files/0011-Add-missing-dependencies-on-chrome-browser-browser.patch
@@ -0,0 +1,47 @@
+From a1c6672d515b7b4e6eee9c76a823f08058ba72de Mon Sep 17 00:00:00 2001
+From: Max Ihlenfeldt <max@igalia.com>
+Date: Thu, 9 May 2024 15:59:45 +0000
+Subject: [PATCH] Add missing dependencies on //chrome/browser:browser
+
+---
+ chrome/browser/devtools/BUILD.gn      | 1 +
+ chrome/browser/extensions/BUILD.gn    | 1 +
+ chrome/browser/safe_browsing/BUILD.gn | 1 +
+ 3 files changed, 3 insertions(+)
+
+diff --git a/chrome/browser/devtools/BUILD.gn b/chrome/browser/devtools/BUILD.gn
+index 159ee0909d33f..2bf53b180b81a 100644
+--- a/chrome/browser/devtools/BUILD.gn
++++ b/chrome/browser/devtools/BUILD.gn
+@@ -129,6 +129,7 @@ static_library("devtools") {
+       "//chrome:resources",
+       "//chrome:strings",
+       "//chrome/app/theme:theme_resources",
++      "//components/enterprise/buildflags",
+       "//chrome/browser:browser_process",
+       "//chrome/browser/media/router:media_router_feature",
+       "//chrome/browser/profiles:profile",
+diff --git a/chrome/browser/extensions/BUILD.gn b/chrome/browser/extensions/BUILD.gn
+index 0ca499538f500..796845826fb74 100644
+--- a/chrome/browser/extensions/BUILD.gn
++++ b/chrome/browser/extensions/BUILD.gn
+@@ -790,6 +790,7 @@ static_library("extensions") {
+     "//chrome/app/theme:chrome_unscaled_resources",
+     "//chrome/app/theme:theme_resources",
+     "//chrome/app/vector_icons",
++    "//components/enterprise/buildflags",
+     "//chrome/browser:browser_process",
+     "//chrome/browser:ntp_background_proto",
+     "//chrome/browser:theme_properties",
+diff --git a/chrome/browser/safe_browsing/BUILD.gn b/chrome/browser/safe_browsing/BUILD.gn
+index 4a67fbf1512e0..a46ac21af63d3 100644
+--- a/chrome/browser/safe_browsing/BUILD.gn
++++ b/chrome/browser/safe_browsing/BUILD.gn
+@@ -23,6 +23,7 @@ static_library("safe_browsing") {
+   deps = [
+     "//build:branding_buildflags",
+     "//chrome/app:generated_resources",
++    "//components/enterprise/buildflags",
+     "//chrome/browser:browser_process",
+     "//chrome/browser/google",
+     "//chrome/browser/profiles",
```

gitgonewithwind, 2024-05-10:
thank you so much.
after your merging in the master branch, I will pull it my host to build.

gitgonewithwind, 2024-05-12:
the patch works well. thanks close this issue

PR #812 (2024-05-13): chromium: Fix build race condition
Fixes #811.

Build and patch changes:
------------------------

Add one backported patch to fix a build race condition that would sometimes lead to build errors.

License changes:
----------------

Added licenses: none.

Removed licenses: none.

Updated licenses: none.

Test-built:
-----------

* chromium-wayland:
 - nanbield, clang, MACHINE=qemuarm64

* chromium-x11:
 - master, clang,   MACHINE=qemuarm
commit 24b4cdf0 (2024-05-13): chromium: Fix build race condition (#812)

Fixes #811.

Build and patch changes:
------------------------

Add one backported patch to fix a build race condition that would
sometimes lead to build errors.

License changes:
----------------

Added licenses: none.

Removed licenses: none.

Updated licenses: none.

Test-built:
-----------

* chromium-wayland:
 - nanbield, clang, MACHINE=qemuarm64

* chromium-x11:
 - master, clang,   MACHINE=qemuarm
closed 2024-05-12 by gitgonewithwind