OSSystems/meta-browser #813: chromium-ozone-wayland-122.0.6261.128 build failed in yocto for master

Gerry546, 2024-05-11:
Hi,

I ran into a problem building the mentioned version of chromium. I looks like the same item as #811 but in this case for the wayland browser.

```bitbake
FAILED: obj/chrome/browser/safe_browsing/safe_browsing/chrome_enterprise_url_lookup_service_factory.o 
aarch64-oe-linux-clang++ -target aarch64-oe-linux  -mcpu=cortex-a72+crc+nocrypto -mbranch-protection=standard -mlittle-endian --dyld-prefix=/usr -Qunused-arguments -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/workspaces/meta-sanctum/build/tmp/work/cortexa72-oe-linux/chromium-ozone-wayland/122.0.6261.128/recipe-sysroot -MMD -MF obj/chrome/browser/safe_browsing/safe_browsing/chrome_enterprise_url_lookup_service_factory.o.d -DUSE_UDEV -DUSE_AURA=1 -DUSE_GLIB=1 -DUSE_OZONE=1 -DOFFICIAL_BUILD -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D_FORTIFY_SOURCE=2 -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -DNO_UNWIND_TABLES -D_GNU_SOURCE -D_LIBCPP_HARDENING_MODE=_LIBCPP_HARDENING_MODE_NONE -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -DGLIB_VERSION_MAX_ALLOWED=GLIB_VERSION_2_56 -DGLIB_VERSION_MIN_REQUIRED=GLIB_VERSION_2_56 -DBASE_USE_PERFETTO_CLIENT_LIBRARY=1 -DSK_ENABLE_SKSL -DSK_UNTIL_CRBUG_1187654_IS_FIXED -DSK_USER_CONFIG_HEADER=\"../../skia/config/SkUserConfig.h\" -DSK_WIN_FONTMGR_NO_SIMULATIONS -DSK_DISABLE_LEGACY_SKSURFACE_METHODS -DSK_DISABLE_LEGACY_GRAPHITE_IMAGE_FACTORIES -DSK_DISABLE_LEGACY_GRAPHITE_IMAGE_METHODS -DSK_DISABLE_LEGACY_SKSURFACE_FACTORIES -DSK_DISABLE_LEGACY_SKSURFACE_FLUSH -DSK_DISABLE_LEGACY_SKSURFACE_AS_IMAGE -DSK_DISABLE_LEGACY_SKSURFACE_DISPLAYLIST -DSK_DISABLE_LEGACY_IMAGE_SUBSET_METHODS -DSK_DISABLE_LEGACY_IMAGE_COLORSPACE_METHODS -DSK_DISABLE_LEGACY_IMAGE_RELEASE_PROCS -DSK_DISABLE_LEGACY_GL_BACKEND_SURFACE -DSK_DISABLE_LEGACY_INIT_DECODERS -DSK_DISABLE_LEGACY_GRDIRECTCONTEXT_FLUSH -DSK_DISABLE_LEGACY_GRDIRECTCONTEXT_BOOLS -DSK_DISABLE_LEGACY_GL_GRDIRECTCONTEXT_FACTORIES -DSK_DISABLE_LEGACY_PNG_WRITEBUFFER -DSK_SLUG_DISABLE_LEGACY_DESERIALIZE -DSK_DISABLE_LEGACY_VK_GRDIRECTCONTEXT_FACTORIES -DSK_DEFAULT_TYPEFACE_IS_EMPTY -DSK_DISABLE_LEGACY_DEFAULT_TYPEFACE -DSK_DISABLE_LEGACY_VULKAN_BACKENDSEMAPHORE -DSK_DISABLE_LEGACY_CREATE_CHARACTERIZATION -DSK_DISABLE_LEGACY_FONTMGR_REFDEFAULT -DSK_DISABLE_LEGACY_FONTMGR_FACTORY -DSK_CODEC_DECODES_JPEG -DSK_ENCODE_JPEG -DSK_ENCODE_PNG -DSK_ENCODE_WEBP -DSK_GANESH -DSK_GPU_WORKAROUNDS_HEADER=\"gpu/config/gpu_driver_bug_workaround_autogen.h\" -DSK_GL -DSK_VULKAN=1 -DSK_GRAPHITE -DSK_DAWN -DVK_USE_PLATFORM_WAYLAND_KHR -DWEBP_EXTERN=extern -DUSE_EGL -DTOOLKIT_VIEWS=1 -DON_FOCUS_PING_ENABLED -DOPENSSL_NO_ASM -DLEVELDB_PLATFORM_CHROMIUM=1 -DU_USING_ICU_NAMESPACE=0 -DU_ENABLE_DYLOAD=0 -DUSE_CHROMIUM_ICU=1 -DU_ENABLE_TRACING=1 -DU_ENABLE_RESOURCE_TRACING=0 -DU_STATIC_IMPLEMENTATION -DICU_UTIL_DATA_IMPL=ICU_UTIL_DATA_FILE -DGOOGLE_PROTOBUF_NO_RTTI -DGOOGLE_PROTOBUF_NO_STATIC_INITIALIZER -DGOOGLE_PROTOBUF_INTERNAL_DONATE_STEAL_INLINE=0 -DHAVE_PTHREAD -DV8_USE_EXTERNAL_STARTUP_DATA -DWEBRTC_ENABLE_AVX2 -DWEBRTC_NON_STATIC_TRACE_EVENT_HANDLERS=0 -DWEBRTC_CHROMIUM_BUILD -DWEBRTC_POSIX -DWEBRTC_LINUX -DABSL_ALLOCATOR_NOTHROW=1 -DLOGGING_INSIDE_WEBRTC -DCRASHPAD_ZLIB_SOURCE_EXTERNAL -DV8_COMPRESS_POINTERS -DV8_COMPRESS_POINTERS_IN_SHARED_CAGE -DV8_31BIT_SMIS_ON_64BIT_ARCH -DV8_ENABLE_SANDBOX -DV8_DEPRECATION_WARNINGS -DV8_USE_PERFETTO -DCPPGC_CAGED_HEAP -DCPPGC_YOUNG_GENERATION -DCPPGC_POINTER_COMPRESSION -DCPPGC_SLIM_WRITE_BARRIER -DATK_LIB_DIR=\"/usr/lib\" -DUSE_ATK_BRIDGE -DOPENSCREEN_TEST_DATA_DIR=\"third_party/openscreen/src/test/data/\" -DENABLE_TRACE_LOGGING -DI18N_ADDRESS_VALIDATION_DATA_URL=\"https://chromium-i18n.appspot.com/ssl-aggregate-address/\" -DEIGEN_MPL2_ONLY -DEIGEN_MAX_ALIGN_BYTES=64 -DEIGEN_HAS_TYPE_TRAITS=0 -DFLATBUFFERS_LOCALE_INDEPENDENT=0 -I../.. -Igen -I../../third_party/perfetto/include -Igen/third_party/perfetto/build_config -Igen/third_party/perfetto -I../../net/third_party/quiche/overrides -I../../net/third_party/quiche/src/quiche/common/platform/default -I../../net/third_party/quiche/src -I../../third_party/skia -Igen/third_party/skia -I../../third_party/wuffs/src/release/c -I../../third_party/vulkan/include -I../../third_party/vulkan-deps/vulkan-headers/src/include -I../../third_party/wayland/src/src -I../../third_party/wayland/include/src -Igen/third_party/dawn/include -I../../third_party/dawn/include -I../../third_party/libwebp/src/src -I../../third_party/libyuv/include -I../../third_party/khronos -I../../gpu -Igen/shim_headers/flac_shim -Igen/third_party/private_membership/src -Igen/third_party/shell-encryption/src -Igen/components/policy/proto -I../../base/allocator/partition_allocator/src -Igen/base/allocator/partition_allocator/src -I../../third_party/abseil-cpp -I../../third_party/boringssl/src/include -I../../third_party/protobuf/src -Igen/protoc_out -I../../third_party/leveldatabase -I../../third_party/leveldatabase/src -I../../third_party/leveldatabase/src/include -I../../third_party/ipcz/include -I../../third_party/ced/src -I../../third_party/icu/source/common -I../../third_party/icu/source/i18n -Igen/net/third_party/quiche/src -I../../third_party/webrtc_overrides -I../../third_party/webrtc -Igen/third_party/webrtc -I../../third_party/libwebm/source -I../../third_party/mesa_headers -I../../third_party/crashpad/crashpad -I../../third_party/crashpad/crashpad/compat/linux -I../../third_party/crashpad/crashpad/compat/non_win -I../../third_party/zlib -I../../third_party/libaom/source/libaom -I../../third_party/libaom/source/config/linux/arm64-cpu-detect -I../../v8/include -Igen/third_party/metrics_proto -I../../third_party/re2/src -I../../third_party/openscreen/src -Igen/third_party/openscreen/src -I../../third_party/jsoncpp/source/include -I../../third_party/libaddressinput/src/cpp/include -I../../third_party/tflite/src -I../../third_party/tflite/src/third_party/xla/third_party/tsl -I../../third_party/pthreadpool/src/include -I../../third_party/eigen3/src -Igen/third_party/tflite/src -I../../third_party/flatbuffers/src/include -I../../third_party/flatbuffers/src/src -I../../third_party/flatbuffers/src -I../../third_party/flatbuffers/generated -I../../third_party/gemmlowp/src -I../../third_party/tflite_support/src -I../../third_party/tflite_support/src/tensorflow_lite_support -Igen/third_party/tflite_support/src -Wall -Wextra -Wimplicit-fallthrough -Wextra-semi -Wunreachable-code-aggressive -Wthread-safety -Wno-missing-field-initializers -Wno-unused-parameter -Wno-psabi -Wloop-analysis -Wno-unneeded-internal-declaration -Wenum-compare-conditional -Wno-unknown-warning-option -Wno-ignored-pragma-optimize -Wno-deprecated-builtins -Wno-bitfield-constant-conversion -Wno-deprecated-this-capture -Wno-invalid-offsetof -Wno-vla-extension -Wno-thread-safety-reference-return -Wshadow -fno-delete-null-pointer-checks -fno-ident -fno-strict-aliasing -fstack-protector -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -pthread -fcolor-diagnostics -fmerge-all-constants -fcrash-diagnostics-dir=../../tools/clang/crashreports -mllvm -instcombine-lower-dbg-declare=0 -ffp-contract=off -flto=thin -fsplit-lto-unit -fwhole-program-vtables -mbranch-protection=standard -mno-outline -ffile-compilation-dir=. -ftrivial-auto-var-init=pattern -O2 -fdata-sections -ffunction-sections -fno-unique-section-names -fno-math-errno -fno-omit-frame-pointer -g0 -fvisibility=hidden -Wheader-hygiene -Wstring-conversion -Wtautological-overlap-compare -Wexit-time-destructors -I/workspaces/meta-sanctum/build/tmp/work/cortexa72-oe-linux/chromium-ozone-wayland/122.0.6261.128/recipe-sysroot/usr/include/glib-2.0 -I/workspaces/meta-sanctum/build/tmp/work/cortexa72-oe-linux/chromium-ozone-wayland/122.0.6261.128/recipe-sysroot/usr/lib/glib-2.0/include -Wno-redundant-parens -Wno-redundant-parens -DPROTOBUF_ALLOW_DEPRECATED=1 -I/workspaces/meta-sanctum/build/tmp/work/cortexa72-oe-linux/chromium-ozone-wayland/122.0.6261.128/recipe-sysroot/usr/include/nss3 -I/workspaces/meta-sanctum/build/tmp/work/cortexa72-oe-linux/chromium-ozone-wayland/122.0.6261.128/recipe-sysroot/usr/include/nspr -I/workspaces/meta-sanctum/build/tmp/work/cortexa72-oe-linux/chromium-ozone-wayland/122.0.6261.128/recipe-sysroot/usr/include/dbus-1.0 -I/workspaces/meta-sanctum/build/tmp/work/cortexa72-oe-linux/chromium-ozone-wayland/122.0.6261.128/recipe-sysroot/usr/lib/dbus-1.0/include -I/workspaces/meta-sanctum/build/tmp/work/cortexa72-oe-linux/chromium-ozone-wayland/122.0.6261.128/re [... cut ...]
In file included from ../../chrome/browser/safe_browsing/chrome_enterprise_url_lookup_service_factory.cc:8:
In file included from ../../chrome/browser/enterprise/connectors/connectors_service.h:14:
../../chrome/browser/enterprise/connectors/connectors_manager.h:16:10: fatal error: 'components/enterprise/buildflags/buildflags.h' file not found
   16 | #include "components/enterprise/buildflags/buildflags.h"
``` 

Since I did not want to hijjack the issue I created a new one. Question: is there a similar patch which I can drop in?

gitgonewithwind, 2024-05-12:
I think you can the same patch file https://github.com/OSSystems/meta-browser/issues/811#issuecomment-2104273455
I just create the patch name a.patch.
an run : cat ../a.patch  | patch -p1    
patched it and I have been building. 4 hours later I will see it will be successfull. I think

MaxIhlenfeldt, 2024-05-13:
There are only minor differences between chromium-x11 and chromium-ozone-wayland. #812 (which was just merged) should fix this for the latter as well.

closed 2024-05-13 by MaxIhlenfeldt