OSSystems/meta-browser #820: chromium-x11-122.0.6261.128 build failed with compilation issue

rjanani-p, 2024-05-23:
Hi, 

I am trying to build chromium-x11 package for 122.0.6261.128(wrlinux_master branch) for qemux86-64 , qemuarm64 targets but getting below error,

ERROR: chromium-x11-122.0.6261.128-r0 do_copy_clang_library: ExecutionError('$/tmp-glibc/work/core2-64-wrs-linux/chromium-x11/122.0.6261.128/temp/run.do_copy_clang_library.1558278', 1, None, None)
ERROR: Logfile of failure stored in:$/tmp-glibc/work/core2-64-wrs-linux/chromium-x11/122.0.6261.128/temp/log.do_copy_clang_library.1558278
Log data follows:
| DEBUG: Executing shell function do_copy_clang_library
| $/tmp-glibc/work/core2-64-wrs-linux/chromium-x11/122.0.6261.128/temp/run.do_copy_clang_library.1558278: line 151: cd: $/tmp-glibc/work/core2-64-wrs-linux/chromium-x11/122.0.6261.128/recipe-sysroot/usr/lib64/clang: No such file or directory
| WARNING: $/tmp-glibc/work/core2-64-wrs-linux/chromium-x11/122.0.6261.128/temp/run.do_copy_clang_library.1558278:151 exit 1 from 'cd "$/tmp-glibc/work/core2-64-wrs-linux/chromium-x11/122.0.6261.128/recipe-sysroot/usr/lib64/clang"'
| WARNING: Backtrace (BB generated script):
|       #1: do_copy_clang_library, $/tmp-glibc/work/core2-64-wrs-linux/chromium-x11/122.0.6261.128/temp/run.do_copy_clang_library.1558278, line 151
|       #2: main, $/tmp-glibc/work/core2-64-wrs-linux/chromium-x11/122.0.6261.128/temp/run.do_copy_clang_library.1558278, line 159
ERROR: Task (/buildarea/eng1/rpandi/wrkspc/ltsrealses/lincd_cland_build_issue/layers/meta-browser/meta-chromium/recipes-browser/chromium/chromium-x11_122.0.6261.128.bb:do_copy_clang_library) failed with exit code '1'
NOTE: Tasks Summary: Attempted 4022 tasks of which 4021 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /buildarea/eng1/rpandi/wrkspc/ltsrealses/lincd_cland_build_issue/layers/meta-browser/meta-chromium/recipes-browser/chromium/chromium-x11_122.0.6261.128.bb:do_copy_clang_library
Summary: There was 1 WARNING message.
Summary: There was 1 ERROR message, returning a non-zero exit code.

while building the chromium-x11 for qemuarm not getting issue. I have tried to reproduce the issue with poky master+meta-browser master, but I am facing issue while adding meta-openembedded dependent package to the chromium build. 

Error log for reference:
ERROR: $/meta-openembedded/meta-oe/recipes-graphics/ttf-fonts/ttf-roboto_2.138.bb: Using S = ${WORKDIR} is no longer supported
ERROR: $/meta-openembedded/meta-gnome/recipes-gnome/gtksourceview/gtksourceview-classic-light_1.0.bb: Using S = ${WORKDIR} is no longer supported

could you please check clang build issue log once and please share your inputs. Thank you.

MaxIhlenfeldt, 2024-05-27:
A you building a revision of meta-browser that includes https://github.com/OSSystems/meta-browser/pull/814?

```
/tmp-glibc/work/core2-64-wrs-linux/chromium-x11/122.0.6261.128/recipe-sysroot/usr/lib64/clang: No such file or directory
```

It's interesting that it's trying to use `/usr/lib64/clang`, I think I've always seen `/usr/lib/clang` being used before. Can you please check if the latter exists (i.e. `tmp-glibc/work/core2-64-wrs-linux/chromium-x11/122.0.6261.128/recipe-sysroot/usr/lib/clang`) and if yes, paste the output of `tree tmp-glibc/work/core2-64-wrs-linux/chromium-x11/122.0.6261.128/recipe-sysroot/usr/lib/clang` here?

Also, can you please paste the contents of `tmp-glibc/work/core2-64-wrs-linux/chromium-x11/122.0.6261.128/temp/run.do_copy_clang_library.1558278` here?

rjanani-p, 2024-05-28:
> A you building a revision of meta-browser that includes #814?
>
> ```
> /tmp-glibc/work/core2-64-wrs-linux/chromium-x11/122.0.6261.128/recipe-sysroot/usr/lib64/clang: No such file or directory
> ```
>
> It's [...]

yes, I am building chromium version 122.0.6261.128 with meta-browser commits up to #814. 

Here I am sharing the output for following command
$ tmp-glibc/work/core2-64-wrs-linux/chromium-x11/122.0.6261.128/recipe-sysroot/usr/lib/clang
$ls
**18.1.5**

and from the log, I have understood my chromium build clang library path is taking /usr/lib64/clang and for poky master build clang lib path it's taking /usr/lib/clang. 
and for my chroimum build few paths it's taking /usr/lib64/.

export STAGING_LIBDIR="$build_lincd_clang_build_issue/tmp-glibc/work/core2-64-wrs-linux/chromium-x11/122.0.6261.128/recipe-sysroot/usr/lib64"

export base_libdir="/usr/lib64"
export libdir="/usr/lib64"

here I am attaching the log file for reference.
[run.do_copy_clang_library.1558278.txt](https://github.com/OSSystems/meta-browser/files/15464284/run.do_copy_clang_library.1558278.txt)

could you please share your input.

rjanani-p, 2024-05-30:
> @kraj could this be a meta-clang issue?

@kraj , could you please check the issue once. I have tried with chromium 123.0.6312.122 version but for that also getting same issue.

kraj, 2024-05-30:
I dont see it being clang issue here, My local packages built with clang finds it in /usr/lib and on architectures where lib64 is default it looks into /use/lib64 correctly.

rjanani-p, 2024-05-30:
@kraj thanks for sharing the information.

@MaxIhlenfeldt , could you please share your input whether I need to update any changes in my chromium build.

rjanani-p, 2024-06-13:
> > @MaxIhlenfeldt , could you please share your input whether I need to update any changes in my chromium build.
> 
> @rjanani-p It seems to me both meta-chromium and meta-clang are doing the right thing here, so I'm out of ideas, sorry.

thanks for the reply @MaxIhlenfeldt , if you have any inputs to resolve the issue, please share. Thank you.