OSSystems/meta-browser #829: chromium: Enable crabbyavif

MaxIhlenfeldt, 2024-08-12:
In #828, we had to disable crabbyavif from being built, because the bindgen invocations for it created build errors (see https://github.com/OSSystems/meta-browser/issues/827#issuecomment-2263051027).

I've been able to fix the `unsupported option '-mbranch-protection=' for target 'x86_64-unknown-linux-gnu'` part, but I'm stuck on the `$HEADER file not found` part.

My current solution is to get all the flags set in our toolchain BUILD.gn file and pass them through to bindgen, so it can pass them to clang. These two are new in our `args.gn`:
```gn
rust_bindgen_extra_host_args = [
  "-isystem/home/gitlab-runner/yocto-chromium/builds/ci-chromium-x11-qemuarm-master/tmp-glibc/work/cortexa15t2hf-neon-oe-linux-gnueabi/chromium-x11/127.0.6533.99/recipe-sysroot-native/usr/include",
  "-isysroot=/home/gitlab-runner/yocto-chromium/builds/ci-chromium-x11-qemuarm-master/tmp-glibc/work/cortexa15t2hf-neon-oe-linux-gnueabi/chromium-x11/127.0.6533.99/recipe-sysroot-native",
  "-stdlib=libc++",
  "-O2",
  "-pipe",
]
rust_bindgen_extra_target_args = [
  "-target",
  "arm-oe-linux-gnueabi",
  "-mfpu=neon",
  "-mfloat-abi=hard",
  "-mcpu=cortex-a15",
  "-mlittle-endian",
  "-Qunused-arguments",
  "-D_TIME_BITS=64",
  "-D_FILE_OFFSET_BITS=64",
  "--sysroot=/home/gitlab-runner/yocto-chromium/builds/ci-chromium-x11-qemuarm-master/tmp-glibc/work/cortexa15t2hf-neon-oe-linux-gnueabi/chromium-x11/127.0.6533.99/recipe-sysroot",
  "-O2",
  "-g",
  "-fmacro-prefix-map=/home/gitlab-runner/yocto-chromium/builds/ci-chromium-x11-qemuarm-master/tmp-glibc/work/cortexa15t2hf-neon-oe-linux-gnueabi/chromium-x11/127.0.6533.99/chromium-127.0.6533.99=/usr/src/debug/chromium-x11/127.0.6533.99",
  "-fdebug-prefix-map=/home/gitlab-runner/yocto-chromium/builds/ci-chromium-x11-qemuarm-master/tmp-glibc/work/cortexa15t2hf-neon-oe-linux-gnueabi/chromium-x11/127.0.6533.99/chromium-127.0.6533.99=/usr/src/debug/chromium-x11/127.0.6533.99",
  "-fmacro-prefix-map=/home/gitlab-runner/yocto-chromium/builds/ci-chromium-x11-qemuarm-master/tmp-glibc/work/cortexa15t2hf-neon-oe-linux-gnueabi/chromium-x11/127.0.6533.99/chromium-127.0.6533.99/out/Release=/usr/src/debug/chromium-x11/127.0.6533.99",
  "-fdebug-prefix-map=/home/gitlab-runner/yocto-chromium/builds/ci-chromium-x11-qemuarm-master/tmp-glibc/work/cortexa15t2hf-neon-oe-linux-gnueabi/chromium-x11/127.0.6533.99/chromium-127.0.6533.99/out/Release=/usr/src/debug/chromium-x11/127.0.6533.99",
  "-fdebug-prefix-map=/home/gitlab-runner/yocto-chromium/builds/ci-chromium-x11-qemuarm-master/tmp-glibc/work/cortexa15t2hf-neon-oe-linux-gnueabi/chromium-x11/127.0.6533.99/recipe-sysroot=",
  "-fmacro-prefix-map=/home/gitlab-runner/yocto-chromium/builds/ci-chromium-x11-qemuarm-master/tmp-glibc/work/cortexa15t2hf-neon-oe-linux-gnueabi/chromium-x11/127.0.6533.99/recipe-sysroot=",
  "-fdebug-prefix-map=/home/gitlab-runner/yocto-chromium/builds/ci-chromium-x11-qemuarm-master/tmp-glibc/work/cortexa15t2hf-neon-oe-linux-gnueabi/chromium-x11/127.0.6533.99/recipe-sysroot-native=",
  "-pipe",
]
```

I've also manually added these two to the extra target args:
```
  "-isysroot=/home/gitlab-runner/yocto-chromium/builds/ci-chromium-x11-qemuarm-master/tmp-glibc/work/cortexa15t2hf-neon-oe-linux-gnueabi/chromium-x11/127.0.6533.99/recipe-sysroot",
  "-isystem/home/gitlab-runner/yocto-chromium/builds/ci-chromium-x11-qemuarm-master/tmp-glibc/work/cortexa15t2hf-neon-oe-linux-gnueabi/chromium-x11/127.0.6533.99/recipe-sysroot/usr/include",
```
but that didn't fix the issue.

I've searched for the header files in question in the sysroot, and this is where they seem to be located:
```
> fdfind stddef.h
usr/include/c++/v1/stddef.h
usr/include/linux/stddef.h

> fdfind stdarg.h
usr/include/c++/14.2.0/tr1/stdarg.h
```

Adding these directories using `-isystem` might work, but seems hacky; also, the `usr/include/c++/14.2.0/tr1/stdarg.h` path looks like it might not be the same across all clang versions / Yocto releases.

I've also tried using bindgen from `meta-clang` instead of Chromium's by adding a dependency on `bindgen-cli-native` and setting `rust_bindgen_root=".../recipe-sysroot-native/usr"` in `args.gn`, but that didn't seem to change anything.

@kraj do you maybe have suggestions what else I could try?

MaxIhlenfeldt, 2024-08-26:
@kraj gentle ping

kraj, 2024-08-26:
hmmm, it seems to be using vendored paths for libclang, Can you check if its using the tools from native sysroot ?

MaxIhlenfeldt, 2024-08-27:
@kraj 
> Can you check if its using the tools from native sysroot ?

What exactly do you mean?

g4bwy, 2025-06-03:
Hi,

Just pitching in on this issue since it seems to have been stalled and worked around for quite a while, but with chromium 138 it became a blocker because upstream generalized dependency on //build/rust/allocator to most crates.

The issue originates with a wrong -resource-dir parameter passed to bindgen: eg. on styhead we have clang 20.1.2, include files are located in ```recipe-sysroot-native/usr/lib/clang/20``` while the libraries are in ```recipe-sysroot-native/usr/lib/clang/20.1.2``` (not sure why meta-clang installs it that way, but nevermind...)

OTOH, //build/rust/bindgen_generator.gni uses clang_version to set clang_resource_dir:
```
    clang_resource_dir =
        rebase_path(clang_base_path + "/lib/clang/" + clang_version,
                    root_build_dir)
    args += [
      "-resource-dir",
      clang_resource_dir,
    ]
```

with the following patch I am able to provide the correct resource dir and bindgen works properly:
```
--- a/build/rust/rust_bindgen_generator.gni
+++ b/build/rust/rust_bindgen_generator.gni
@@ -219,8 +219,9 @@ template("rust_bindgen_generator") {
     # says the wrong thing. We point it to our clang's resource dir which will
     # make it behave consistently with our other command line flags and allows
     # system headers to be found.
+    clang_split_version = string_split(clang_version, ".")
     clang_resource_dir =
-        rebase_path(clang_base_path + "/lib/clang/" + clang_version,
+        rebase_path(clang_base_path + "/lib/clang/" + clang_split_version[0],
                     root_build_dir)
     args += [
       "-resource-dir",
```