OSSystems/meta-browser #834: Question on why chromium-x11 (kirkstone) always builds and does not use sstate cache

bbergquist0930, 2024-09-04:
I have a shared sstate-cache and for 99% of the recipes that is being used correctly however for chromium-x11, it always seems to want to compile and 3 hours later I have it built.  

Once built on my system, then it does not seem to get rebuilt.  But if I use a different system with the connection to the up to date sstate-cache, it will rebuild again on that system.

There must be some sort of dependency that seems to system specific that is in the signatures but I have not been able to find it and exclude it yet.  

So my question is if anyone knows what this might be?

OldManYellsAtCloud, 2024-09-04:
Bitbake generates a hash/signature of the recipe, and checks if the given signature is already stored in the sstate cache. If the signature doesn't exist, or it is different, then it it ignores the cache.

You can compare the signatures (with bitbake-dumpsig) of the repeated steps between two runs to see what change has triggered the rebuilding. See https://wiki.yoctoproject.org/wiki/TipsAndTricks/Understanding_what_changed_(diffsigs_etc)

MaxIhlenfeldt, 2024-11-25:
Closing this for now, happy to reopen if the problem persists and you have more input.

closed 2024-11-25 by MaxIhlenfeldt