OSSystems/meta-browser #841: Update to Chromium 130

MaxIhlenfeldt, 2024-10-17:
Release notes:
* https://chromereleases.googleblog.com/2024/10/stable-channel-update-for-desktop_15.html

AndreasUfert, 2024-10-17:
Oh, I am definitely curious about 130, since 129 somehow broke something in our Yocto image. While with 128 everything renders fine, 129 doesn't bother to render *any* font elements, not only on our kiosk application but also if I redirect the kiosk to youtube.com or something like that. It just shows a more or less gray or white canvas.

I can't really file an actual bug report because there's not much to report on - it worked with 128, it doesn't with 129, while neither the remote debug console nor the Chromium debug log show any hint to what could have been broken.

So my only hope is 130 to see if the problem somehow magically disappears.

MaxIhlenfeldt, 2024-10-24:
@AndreasUfert feel free to try out 130 already using https://github.com/MaxIhlenfeldt/meta-browser/tree/130.0.6723.58. Hopefully that'll fix your issue!

closed 2024-10-29 by MaxIhlenfeldt

AndreasUfert, 2024-10-30:
> @AndreasUfert feel free to try out 130 already using https://github.com/MaxIhlenfeldt/meta-browser/tree/130.0.6723.58. Hopefully that'll fix your issue!

Unfortunately, it does not. Anyway, as it seems we're the only ones affected by this, I will focus on our Yocto image and how X11, chromium-x11 and other components interact there.

AndreasUfert, 2024-10-30:
Okay, that was quick (sometimes trial & error is the most effective method): Turns out, `--enable-features="RawDraw"` is broken with 129 and 130, at least on our Yocto image.

Disabled it, works now. Thank you!

MaxIhlenfeldt, 2024-10-30:
> Turns out, --enable-features="RawDraw" is broken with 129 and 130, at least on our Yocto image.

Might be worth to file a bug at https://issues.chromium.org.

PR #844 (2024-10-29): chromium: Update to 130.0.6723.58
Fixes #841.

Release notes:
    https://chromereleases.googleblog.com/2024/10/stable-channel-update-for-desktop_15.html
    https://chromereleases.googleblog.com/2024/10/stable-channel-update-for-desktop_22.html

Build and patch changes:
------------------------

Add one patch to fix a build error on scarthgap with clang 18.

Rebase remaining patches.

License changes:
----------------

Added licenses:
  * third_party/android_deps/libs/com_google_dagger_hilt_core/LICENSE
  - https://chromium-review.googlesource.com/c/chromium/src/+/5823738

  * third_party/and [... cut]
commit 3e974410 (2024-10-29): chromium: Update to 130.0.6723.58 (#844)

Fixes #841.

Release notes:
    https://chromereleases.googleblog.com/2024/10/stable-channel-update-for-desktop_15.html
    https://chromereleases.googleblog.com/2024/10/stable-channel-update-for-desktop_22.html

Build and patch changes:
------------------------

Add one patch to fix a build error on scarthgap with clang 18.

Rebase remaining patches.

License changes:
----------------

Added licenses:
  * third_party/android_deps/libs/com_google_dagger_hilt_core/LICENSE
  - https://chromium-review.googlesource.com/c/chromium/src/+/5823738

  * third_p [... cut]