OSSystems/meta-browser #842: chromium: kirkstone: Update to Chromium 130

adalessandro, 2024-10-21:
Yocto Kirkstone is LTS (until Apr. 2026) but due to its dependency on `meta-clang` [0], the chromium version is freezed at `120.0.6099.224`.

This issue is to keep track of ongoing efforts on updating Chromium in Kirkstone.
Note that Chromium 130 [1] has been released and version will get bumped in `master` [2].

As part of this task, `meta-clang` needs to backport `clang >= 16` to kirkstone, see comment https://github.com/OSSystems/meta-browser/issues/716#issuecomment-2425448756

[0] https://github.com/OSSystems/meta-browser/issues/716
[1] https://chromereleases.googleblog.com/2024/10/stable-channel-update-for-desktop_15.html
[2] https://github.com/OSSystems/meta-browser/issues/841

adalessandro, 2024-10-21:
From @MaxIhlenfeldt https://github.com/OSSystems/meta-browser/issues/716#issuecomment-2426722443:

> Thanks for looking into this! Note that clang 16 is the bare minimum we need to build Chromium, but that might change again (or might have changed already since the last time we built kirkstone). Thus, having clang 17 or 18 (scarthgap uses 18) would be even better. Not sure how much more work it'd be to backport clang 18?

Agreed, the best approach here might be backporting from scarthgap, which currently provides clang `18.1.6`.

adalessandro, 2024-10-21:
@MaxIhlenfeldt BTW, have there been any effort on using the prebuilt clang toolchain (shipped along the chromium sources) instead of `clang-native`? Not sure if that's even possible.

MaxIhlenfeldt, 2024-10-21:
I think @kraj probably knows the reason we're not using the prebuilt toolchain.

adalessandro, 2024-10-21:
@MaxIhlenfeldt is there a specific `scarthgap` branch in `meta-browser`?

MaxIhlenfeldt, 2024-10-22:
scarthgap support is maintained on the `master` branch. `kirkstone` only has its own branch because we stopped supporting it on `master`.

adalessandro, 2024-10-22:
> I think @kraj probably knows the reason we're not using the prebuilt toolchain.

AFAICS, the prebuilt toolchain that is shipped along the chromium package comes in fact prebuilt and expects the sysroot to include several header and libraries. Their build system then pulls a prepacked debian image for that purpose, but pointing that to the yocto sysroot messes up everything.

@kraj that's my understanding from what I could see. Let me know if you agree or have other insights.

adalessandro, 2025-01-06:
Closing in favor of https://github.com/OSSystems/meta-browser/pull/843

closed 2025-01-06 by adalessandro