OSSystems/meta-browser #845: Chromium 129.0.6668.100 compile fails at the very end during linking with std::bad_alloc error

neenareddi, 2024-10-31:
Hi,

Chromium 129.0.6668.100 compile fails at the very end during linking with std::bad_alloc error. Built this using meta-clang styhead branch. We tried to build on a machine with high memory but it still failed. 

[image]

Did anyone run into this issue ? Any help would be much appreciated.

MaxIhlenfeldt, 2024-10-31:
I've been seeing this issue in my builds for the last few releases, but @SoumyaWind was able to build fine.

Maybe it's time to actually submit a bug report to LLVM as suggested in the output. Would it be possible for you to do that?

WujianSun, 2024-11-01:
On my side also meet the same issue on NXP imx boards

jcalvinowens, 2025-02-18:
I ran into this today too, the issue is the kernel's limit on the number of memory mappings each process is allowed to create: https://docs.kernel.org/admin-guide/sysctl/vm.html#max-map-count

You can set the limit to a larger number easily, just run this as root:
```
echo 1048576 > /proc/sys/vm/max_map_count
```

To make it permanent, add this to /etc/sysctl.conf:
```
vm.max_map_count = 1048576
```

stu-spp, 2025-03-05:
I was running into this problem too. Increasing the `vm.max_map_count` to `1048576` fixed it for me. Thank you!!!

MaxIhlenfeldt, 2025-03-06:
Thanks for finding the fix!

closed 2025-03-06 by MaxIhlenfeldt