OSSystems/meta-browser #847: armv7: LLD: output file too large

MastaG, 2024-11-08:
I'm building chromium-ozone-wayland for armv7 (32bit) but the linker doesn't seem to be happy at the end.
My local.conf:
```
TOOLCHAIN ?= "clang"
RUNTIME = "llvm"
TC_CXX_RUNTIME = "llvm"
LIBCPLUSPLUS = "-stdlib=libc++"
PREFERRED_PROVIDER_llvm = "clang"
PREFERRED_PROVIDER_llvm-native = "clang-native"
PREFERRED_PROVIDER_nativesdk-llvm = "nativesdk-clang"
PROVIDES:pn-clang = "llvm"
PROVIDES:pn-clang-native = "llvm-native"
PROVIDES:pn-nativesdk-clang = "nativesdk-llvm"
```
The log output:

```
"python3" "../../build/toolchain/gcc_link_wrapper.py" --output="./chrome" -- arm-oe-linux-gnueabi-clang++ -target arm-oe-linux-gnueabi   -mfpu=neon-vfpv4 -mfloat-abi=hard -mcpu=cortex-a15 -mlittle-endian --dyld-prefix=/usr -Qunused-arguments -D_TIME_BITS=64 -D_FILE_OFFSET_BITS=64 --sysroot=/home/mastag/src/odroid-oe-core/build/tmp/work/cortexa15t2hf-neon-vfpv4-oe-linux-gnueabi/chromium-ozone-wayland/130.0.6723.91/recipe-sysroot -Wl,--version-script=../../build/linux/chrome.map -fuse-ld=lld -Wl,--fatal-warnings -Wl,--build-id=sha1 -fPIC -Wl,-z,noexecstack -Wl,-z,relro -Wl,-z,now -Wl,--icf=all -Wl,--color-diagnostics -Wl,-mllvm,-instcombine-lower-dbg-declare=0 -Wl,--thinlto-cache-dir=thinlto-cache -Wl,--thinlto-cache-policy=cache_size=10\%:cache_size_bytes=40g:cache_size_files=100000 -flto=thin -Wl,--thinlto-jobs=all -Wl,-mllvm,-import-instr-limit=30 -Wl,-mllvm,-disable-auto-upgrade-debug-info -Wl,-mllvm,-inlinehint-threshold=360 -fwhole-program-vtables -march=armv7-a -Wl,--undefined-version -Wl,--no-call-graph-profile-sort -Wl,-O2 -Wl,--gc-sections -Wl,-z,defs -Wl,--as-needed -rdynamic -pie -Wl,--disable-new-dtags -Wl,--lto-O2 prebuilt_rustc_sysroot/lib/rustlib/armv7-oe-linux-gnueabihf/lib/libstd.rlib prebuilt_rustc_sysroot/lib/rustlib/armv7-oe-linux-gnueabihf/lib/liballoc.rlib prebuilt_rustc_sysroot/lib/rustlib/armv7-oe-linux-gnueabihf/lib/libcfg_if.rlib prebuilt_rustc_sysroot/lib/rustlib/armv7-oe-linux-gnueabihf/lib/libcompiler_builtins.rlib prebuilt_rustc_sysroot/lib/rustlib/armv7-oe-linux-gnueabihf/lib/libcore.rlib prebuilt_rustc_sysroot/lib/rustlib/armv7-oe-linux-gnueabihf/lib/libgetopts.rlib prebuilt_rustc_sysroot/lib/rustlib/armv7-oe-linux-gnueabihf/lib/libhashbrown.rlib prebuilt_rustc_sysroot/lib/rustlib/armv7-oe-linux-gnueabihf/lib/libpanic_abort.rlib prebuilt_rustc_sysroot/lib/rustlib/armv7-oe-linux-gnueabihf/lib/libpanic_unwind.rlib prebuilt_rustc_sysroot/lib/rustlib/armv7-oe-linux-gnueabihf/lib/librustc_demangle.rlib prebuilt_rustc_sysroot/lib/rustlib/armv7-oe-linux-gnueabihf/lib/libstd_detect.rlib prebuilt_rustc_sysroot/lib/rustlib/armv7-oe-linux-gnueabihf/lib/libtest.rlib prebuilt_rustc_sysroot/lib/rustlib/armv7-oe-linux-gnueabihf/lib/libunicode_width.rlib prebuilt_rustc_sysroot/lib/rustlib/armv7-oe-linux-gnueabihf/lib/libunwind.rlib prebuilt_rustc_sysroot/lib/rustlib/armv7-oe-linux-gnueabihf/lib/libaddr2line.rlib prebuilt_rustc_sysroot/lib/rustlib/armv7-oe-linux-gnueabihf/lib/libadler.rlib prebuilt_rustc_sysroot/lib/rustlib/armv7-oe-linux-gnueabihf/lib/libgimli.rlib prebuilt_rustc_sysroot/lib/rustlib/armv7-oe-linux-gnueabihf/lib/liblibc.rlib prebuilt_rustc_sysroot/lib/rustlib/armv7-oe-linux-gnueabihf/lib/libmemchr.rlib prebuilt_rustc_sysroot/lib/rustlib/armv7-oe-linux-gnueabihf/lib/libminiz_oxide.rlib prebuilt_rustc_sysroot/lib/rustlib/armv7-oe-linux-gnueabihf/lib/libobject.rlib -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed   -fmacro-prefix-map=/home/mastag/src/odroid-oe-core/build/tmp/work/cortexa15t2hf-neon-vfpv4-oe-linux-gnueabi/chromium-ozone-wayland/130.0.6723.91/chromium-130.0.6723.91=/usr/src/debug/chromium-ozone-wayland/130.0.6723.91  -fdebug-prefix-map=/home/mastag/src/odroid-oe-core/build/tmp/work/cortexa15t2hf-neon-vfpv4-oe-linux-gnueabi/chromium-ozone-wayland/130.0.6723.91/chromium-130.0.6723.91=/usr/src/debug/chromium-ozone-wayland/130.0.6723.91  -fmacro-prefix-map=/home/mastag/src/odroid-oe-core/build/tmp/work/cortexa15t2hf-neon-vfpv4-oe-linux-gnueabi/chromium-ozone-wayland/130.0.6723.91/chromium-130.0.6723.91/out/Release=/usr/src/debug/chromium-ozone-wayland/130.0.6723.91  -fdebug-prefix-map=/home/mastag/src/odroid-oe-core/build/tmp/work/cortexa15t2hf-neon-vfpv4-oe-linux-gnueabi/chromium-ozone-wayland/130.0.6723.91/chromium-130.0.6723.91/out/Release=/usr/src/debug/chromium-ozone-wayland/130.0.6723.91  -fdebug-prefix-map=/home/mastag/src/odroid-oe-core/build/tmp/work/cortexa15t2hf-neon-vfpv4-oe-linux-gnueabi/chromium-ozone-wayland/130.0.6723.91/recipe-sysroot=  -fmacro-prefix-map=/home/mastag/src/odroid-oe-core/build/tmp/work/cortexa15t2hf-neon-vfpv4-oe-linux-gnueabi/chromium-ozone-wayland/130.0.6723.91/recipe-sysroot=  -fdebug-prefix-map=/home/mastag/src/odroid-oe-core/build/tmp/work/cortexa15t2hf-neon-vfpv4-oe-linux-gnueabi/chromium-ozone-wayland/130.0.6723.91/recipe-sysroot-native=  -fmacro-prefix-map=/home/mastag/src/odroid-oe-core/build/tmp/work/cortexa15t2hf-neon-vfpv4-oe-linux-gnueabi/chromium-ozone-wayland/130.0.6723.91/recipe-sysroot-native=   -stdlib=libc++ -fuse-ld=lld -o "./chrome" -Wl,--start-group @"./chrome.rsp" -Wl,--end-group   /home/mastag/src/odroid-oe-core/build/tmp/work/cortexa15t2hf-neon-vfpv4-oe-linux-gnueabi/chromium-ozone-wayland/130.0.6723.91/recipe-sysroot-native/usr/lib/clang/latest/lib/linux/libclang_rt.builtins-armhf.a -latomic -ldl -lpthread -lrt -lgmodule-2.0 -lgobject-2.0 -lgthread-2.0 -lglib-2.0 -lsmime3 -lnss3 -lsoftokn3 -lnssutil3 -lplds4 -lplc4 -lnspr4 -ldbus-1 -latk-1.0 -latk-bridge-2.0 -lcups -lgio-2.0 -lresolv -lexpat -luuid -ldrm -lxcb -lxkbcommon -lm -lz -lX11 -lXcomposite -lXdamage -lXext -lXfixes -lXrender -lXrandr -lXtst -lgbm -lEGL -lXi -lpci -lffi -lpangocairo-1.0 -lpango-1.0 -lharfbuzz -lcairo -ludev -lasound -latspi -lxshmfence -Wl,--start-group obj/third_party/rust/cxx/v1/lib/libcxx_lib.rlib obj/third_party/rust/serde_json_lenient/v0_2/wrapper/libthird_uparty_srust_sserde_ujson_ulenient_sv0_u2_swrapper_cwrapper.rlib obj/third_party/rust/serde/v1/lib/libserde_lib.rlib obj/third_party/rust/serde_json_lenient/v0_2/lib/libserde_json_lenient_lib.rlib obj/third_party/rust/itoa/v1/lib/libitoa_lib.rlib obj/third_party/rust/ryu/v1/lib/libryu_lib.rlib obj/build/rust/chromium_prelude/libchromium.rlib obj/skia/libskia_cbridge_urust_uside.rlib obj/third_party/rust/font_types/v0_7/lib/libfont_types_lib.rlib obj/third_party/rust/bytemuck/v1/lib/libbytemuck_lib.rlib obj/third_party/rust/read_fonts/v0_22/lib/libread_fonts_lib.rlib obj/third_party/rust/skrifa/v0_22/lib/libskrifa_lib.rlib obj/third_party/blink/common/rust_crash/libthird_uparty_sblink_scommon_srust_ucrash_crs.rlib obj/components/qr_code_generator/libcomponents_sqr_ucode_ugenerator_cqr_ucode_ugenerator_uffi_uglue.rlib obj/third_party/rust/qr_code/v2/lib/libqr_code_lib.rlib -Wl,--end-group 
arm-oe-linux-gnueabi-ld.lld: error: output file too large: 6966937872 bytes
section sizes:
.interp 29
.note.ABI-tag 32
.note.crashpad.info 28
.note.gnu.build-id 36
.dynsym 26160
.gnu.version 3270
.gnu.version_r 848
.gnu.hash 372
.dynstr 36071
.rel.dyn 7120800
.ARM.exidx 8808
.rel.plt 12088
.rodata 14734757
.ARM.extab 12996
.eh_frame_hdr 12
.eh_frame 4
.text 146279256
.init 28
.fini 8
malloc_hook 500
.plt 24208
.tdata 176
.tbss 2980
.fini_array 4
.init_array 8
.data.rel.ro 4882544
.dynamic 464
.got 56168
.got.plt 6056
.relro_padding 1136
.data 381920
.tm_clone_table 0
protected_memory 4118
protected_memory_buffer 1
.bss 1848236
.comment 133
.ARM.attributes 59
.debug_aranges 264
.debug_info 3698584221
.debug_abbrev 142859418
.debug_line 280798254
.debug_str 1396758032
.debug_line_str 4129043
.debug_rnglists 154099059
.debug_loclists 331209667
.debug_str_offsets 594928852
.debug_addr 38065804
.debug_frame 22817568
.symtab 31863328
.shstrtab 586
.strtab 97220723

arm-oe-linux-gnueabi-clang++: error: linker command failed with exit code 1 (use -v to see invocation)
ninja: build stopped: subcommand failed.
WARNING: /home/mastag/src/odroid-oe-core/build/tmp/work/cortexa15t2hf-neon-vfpv4-oe-linux-gnueabi/chromium-ozone-wayland/130.0.6723.91/temp/run.do_compile.57401:149 exit 1 from 'ninja -v -j 32 chrome chrome_sandbox chromedriver.unstripped'
WARNING: Backtrace (BB generated script): 
        #1: do_compile, /home/mastag/src/odroid-oe-core/build/tmp/work/cortexa15t2hf-neon-vfpv4-oe-linux-gnueabi/chromium-ozone-wayland/130.0.6723.91/temp/run.do_compile.57401, line 149
        #2: main, /home/mastag/src/odroid-oe-core/build/tmp/work/cortexa15t2hf-neon-vfpv4-oe-linux-gnueabi/chromium-ozone-wayland/130.0.6723.91/temp/run.do_compile.57401, line 153
```

MaxIhlenfeldt, 2024-11-12:
I think @SoumyaWind was dealing with something similar?

SoumyaWind, 2024-11-12:
Yes, the issue got resolved with the following debug configuration:
is_debug=true, is_official_build=false, symbol_level=2

Additionally, the following dependencies were required:
DISTRO_FEATURES with usrmerge and systemd
VIRTUAL-RUNTIME_init_manager set to systemd
systemd and systemd-dev in IMAGE_INSTALL

Refer: https://github.com/OSSystems/meta-browser/pull/844#issuecomment-2443450417

MastaG, 2024-11-13:
@SoumyaWind
I don't really get why you're purposely enabling and increasing debug symbols by setting: `is_debug=true` and `symbol_level=2`.
Doesn't that increase the binary even more?

The recipe enforces `symbol_level=0`.
However bitbake still seems to add the `-g` flag.

I worked around this by adding:
```
CFLAGS:remove:arm = "-g"
CXXFLAGS:remove:arm = "-g"
```
To my bbappend.
I know this strips away all debug symbols, as `symbol_level=0`  basically sets `-g0`.
But hey... at least it builds now.

prevas-emkan, 2025-01-24:
I'm also seeing this "output file too large" error when building Chromium 131.0.6778.139 with Clang 19.1.0. More specifically, I'm using these commits:

- `meta-browser` commit `502433e1efa0 ("chromium: Update to 131.0.6778.139 (#861)")`
- `meta-clang` commit `c709a5196f1e ("clang_git.bb: assign PYTHON_PN to LLDB_PYTHON_EXE_RELATIVE_PATH")`

I tried disabling optimizations and debug symbols without any luck.

```
$ cat chromium-ozone-wayland_%.bbappend
# Disable ThinLTO optimization to reduce memory overhead.
# https://chromium.googlesource.com/chromium/src/build/config/+/refs/heads/main/compiler/BUILD.gn#119
GN_ARGS += "use_thin_lto=false thin_lto_enable_cache=false thin_lto_enable_optimizations=false"

# The 'is_official_build' flag enables further optimizations, so disable it.
# https://chromium.googlesource.com/chromium/src/build/config/+/refs/heads/main/compiler/BUILD.gn
# See also meta-browser/meta-chromium/recipes-browser/chromium/chromium-gn.inc
GN_ARGS += "is_official_build=false"

# Disable debug symbols.
GN_ARGS += "is_debug=false"

# Disable blink and v8 debug symbols.
# https://chromium.googlesource.com/chromium/src/+/main/docs/linux/build_instructions.md#disable-debug-symbols-for-blink-and-v8
GN_ARGS += "blink_symbol_level=0 v8_symbol_level=0"

# Do not generate unwind tables.
GN_ARGS += "exclude_unwind_tables=true"
```

Below is the relevant part of `log_do.compile` with the above changes to the `chromium-ozone-wayland` recipe:

```
"python3" "../../build/toolchain/gcc_link_wrapper.py" --output="./chrome" -- arm-oe-linux-gnueabi-clang++ -target arm-oe-linux-gnueabi   -mfpu=neon -mfloat-abi=hard -mcpu=cortex-a9 -mlittle-endian --dyld-prefix=/usr -Qunused-arguments -D_TIME_BITS=64 -D_FILE_OFFSET_BITS=64 --sysroot=path/to/build/tmp/work/cortexa9t2hf-neon-oe-linux-gnueabi/chromium-ozone-wayland/131.0.6778.139/recipe-sysroot -Wl,--version-script=../../build/linux/chrome.map -fuse-ld=lld -Wl,--fatal-warnings -Wl,--build-id=fast -fPIC -Wl,-z,noexecstack -Wl,-z,relro -Wl,-z,now -Wl,--icf=all -Wl,--color-diagnostics -Wl,--undefined-version -Wl,--no-call-graph-profile-sort -Wl,--gc-sections -Wl,-z,defs -Wl,--as-needed -rdynamic -pie -Wl,--disable-new-dtags prebuilt_rustc_sysroot/lib/rustlib/armv7-oe-linux-gnueabihf/lib/libstd.rlib prebuilt_rustc_sysroot/lib/rustlib/armv7-oe-linux-gnueabihf/lib/liballoc.rlib prebuilt_rustc_sysroot/lib/rustlib/armv7-oe-linux-gnueabihf/lib/libcfg_if.rlib prebuilt_rustc_sysroot/lib/rustlib/armv7-oe-linux-gnueabihf/lib/libcompiler_builtins.rlib prebuilt_rustc_sysroot/lib/rustlib/armv7-oe-linux-gnueabihf/lib/libcore.rlib prebuilt_rustc_sysroot/lib/rustlib/armv7-oe-linux-gnueabihf/lib/libgetopts.rlib prebuilt_rustc_sysroot/lib/rustlib/armv7-oe-linux-gnueabihf/lib/libhashbrown.rlib prebuilt_rustc_sysroot/lib/rustlib/armv7-oe-linux-gnueabihf/lib/libpanic_abort.rlib prebuilt_rustc_sysroot/lib/rustlib/armv7-oe-linux-gnueabihf/lib/libpanic_unwind.rlib prebuilt_rustc_sysroot/lib/rustlib/armv7-oe-linux-gnueabihf/lib/librustc_demangle.rlib prebuilt_rustc_sysroot/lib/rustlib/armv7-oe-linux-gnueabihf/lib/libstd_detect.rlib prebuilt_rustc_sysroot/lib/rustlib/armv7-oe-linux-gnueabihf/lib/libtest.rlib prebuilt_rustc_sysroot/lib/rustlib/armv7-oe-linux-gnueabihf/lib/libunicode_width.rlib prebuilt_rustc_sysroot/lib/rustlib/armv7-oe-linux-gnueabihf/lib/libunwind.rlib prebuilt_rustc_sysroot/lib/rustlib/armv7-oe-linux-gnueabihf/lib/libaddr2line.rlib prebuilt_rustc_sysroot/lib/rustlib/armv7-oe-linux-gnueabihf/lib/libadler.rlib prebuilt_rustc_sysroot/lib/rustlib/armv7-oe-linux-gnueabihf/lib/libgimli.rlib prebuilt_rustc_sysroot/lib/rustlib/armv7-oe-linux-gnueabihf/lib/liblibc.rlib prebuilt_rustc_sysroot/lib/rustlib/armv7-oe-linux-gnueabihf/lib/libmemchr.rlib prebuilt_rustc_sysroot/lib/rustlib/armv7-oe-linux-gnueabihf/lib/libminiz_oxide.rlib prebuilt_rustc_sysroot/lib/rustlib/armv7-oe-linux-gnueabihf/lib/libobject.rlib -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed   -fmacro-prefix-map=path/to/build/tmp/work/cortexa9t2hf-neon-oe-linux-gnueabi/chromium-ozone-wayland/131.0.6778.139/chromium-131.0.6778.139=/usr/src/debug/chromium-ozone-wayland/131.0.6778.139  -fdebug-prefix-map=path/to/build/tmp/work/cortexa9t2hf-neon-oe-linux-gnueabi/chromium-ozone-wayland/131.0.6778.139/chromium-131.0.6778.139=/usr/src/debug/chromium-ozone-wayland/131.0.6778.139  -fmacro-prefix-map=path/to/build/tmp/work/cortexa9t2hf-neon-oe-linux-gnueabi/chromium-ozone-wayland/131.0.6778.139/chromium-131.0.6778.139/out/Release=/usr/src/debug/chromium-ozone-wayland/131.0.6778.139  -fdebug-prefix-map=path/to/build/tmp/work/cortexa9t2hf-neon-oe-linux-gnueabi/chromium-ozone-wayland/131.0.6778.139/chromium-131.0.6778.139/out/Release=/usr/src/debug/chromium-ozone-wayland/131.0.6778.139  -fdebug-prefix-map=path/to/build/tmp/work/cortexa9t2hf-neon-oe-linux-gnueabi/chromium-ozone-wayland/131.0.6778.139/recipe-sysroot=  -fmacro-prefix-map=path/to/build/tmp/work/cortexa9t2hf-neon-oe-linux-gnueabi/chromium-ozone-wayland/131.0.6778.139/recipe-sysroot=  -fdebug-prefix-map=path/to/build/tmp/work/cortexa9t2hf-neon-oe-linux-gnueabi/chromium-ozone-wayland/131.0.6778.139/recipe-sysroot-native=  -fmacro-prefix-map=path/to/build/tmp/work/cortexa9t2hf-neon-oe-linux-gnueabi/chromium-ozone-wayland/131.0.6778.139/recipe-sysroot-native=   -stdlib=libc++ -o "./chrome" -Wl,--start-group @"./chrome.rsp" -Wl,--end-group   path/to/build/tmp/work/cortexa9t2hf-neon-oe-linux-gnueabi/chromium-ozone-wayland/131.0.6778.139/recipe-sysroot-native/usr/lib/clang/l [... cut ...]
.ARM.attributes 57
.debug_aranges 264
.debug_info 3642812853
.debug_abbrev 124336568
.debug_line 360685042
.debug_str 1510751854
.debug_line_str 3569863
.debug_rnglists 164653636
.debug_str_offsets 542818540
.debug_addr 55206080
.debug_frame 59114348
.debug_loclists 460727013
.symtab 46594896
.shstrtab 586
.strtab 149165118

arm-oe-linux-gnueabi-clang++: error: linker command failed with exit code 1 (use -v to see invocation)
ninja: build stopped: subcommand failed.
WARNING: exit code 1 from a shell command.
```