OSSystems/meta-browser #851: libxslt was skipped: Recipe is blacklisted

Arjun-741, 2024-11-20:
I added into meta-browser recipe into my yocto build system and try to build got it issue

 ERROR: Nothing PROVIDES 'libxslt' (but /home/mkarpurapu/workspace/HLOS_IDEA_FORGE/poky/meta-browser/meta-chromium/recipes-browser/chromium/chromium-ozone-wayland_130.0.6723.91.bb DEPENDS on or otherwise requires it)
libxslt was skipped: Recipe is blacklisted: Using ubuntu toolchain
ERROR: Required build target 'chromium-ozone-wayland' has no buildable providers.
Missing or unbuildable dependency chain was: ['chromium-ozone-wayland', 'libxslt']

MastaG, 2024-11-20:
libxslt should be in openembedded-core: `openembedded-core/meta/recipes-support/libxslt`
But it's not a hard requirement.
I've created a bbappend for chromium which removes the requirement for some system libraries, so the build can also run on other distros such as Ubuntu.
See: https://github.com/MastaG/odroid-oe-core/blob/master/odroid-os/recipes-browser/chromium/chromium-%25.bbappend
```
GN_UNBUNDLE_LIBS = ""
DEPENDS:remove = "\
        flac \
        jpeg \
        libxslt \
"
```
I understand that people want to link against system libraries as much as possible due to security updates.
But it makes chromium less portable for me.
I'd like to link statically with its bundled libraries as much as possible.

Arjun-741, 2024-11-22:
Not working

MaxIhlenfeldt, 2024-11-25:
libxslt being blacklisted on Ubuntu is outside of our control. The bbappend approach from https://github.com/OSSystems/meta-browser/issues/851#issuecomment-2488249372 seems like it should work (if the error is that meta-browser depends on libxslt and you remove that dependency, the error should be gone?).

closed 2024-11-25 by MaxIhlenfeldt