OSSystems/meta-browser #853: Broken Firefox build on Dunfell

mdanikin, 2024-11-22:
Has anyone ever successfully built Firefox on dunfell?
Before the commit cb79ef075dab7ff4f680138a9959f5635c2bb313 build failed during the compile time; see #717 
After this commit, rust-common-without-python3native.bbclass inherits rust-target-config.bbclass.
It should be a part of poky/oe-core, but it is not available in dunfell, as well as kirkstone.

MaxIhlenfeldt, 2024-11-25:
I'm afraid this repo's Firefox recipe is horribly out of date. You might have better luck with the fork mentioned in #822. Sorry that that's all the help I can provide!

closed 2024-11-25 by MaxIhlenfeldt

OldManYellsAtCloud, 2024-11-25:
I have a fork with a more updated version, that I used to test-build with Dunfell regularly, up to version 115.15esr (which is also outdated now, though not as much as v68). If interested, you can find the last commit supporting Dunfell here: https://github.com/OldManYellsAtCloud/meta-browser/tree/34932860a695967ea417d25e7353372e7d2a0974

(I plan to upstream it at some point, though to be honest, have no idea when will that happen)

MaxIhlenfeldt, 2024-11-25:
> I plan to upstream it at some point, though to be honest, have no idea when will that happen

We'd love to review and merge those changes!