OSSystems/meta-browser #854: Building CEF

MattCrossley99, 2024-12-03:
Hi, I'm attempting to build Chromium Embedded Framework for a Yocto image.

It appears there was previously a recipe for building cef in this repo but there no longer is, is there a way to build CEF easily using the current recipes? I'd like to explore this before going the route of creating my own recipe.

MaxIhlenfeldt, 2024-12-04:
I'm not familiar with the CEF build process, but if we dropped support for CEF (must've been before I became a maintainer) there probably was a reason? Unfortunately I don't think I can help you with this, but maybe someone else can chime in.

In any case, if you get it to work and think we should add back CEF support, we can reopen this issue to discuss if we want to incorporate the changes.

closed 2024-12-04 by MaxIhlenfeldt

rakuco, 2024-12-04:
> I'm not familiar with the CEF build process, but if we dropped support for CEF (must've been before I became a maintainer) there probably was a reason? Unfortunately I don't think I can help you with this, but maybe someone else can chime in.

I just have a vague memory of this, to be honest. The recipe was removed in commit 9162c5a062e75129d2065377ffa680a983855224 but had apparently been broken for a while; IIRC getting this cleaned up was also necessary for #62 (which covered updating the recipe to its current format, the previous one shared files between CEF and chromium-{x11,ozone-wayland}).

This happened a long time ago, so CEF's build process has probably changed and I'd start from scratch if I were to create a CEF recipe again.