OSSystems/meta-browser #869: Chromium V4L2 Support

chrisvollorenew, 2025-02-10:
Hello! Currently the repo supports chromium v131. However, according to my research, v4l2 support was added to chromium in version 132. Has anybody tried v4l2 on chromium? What might be involved in upgrading chromium to version 132?

MaxIhlenfeldt, 2025-02-11:
I can't say anything about v4l2, but the update to m132 is being worked on in #867.

MaxIhlenfeldt, 2025-02-11:
Out of curiosity, where did you learn about v4l2 support being added in m132? A quick search didn't yield anything for me.

chrisvollorenew, 2025-02-11:
@MaxIhlenfeldt thank you! I'm looking forward to trying it. Is it able to build now? If it builds, our team can test it and provide feedback. It looks like you're getting close?

Here is more info about v4l2 support in 132+ https://github.com/saiarcot895/chromium-ubuntu-build/issues/65#issuecomment-2533504008

avi951, 2025-04-18:
I am trying to build Chromium, but not able to get `Video Decode: Hardware accelerated`, getting Software rendering in Rockchip RK3588. Any Idea on it?
I have noticed that in Debian Chromium package `132.0.6834.159` supports `Hardware accelerated` support by default in Rockchip RK3588.