OSSystems/meta-browser #871: chromium-ozone-wayland_101.0.4951.54 build error

phyljh, 2025-02-26:
I tried to build chromium-ozone-wayland_101.0.4951.54.bb  on dunefell and build failed as below.

[13830/55505] clang++ -MMD -MF yocto_native/obj/base/third_party/symbolize/symbolize/symbolize.o.d -DUSE_UDEV -DUSE_AURA=1 -DUSE_GLIB=1 -DUSE_OZONE=1 -DOFFICIAL_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -DNO_UNWIND_TABLES -D_GNU_SOURCE -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -I../.. -Iyocto_native/gen -fno-delete-null-pointer-checks -fno-ident -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -pthread -fcolor-diagnostics -fmerge-all-constants -fcrash-diagnostics-dir=../../tools/clang/crashreports -mllvm -instcombine-lower-dbg-declare=0 -ffp-contract=off -m64 -march=x86-64 -msse3 -Xclang -fdebug-compilation-dir -Xclang . -ftrivial-auto-var-init=pattern -O2 -fdata-sections -ffunction-sections -fno-unique-section-names -fno-omit-frame-pointer -g0 -fvisibility=hidden -Wheader-hygiene -Wstring-conversion -Wtautological-overlap-compare -Wall -Wno-unused-variable -Wno-c++11-narrowing -Wno-unused-but-set-variable -Wno-misleading-indentation -Wno-missing-field-initializers -Wno-unused-parameter -Wloop-analysis -Wno-unneeded-internal-declaration -Wenum-compare-conditional -Wno-unknown-warning-option -Wno-psabi -Wno-ignored-pragma-optimize -Wno-unqualified-std-cast-call -std=c++17 -Wno-trigraphs -fno-aligned-new -fno-exceptions -fno-rtti -fvisibility-inlines-hidden -isystem/home/jaehyunlee/workspace/homerobot0226/apps_proc/build/tmp-glibc/work/aarch64-oe-linux/chromium-ozone-wayland/101.0.4951.54-r0/recipe-sysroot-native/usr/include   -isystem/home/jaehyunlee/workspace/homerobot0226/apps_proc/build/tmp-glibc/work/aarch64-oe-linux/chromium-ozone-wayland/101.0.4951.54-r0/recipe-sysroot-native/usr/include   -O2 -pipe -c ../../base/third_party/symbolize/symbolize.cc -o yocto_native/obj/base/third_party/symbolize/symbolize/symbolize.o
| FAILED: yocto_native/obj/base/third_party/symbolize/symbolize/symbolize.o
| clang++ -MMD -MF yocto_native/obj/base/third_party/symbolize/symbolize/symbolize.o.d -DUSE_UDEV -DUSE_AURA=1 -DUSE_GLIB=1 -DUSE_OZONE=1 -DOFFICIAL_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -DNO_UNWIND_TABLES -D_GNU_SOURCE -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -I../.. -Iyocto_native/gen -fno-delete-null-pointer-checks -fno-ident -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -pthread -fcolor-diagnostics -fmerge-all-constants -fcrash-diagnostics-dir=../../tools/clang/crashreports -mllvm -instcombine-lower-dbg-declare=0 -ffp-contract=off -m64 -march=x86-64 -msse3 -Xclang -fdebug-compilation-dir -Xclang . -ftrivial-auto-var-init=pattern -O2 -fdata-sections -ffunction-sections -fno-unique-section-names -fno-omit-frame-pointer -g0 -fvisibility=hidden -Wheader-hygiene -Wstring-conversion -Wtautological-overlap-compare -Wall -Wno-unused-variable -Wno-c++11-narrowing -Wno-unused-but-set-variable -Wno-misleading-indentation -Wno-missing-field-initializers -Wno-unused-parameter -Wloop-analysis -Wno-unneeded-internal-declaration -Wenum-compare-conditional -Wno-unknown-warning-option -Wno-psabi -Wno-ignored-pragma-optimize -Wno-unqualified-std-cast-call -std=c++17 -Wno-trigraphs -fno-aligned-new -fno-exceptions -fno-rtti -fvisibility-inlines-hidden -isystem/home/jaehyunlee/workspace/homerobot0226/apps_proc/build/tmp-glibc/work/aarch64-oe-linux/chromium-ozone-wayland/101.0.4951.54-r0/recipe-sysroot-native/usr/include   -isystem/home/jaehyunlee/workspace/homerobot0226/apps_proc/build/tmp-glibc/work/aarch64-oe-linux/chromium-ozone-wayland/101.0.4951.54-r0/recipe-sysroot-native/usr/include   -O2 -pipe -c ../../base/third_party/symbolize/symbolize.cc -o yocto_native/obj/base/third_party/symbolize/symbolize/symbolize.o
| In file included from ../../base/third_party/symbolize/symbolize.cc:65:
| ../../base/third_party/symbolize/symbolize.h:123:53: error: no member named 'exchange' in namespace 'std'; did you mean '__exchange'?
|   FileDescriptor(FileDescriptor&& other) : fd_(std::exchange(other.fd_, -1)) {}
|                                                ~~~~~^~~~~~~~
|                                                     __exchange
| /usr/lib/gcc/x86_64-linux-gnu/12/../../../../include/c++/12/bits/move.h:157:5: note: '__exchange' declared here
|     __exchange(_Tp& __obj, _Up&& __new_val)
|     ^
| In file included from ../../base/third_party/symbolize/symbolize.cc:65:
| ../../base/third_party/symbolize/symbolize.h:127:18: error: no member named 'exchange' in namespace 'std'; did you mean '__exchange'?
|       fd_ = std::exchange(rhs.fd_, -1);
|             ~~~~~^~~~~~~~
|                  __exchange
| /usr/lib/gcc/x86_64-linux-gnu/12/../../../../include/c++/12/bits/move.h:157:5: note: '__exchange' declared here
|     __exchange(_Tp& __obj, _Up&& __new_val)
|     ^
| 2 errors generated.

phyljh, 2025-02-27:
I changed denfell-clang12 into denfell-clang14 and built successfully.
Thanks

rakuco, 2025-03-06:
Glad to hear it's working now!