OSSystems/meta-browser #877: Clang upgrade to 20.1.0 in meta-clang

kraj, 2025-03-11:
This is for information, since chromium builds will be directly impacted. There is [PR for upgrading clang from 19 -> 20](https://github.com/kraj/meta-clang/pull/1060) in meta-clang right now under test, I have chroumium-ozone-wayland builds working for x86_64 but would appreciate any help in testing other chromium configs that you might be interested in and giving feedback on how it goes with clang20

OldManYellsAtCloud, 2025-04-27:
Just wanted to look at some Firefox issues with clang20, but I have faced 2 problems with the latest meta-clang when I tried to compile wasi-sdk.

1. By default, it fails with the following error:
```
| CMake Error at /yocto/walnascar/build/tmp/work/x86_64-linux/wasi-sdk-native/git/toolchain.cmake:11 (find_program):
|   Could not find CMAKE_AR using the following names: llvm-ar
```

I kinda recall some similar mesa issues, though I might be mistaken... anyway. I can get over this if I add `llvm-native` to the dependencies.

2. Once I get over the first issue, I get a second one (during configure):
```
|     Run Build Command(s): ninja -v cmTC_193c7
|     [1/2] /yocto/walnascar/build/tmp/work/x86_64-linux/wasi-sdk-native/git/recipe-sysroot-native/usr/bin/clang   -isystem/yocto/walnascar/build/tmp/work/x86_64-linux/wasi-sdk-native/git/recipe-sysroot-native/usr/include -O2 -pipe -o CMakeFiles/cmTC_193c7.dir/testCCompiler.c.o -c /yocto/walnascar/build/tmp/work/x86_64-linux/wasi-sdk-native/git/build/CMakeFiles/CMakeScratch/TryCompile-fe8oht/testCCompiler.c
|     FAILED: CMakeFiles/cmTC_193c7.dir/testCCompiler.c.o
|     /yocto/walnascar/build/tmp/work/x86_64-linux/wasi-sdk-native/git/recipe-sysroot-native/usr/bin/clang   -isystem/yocto/walnascar/build/tmp/work/x86_64-linux/wasi-sdk-native/git/recipe-sysroot-native/usr/include -O2 -pipe -o CMakeFiles/cmTC_193c7.dir/testCCompiler.c.o -c /yocto/walnascar/build/tmp/work/x86_64-linux/wasi-sdk-native/git/build/CMakeFiles/CMakeScratch/TryCompile-fe8oht/testCCompiler.c
|     /yocto/walnascar/build/tmp/work/x86_64-linux/wasi-sdk-native/git/recipe-sysroot-native/usr/bin/clang: symbol lookup error: /yocto/walnascar/build/tmp/work/x86_64-linux/wasi-sdk-native/git/recipe-sysroot-native/usr/bin/clang: undefined symbol: LLVMInitializeAArch64TargetInfo, version LLVM_20.1
|     ninja: build stopped: subcommand failed.
```

Before I'd dive into it, wanted to ask if maybe you have any idea from top of your head about  undefined `LLVMInitializeAArch64TargetInfo`... or maybe I should stay in clang19 until clang20 gets into oe-core?

kraj, 2025-04-27:
I think it is due to llvm doing clang builds also in core, I have patches in flight to fix this such that whole clang is in core then we can just use it in meta-clang. if you are interested to try it out use

https://web.git.yoctoproject.org/poky-contrib/log/?h=kraj/clang-merge-2

and

https://github.com/kraj/meta-clang/pull/1088

applied to clang master branch

OldManYellsAtCloud, 2025-04-29:
Thanks. Unfortunately when I try to clone the poky-contrib repo, I get only error 500. Not sure if the server is being hammered again, or it just requires special permission. Anyway, after I applied your latest clang patches from the oe-core mailing list I managed to compile  wasi-sdk successfully, without meta-clang and without doing anything extra. Thanks for the hint.

kraj, 2025-04-29:
> Thanks. Unfortunately when I try to clone the poky-contrib repo, I get only error 500. Not sure if the server is being hammered again, or it just requires special permission. Anyway, after I applied [...]

Yeah the link usually says web.git.yoctoproject.org to mitigate the hammers, so you  have to manually remove `web.` from the URI