OSSystems/meta-browser #881: chromium-ozone-wayland does not compile on styhead

DasRoteSkelett, 2025-04-02:
Hi!

I have been trying to compile chromium-ozone-wayland a couple of times with the last updates and it fails. Here are the relevant logs:
```
Build Configuration:
BB_VERSION           = "2.9.1"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "x86_64-poky-linux"
MACHINE              = "qemux86-64"
DISTRO               = "poky"
DISTRO_VERSION       = "5.1.4"
TUNE_FEATURES        = "m64 core2"
TARGET_FPU           = ""
meta                 
meta-poky            
meta-yocto-bsp       = "styhead:70dc28ac287bf35541270cae1d99130a0f6b7b5f"
meta-oe              
meta-python          = "styhead:c93994f1bb601c57548b588d4f77a044d90822cf"
meta-clang           = "styhead:c709a5196f1e8654425b43b478064395386c36d4"
meta-chromium        = "master:bb8662912354dae13634c0ec35c3803c344b1e72"

[...]
"python3" "../../build/toolchain/gcc_link_wrapper.py" --output="./chrome" -- x86_64-poky-linux-clang++ -target x86_64-poky-linux  -m64 -march=core2 -mtune=core2 -msse3 -mfpmath=sse -mlittle-endian -Qunused-arguments -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/mschoepf/yocto/vanilla/build/tmp/work/core2-64-poky-linux/chromium-ozone-wayland/132.0.6834.83/recipe-sysroot -Wl,--version-script=../../build/linux/chrome.map -fuse-ld=lld -Wl,--fatal-warnings -Wl,--build-id=sha1 -fPIC -Wl,-z,noexecstack -Wl,-z,relro -Wl,-z,now -Wl,--icf=all -Wl,--color-diagnostics -Wl,-mllvm,-instcombine-lower-dbg-declare=0 -Wl,--thinlto-cache-dir=thinlto-cache -Wl,--thinlto-cache-policy=cache_size=10\%:cache_size_bytes=40g:cache_size_files=100000 -flto=thin -Wl,--thinlto-jobs=all -Wl,-mllvm,-import-instr-limit=30 -Wl,-mllvm,-disable-auto-upgrade-debug-info -Wl,-mllvm,-inlinehint-threshold=360 -fwhole-program-vtables -Wl,--undefined-version -Wl,--no-call-graph-profile-sort -m64 -Wl,-O2 -Wl,--gc-sections -Wl,-z,defs -Wl,--as-needed -rdynamic -pie -Wl,--disable-new-dtags -Wl,--lto-O2 prebuilt_rustc_sysroot/lib/rustlib/x86_64-poky-linux-gnu/lib/libstd.rlib prebuilt_rustc_sysroot/lib/rustlib/x86_64-poky-linux-gnu/lib/liballoc.rlib prebuilt_rustc_sysroot/lib/rustlib/x86_64-poky-linux-gnu/lib/libcfg_if.rlib prebuilt_rustc_sysroot/lib/rustlib/x86_64-poky-linux-gnu/lib/libcompiler_builtins.rlib prebuilt_rustc_sysroot/lib/rustlib/x86_64-poky-linux-gnu/lib/libcore.rlib prebuilt_rustc_sysroot/lib/rustlib/x86_64-poky-linux-gnu/lib/libgetopts.rlib prebuilt_rustc_sysroot/lib/rustlib/x86_64-poky-linux-gnu/lib/libhashbrown.rlib prebuilt_rustc_sysroot/lib/rustlib/x86_64-poky-linux-gnu/lib/libpanic_abort.rlib prebuilt_rustc_sysroot/lib/rustlib/x86_64-poky-linux-gnu/lib/libpanic_unwind.rlib prebuilt_rustc_sysroot/lib/rustlib/x86_64-poky-linux-gnu/lib/librustc_demangle.rlib prebuilt_rustc_sysroot/lib/rustlib/x86_64-poky-linux-gnu/lib/libstd_detect.rlib prebuilt_rustc_sysroot/lib/rustlib/x86_64-poky-linux-gnu/lib/libtest.rlib prebuilt_rustc_sysroot/lib/rustlib/x86_64-poky-linux-gnu/lib/libunicode_width.rlib prebuilt_rustc_sysroot/lib/rustlib/x86_64-poky-linux-gnu/lib/libunwind.rlib prebuilt_rustc_sysroot/lib/rustlib/x86_64-poky-linux-gnu/lib/libaddr2line.rlib prebuilt_rustc_sysroot/lib/rustlib/x86_64-poky-linux-gnu/lib/libadler.rlib prebuilt_rustc_sysroot/lib/rustlib/x86_64-poky-linux-gnu/lib/libgimli.rlib prebuilt_rustc_sysroot/lib/rustlib/x86_64-poky-linux-gnu/lib/liblibc.rlib prebuilt_rustc_sysroot/lib/rustlib/x86_64-poky-linux-gnu/lib/libmemchr.rlib prebuilt_rustc_sysroot/lib/rustlib/x86_64-poky-linux-gnu/lib/libminiz_oxide.rlib prebuilt_rustc_sysroot/lib/rustlib/x86_64-poky-linux-gnu/lib/libobject.rlib -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed   -fmacro-prefix-map=/home/mschoepf/yocto/vanilla/build/tmp/work/core2-64-poky-linux/chromium-ozone-wayland/132.0.6834.83/chromium-132.0.6834.83=/usr/src/debug/chromium-ozone-wayland/132.0.6834.83  -fdebug-prefix-map=/home/mschoepf/yocto/vanilla/build/tmp/work/core2-64-poky-linux/chromium-ozone-wayland/132.0.6834.83/chromium-132.0.6834.83=/usr/src/debug/chromium-ozone-wayland/132.0.6834.83  -fmacro-prefix-map=/home/mschoepf/yocto/vanilla/build/tmp/work/core2-64-poky-linux/chromium-ozone-wayland/132.0.6834.83/chromium-132.0.6834.83/out/Release=/usr/src/debug/chromium-ozone-wayland/132.0.6834.83  -fdebug-prefix-map=/home/mschoepf/yocto/vanilla/build/tmp/work/core2-64-poky-linux/chromium-ozone-wayland/132.0.6834.83/chromium-132.0.6834.83/out/Release=/usr/src/debug/chromium-ozone-wayland/132.0.6834.83  -fdebug-prefix-map=/home/mschoepf/yocto/vanilla/build/tmp/work/core2-64-poky-linux/chromium-ozone-wayland/132.0.6834.83/recipe-sysroot=  -fmacro-prefix-map=/home/mschoepf/yocto/vanilla/build/tmp/work/core2-64-poky-linux/chromium-ozone-wayland/132.0.6834.83/recipe-sysroot=  -fdebug-prefix-map=/home/mschoepf/yocto/vanilla/build/tmp/work/core2-64-poky-linux/chromium-ozone-wayland/132.0.6834.83/recipe-sysroot-native=  -fmacro-prefix-map=/home/mschoepf/yocto/vanilla/build/tmp/work/core2-64-poky-linux/chromium-ozone-wayland/132.0.6834.83/recipe-sysroot-native=  -Wl,-z,relro,-z,now  -stdlib=libc++ -o "./chrome" -Wl,--start-group @"./chrome.rsp" -Wl,--end-group   /home/mschoepf/yocto/vanilla/build/tmp/work/core2-64-poky-linux/chromium-ozone-wayland/132.0.6834.83/recipe-sysroot-native/usr/lib/clang/latest/lib/linux/libclang_rt.builtins-x86_64.a -latomic -ldl -lpthread -lrt -lgmodule-2.0 -lgobject-2.0 -lgthread-2.0 -lglib-2.0 -lsmime3 -lnss3 -lsoftokn3 -lnssutil3 -lplds4 -lplc4 -lnspr4 -ldbus-1 -latk-1.0 -latk-bridge-2.0 -lgio-2.0 -lresolv -lexpat -ljpeg -luuid -lxkbcommon -lpci -lffi -lgbm -lpangocairo-1.0 -lpango-1.0 -lharfbuzz -lcairo -ludev -lasound -lFLAC -lm -lz -latspi -lxslt -lxml2 -Wl,--start-group obj/third_party/rust/cxx/v1/lib/libcxx_lib.rlib obj/third_party/rust/serde_json_lenient/v0_2/wrapper/libthird_uparty_srust_sserde_ujson_ulenient_sv0_u2_swrapper_cwrapper.rlib obj/third_party/rust/serde/v1/lib/libserde_lib.rlib obj/third_party/rust/serde_json_lenient/v0_2/lib/libserde_json_lenient_lib.rlib obj/third_party/rust/itoa/v1/lib/libitoa_lib.rlib obj/third_party/rust/ryu/v1/lib/libryu_lib.rlib obj/build/rust/chromium_prelude/libchromium.rlib obj/skia/libskia_cbridge_urust_uside.rlib obj/third_party/rust/font_types/v0_7/lib/libfont_types_lib.rlib obj/third_party/rust/bytemuck/v1/lib/libbytemuck_lib.rlib obj/third_party/rust/read_fonts/v0_23/lib/libread_fonts_lib.rlib obj/third_party/rust/skrifa/v0_24/lib/libskrifa_lib.rlib obj/third_party/blink/common/rust_crash/libthird_uparty_sblink_scommon_srust_ucrash_crs.rlib obj/components/qr_code_generator/libcomponents_sqr_ucode_ugenerator_cqr_ucode_ugenerator_uffi_uglue.rlib obj/third_party/rust/qr_code/v2/lib/libqr_code_lib.rlib -Wl,--end-group 
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
terminate called recursively
terminate called recursively
x86_64-poky-linux-clang++: error: unable to execute command: Aborted (core dumped)
x86_64-poky-linux-clang++: error: linker command failed due to signal (use -v to see invocation)
ninja: build stopped: subcommand failed.
WARNING: /home/mschoepf/yocto/vanilla/build/tmp/work/core2-64-poky-linux/chromium-ozone-wayland/132.0.6834.83/temp/run.do_compile.2560732:149 exit 1 from 'ninja -v -j 32 chrome chrome_sandbox chromedriver.unstripped'
WARNING: Backtrace (BB generated script): 
	#1: do_compile, /home/mschoepf/yocto/vanilla/build/tmp/work/core2-64-poky-linux/chromium-ozone-wayland/132.0.6834.83/temp/run.do_compile.2560732, line 149
	#2: main, /home/mschoepf/yocto/vanilla/build/tmp/work/core2-64-poky-linux/chromium-ozone-wayland/132.0.6834.83/temp/run.do_compile.2560732, line 153
```
I tried this on two different machines, one with 32 GB, the other one has 128GB of memory. Hence, I do not suspect that I ran out of memory here. Any advice or how to debug this is appreciated. 

Regards,
   Matthias

MaxIhlenfeldt, 2025-04-02:
Can you try if https://github.com/OSSystems/meta-browser/issues/845#issuecomment-2664769837 fixes the issue for you as well?

DasRoteSkelett, 2025-04-02:
Hi @MaxIhlenfeldt ,

Thanks, that did the trick for me. Maybe it is worth to document it in the readme? If so, I will be happy to file a PR in that direction. 

Regards,
   Matthias

MaxIhlenfeldt, 2025-04-02:
Glad to hear that! And good idea, please do send a PR that adds a note to the Chromium readme's "Build requirements" section.

closed 2025-04-02 by MaxIhlenfeldt