OSSystems/meta-browser #888: Chromium Scarthgap compile error

Daniel-Wait, 2025-04-18:
Hi, I'm hitting a fatal error in the do_compile step of `bitbake chromium-ozone-wayland'.

``` C++
../../third_party/angle/src/gpu_info_util/SystemInfo_libpci.cpp:12:10: fatal error: 'pci/pci.h' file not found 12 | #include <pci/pci.h>
| ^~~~~~~~~~~
1 error generated.
```

I'm not sure why the pci header isn't found, the pciutils dependency is definitely there.
I suppose that I am either missing something, or the header in one of the following locations ought to be used:

``` bash
~/repos/yocto/build-mine/tmp/work$ find ./ -wholename '*/pci/pci.h'
./cortexa72-poky-linux/chromium-ozone-wayland/132.0.6834.83/chromium-132.0.6834.83/build/linux/debian_bullseye_i386-sysroot/usr/include/i386-linux-gnu/pci/pci.h
./cortexa72-poky-linux/chromium-ozone-wayland/132.0.6834.83/chromium-132.0.6834.83/build/linux/debian_bullseye_amd64-sysroot/usr/include/x86_64-linux-gnu/pci/pci.h
./x86_64-linux/gn-native/132.0.6834.83/chromium-132.0.6834.83/build/linux/debian_bullseye_i386-sysroot/usr/include/i386-linux-gnu/pci/pci.h
./x86_64-linux/gn-native/132.0.6834.83/chromium-132.0.6834.83/build/linux/debian_bullseye_amd64-sysroot/usr/include/x86_64-linux-gnu/pci/pci.h
```

Chromium=v132.0.6834.83
I tried Chromium=v123.0.6312.122 and got the same error (commit=1ed2254d72a4c25879014c98be287a7e3e22904c).
This is my first time trying chromium, so I'd not be surprised if I made a mistake of ignorance.

====== Git Repository Information =======

Repository: meta-browser
Upstream remotes:
  origin: https://github.com/OSSystems/meta-browser/
Current commit: 2dabbb1bc1488f244bf66674fe7268a44dcc0d79
Tags: none
Commit message: scarthgap: chromium: Update to 132.0.6834.83 (#872)
Branch: scarthgap
----------------------------------------

Repository: meta-clang
Upstream remotes:
  origin: https://github.com/kraj/meta-clang
Current commit: eaa08939eaec9f620b14742ff3ac568553683034
Tags: none
Commit message: bcc: fix CVE-2024-2314
Branch: scarthgap
----------------------------------------

Repository: scarthgap-rust-mixin
Upstream remotes:
  origin: git://git.yoctoproject.org/meta-lts-mixins
Current commit: 65b35821c8e39b13b20f0413c315d0c8f1a64a11
Tags: none
Commit message: rust-llvm: Disable libedit
Branch: scarthgap/rust
----------------------------------------

Repository: meta-openembedded
Upstream remotes:
  origin: git://git.openembedded.org/meta-openembedded
Current commit: 67ad83dd7c2485dae0c90eac345007af6195b84d
Tags: none
Commit message: krb5: fix CVE-2025-24528
Branch: scarthgap

OldManYellsAtCloud, 2025-04-19:
> the pciutils dependency is definitely there.

Strangely, even though this should be true based on the main recipe, your `find` results say it otherwise. The pciutils dependency should install its headers in `~/repos/yocto/build-mine/tmp/work/cortexa72-poky-linux/chromium-ozone-wayland/132.0.6834.83/recipe-sysroot/usr/include` folder, which doesn't appear in your output.

Do you have any bbappends or similar modifiers that would change the chromium recipe?
What's the output of `bitbake -e chromium-ozone-wayland | grep ^DEPENDS`? It should list all the effective dependencies.

Daniel-Wait, 2025-04-19:
Thanks so much for the reply!
I tried cleaning and rebuilding both chromium and pciutils.
Strangely, after purging the build target and starting from scratch with a new target, the issue resolved.
I'm entirely unsure what changed or failed previously.

Closing the issue now and can confirm that all worked for the originally defined commits.
Thanks again and apologies for the false issue!

closed 2025-04-19 by Daniel-Wait