OSSystems/meta-browser #908: Build error adding firefox recipe in scarthgap branch

shirasp-bh, 2025-08-01:
I am facing build error while attempting to add firefox recipe in scarthgap branch. Attaching the build logs. Could you please help me to fix the build error.

petalinux20251@petalinux20251-virtual-machine:~/xilinx-zcu102-2025.1$ petalinux-build -c rootfs
[INFO] Building rootfs
[INFO] Bitbake is not available, some functionality may be reduced.
[INFO] Using HW file: /home/petalinux20251/xilinx-zcu102-2025.1/project-spec/hw-description/system-top.dts
[INFO] Getting Platform info from HW file
[INFO] Silentconfig project
[INFO] Silentconfig rootfs
[INFO] Generating configuration files
[INFO] Generating workspace directory
NOTE: Starting bitbake server...
NOTE: Started PRServer with DBfile: /home/petalinux20251/xilinx-zcu102-2025.1/build/cache/prserv.sqlite3, Address: 127.0.0.1:45489, PID: 270468
INFO: Specified workspace already set up, leaving as-is
[INFO] bitbake petalinux-image-minimal -c do_image_complete
NOTE: Started PRServer with DBfile: /home/petalinux20251/xilinx-zcu102-2025.1/build/cache/prserv.sqlite3, Address: 127.0.0.1:35629, PID: 270558
Loading cache: 100% |                                                                                               | ETA:  --:--:--
Loaded 0 entries from dependency cache.
Parsing recipes: 100% |##############################################################################################| Time: 0:15:57
Parsing of 18747 .bb files complete (0 cached, 18747 parsed). 26850 targets, 13658 skipped, 84 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies
NOTE: Resolving any missing task queue dependencies
NOTE: Resolving any missing task queue dependencies
NOTE: Resolving any missing task queue dependencies
NOTE: Resolving any missing task queue dependencies
NOTE: Resolving any missing task queue dependencies
NOTE: Resolving any missing task queue dependencies
NOTE: Resolving any missing task queue dependencies
NOTE: Resolving any missing task queue dependencies
Checking sstate mirror object availability: 100% |###################################################################| Time: 0:00:04
Sstate summary: Wanted 1370 Local 39 Mirrors 1235 Missed 96 Current 2968 (92% match, 97% complete)
Removing 3 stale sstate objects for arch xlnx_zynqmp_zcu102_rev1_0: 100% |###########################################| Time: 0:00:00
NOTE: Executing Tasks
Setscene tasks: 4338 of 4338
WARNING: gtk+-2.24.33-r0 do_fetch: Failed to fetch URL http://ftp.gnome.org/pub/gnome/sources/gtk+/2.24/gtk+-2.24.33.tar.xz, attempting MIRRORS if available
WARNING: libnotify-0.8.3-r0 do_fetch: Failed to fetch URL https://download.gnome.org/sources//libnotify/0.8/libnotify-0.8.3.tar.xz;name=archive, attempting MIRRORS if available
Setscene tasks: 4338 of 4338
WARNING: gnome-themes-extra-3.28-r0 do_fetch: Failed to fetch URL https://download.gnome.org/sources//gnome-themes-extra/3.28/gnome-themes-extra-3.28.tar.xz;name=archive, attempting MIRRORS if available
Setscene tasks: 4338 of 4338
Setscene tasks: 4338 of 4338
Setscene tasks: 4338 of 4338
Setscene tasks: 4338 of 4338
Setscene tasks: 4338 of 4338
ERROR: firefox-68.9.0esr-r0 do_configure: ExecutionError('/home/petalinux20251/xilinx-zcu102-2025.1/build/tmp/work/cortexa72-cortexa53-amd-linux/firefox/68.9.0esr/temp/run.do_configure.670136', 2, None, None)
ERROR: Logfile of failure stored in: /home/petalinux20251/xilinx-zcu102-2025.1/build/tmp/work/cortexa72-cortexa53-amd-linux/firefox/68.9.0esr/temp/log.do_configure.670136
Log data follows:
| DEBUG: Executing python function extend_recipe_sysroot
| NOTE: Direct dependencies are ['/home/petalinux20251/meta-clang/recipes-devtools/clang/clang-cross_git.bb:do_populate_sysroot', '/home/petalinux20251/xilinx-zcu102-2025.1/components/yocto/layers/meta-openembedded/meta-oe/recipes-gnome/gtk+/gtk+_2.24.33.bb:do_populate_sysroot', '/home/petalinux20251/xilinx-zcu102-2025.1/components/yocto/layers/meta-xilinx/meta-xilinx-mali400/recipes-graphics/libgles/libmali-xlnx_r9p0-01rel0.bb:do_populate_sysroot', '/home/petalinux20251/xilinx-zcu102-2025.1/components/yocto/layers/poky/meta/recipes-core/dbus/dbus-glib_0.112.bb:do_populate_sysroot', '/home/petalinux20251/xilinx-zcu102-2025.1/components/yocto/layers/poky/meta/recipes-core/glibc/glibc_2.39.bb:do_populate_sysroot', '/home/petalinux20251/xilinx-zcu102-2025.1/components/yocto/layers/poky/meta/recipes-devtools/gcc/gcc-cross_13.3.bb:do_populate_sysroot', '/home/petalinux20251/xilinx-zcu102-2025.1/components/yocto/layers/poky/meta/recipes-devtools/gcc/gcc-runtime_13.3.bb:do_populate_sysroot', '/home/petalinux20251/xilinx-zcu102-2025.1/components/yocto/layers/poky/meta/recipes-devtools/gcc/libgcc_13.3.bb:do_populate_sysroot', '/home/petalinux20251/xilinx-zcu102-2025.1/components/yocto/layers/poky/meta/recipes-devtools/quilt/quilt-native_0.67.bb:do_populate_sysroot', '/home/petalinux20251/xilinx-zcu102-2025.1/components/yocto/layers/poky/meta/recipes-devtools/rust/libstd-rs_1.75.0.bb:do_populate_sysroot', '/home/petalinux20251/xilinx-zcu102-2025.1/components/yocto/layers/poky/meta/recipes-gnome/gtk+/gtk+3_3.24.41.bb:do_populate_sysroot', '/home/petalinux20251/xilinx-zcu102-2025.1/components/yocto/layers/poky/meta/recipes-gnome/libnotify/libnotify_0.8.3.bb:do_populate_sysroot', '/home/petalinux20251/xilinx-zcu102-2025.1/components/yocto/layers/poky/meta/recipes-graphics/cairo/cairo_1.18.0.bb:do_populate_sysroot', '/home/petalinux20251/xilinx-zcu102-2025.1/components/yocto/layers/poky/meta/recipes-graphics/mesa/mesa-gl_24.0.7.bb:do_populate_sysroot', '/home/petalinux20251/xilinx-zcu102-2025.1/components/yocto/layers/poky/meta/recipes-graphics/startup-notification/startup-notification_0.12.bb:do_populate_sysroot', '/home/petalinux20251/xilinx-zcu102-2025.1/components/yocto/layers/poky/meta/recipes-graphics/xorg-lib/libxi_1.8.1.bb:do_populate_sysroot', '/home/petalinux20251/xilinx-zcu102-2025.1/components/yocto/layers/poky/meta/recipes-graphics/xorg-lib/libxt_1.3.0.bb:do_populate_sysroot', '/home/petalinux20251/xilinx-zcu102-2025.1/components/yocto/layers/poky/meta/recipes-multimedia/alsa/alsa-lib_1.2.11.bb:do_populate_sysroot', '/home/petalinux20251/xilinx-zcu102-2025.1/components/yocto/layers/poky/meta/recipes-multimedia/pulseaudio/pulseaudio_17.0.bb:do_populate_sysroot', '/home/petalinux20251/xilinx-zcu102-2025.1/components/yocto/layers/poky/meta/recipes-support/curl/curl_8.7.1.bb:do_populate_sysroot', '/home/petalinux20251/xilinx-zcu102-2025.1/components/yocto/layers/poky/meta/recipes-support/icu/icu_74-2.bb:do_populate_sysroot', '/home/petalinux20251/xilinx-zcu102-2025.1/components/yocto/layers/poky/meta/recipes-support/libevent/libevent_2.1.12.bb:do_populate_sysroot', 'virtual:native:/home/petalinux20251/meta-browser/meta-firefox/recipes-devtools/cbindgen/cbindgen_0.23.0.bb:do_populate_sysroot', 'virtual:native:/home/petalinux20251/xilinx-zcu102-2025.1/components/yocto/layers/meta-openembedded/meta-oe/recipes-devtools/nodejs/nodejs_20.18.2.bb:do_populate_sysroot', 'virtual:native:/home/petalinux20251/xilinx-zcu102-2025.1/components/yocto/layers/meta-openembedded/meta-oe/recipes-devtools/yasm/yasm_git.bb:do_populate_sysroot', 'virtual:native:/home/petalinux20251/xilinx-zcu102-2025.1/components/yocto/layers/poky/meta/recipes-devtools/gnu-config/gnu-config_git.bb:do_populate_sysroot', 'virtual:native:/home/petalinux20251/xilinx-zcu102-2025.1/components/yocto/layers/poky/meta/recipes-devtools/nasm/nasm_2.16.03.bb:do_populate_sysroot', 'virtual:native:/home/petalinux20251/xilinx-zcu102-2025.1/components/yocto/layers/poky/meta/recipes-devtools/patch/patch_2.7.6.bb:do_populate_sysroot', 'virtual:native:/home/petalinux20251/xilinx-zcu102-2025.1/components/yocto/layers/poky/meta/recipes-devtools/pkgconfig/pkgconfig_git.bb:do_populate_sysroot', 'virtual:native:/home/petalinux20251/xilinx-zcu102-2025.1/components/yocto/layers/poky/meta/recipes-devtools/pseudo/pseudo_git.bb:do_populate_sysroot', 'virtual:native:/home/petalinux20251/xilinx-zcu102-2025.1/components/yocto/layers/poky/meta/recipes-devtools/rust/cargo_1.75.0.bb:do_populate_sysroot', 'virtual:native:/home/petalinux20251/xilinx-zcu102-2025.1/components/yocto/layers/poky/meta/recipes-devtools/rust/rust_1.75.0.bb:do_populate_sysroot', 'virtual:native:/home/petalinux20251/xilinx-zcu102-2025.1/components/yocto/layers/poky/meta/recipes-extended/unzip/unzip_6.0.bb:do_populate_sysroot', 'virtual:native:/home/petalinux20251/xilinx-zcu102-2025.1/components/yocto/layers/poky/meta/recipes-extended/xz/xz_5.4.6.bb:do_populate_sysroot', 'virtual:na [... cut ...]
NOTE: Tasks Summary: Attempted 10048 tasks of which 9855 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/petalinux20251/meta-browser/meta-firefox/recipes-browser/firefox/firefox_68.9.0esr.bb:do_configure
    log: /home/petalinux20251/xilinx-zcu102-2025.1/build/tmp/work/cortexa72-cortexa53-amd-linux/firefox/68.9.0esr/temp/log.do_configure.670136
Summary: There were 4 WARNING messages.
Summary: There was 1 ERROR message, returning a non-zero exit code.
[ERROR] Command bitbake petalinux-image-minimal -c do_image_complete failed

OldManYellsAtCloud, 2025-08-01:
Firefox 68.9.0 is so old, soon it can drive a car...

Try to build the latest available version from the master branch, 140.1.0esr or 141.0.

shirasp-bh, 2025-08-08:
When i try to build the firefox141 recipe, it throws the following error.

WARNING: firefox-141.0-r0 do_fetch: Failed to fetch URL git://github.com/mozilla/application-services.git;protocol=https;branch=main;name=application-services;destsuffix=application-services, attempting MIRRORS if available
Setscene tasks: 4280 of 4280
Setscene tasks: 4280 of 4280
ERROR: firefox-141.0-r0 do_fetch: Fetcher failure: Fetch command export PSEUDO_DISABLED=1; export SSL_CERT_DIR="/home/petalinux20251/petalinux-2025.1/sysroots/x86_64-petalinux-linux/usr/lib/ssl-3/certs"; export SSL_CERT_FILE="/home/petalinux20251/petalinux-2025.1/sysroots/x86_64-petalinux-linux/usr/lib/ssl-3/certs/ca-certificates.crt"; export DBUS_SESSION_BUS_ADDRESS="unix:path=/run/user/1000/bus"; export SSH_AUTH_SOCK="/run/user/1000/keyring/ssh"; export GIT_SSL_CAINFO="/home/petalinux20251/petalinux-2025.1/sysroots/x86_64-petalinux-linux/etc/ssl/certs/ca-certificates.crt"; export PATH="/home/petalinux20251/xilinx-zcu102-2025.1/build/tmp/sysroots-uninative/x86_64-linux/usr/bin:/home/petalinux20251/xilinx-zcu102-2025.1/build/tmp/work/cortexa72-cortexa53-amd-linux/firefox/141.0/recipe-sysroot-native/usr/bin/python3-native:/home/petalinux20251/xilinx-zcu102-2025.1/components/yocto/layers/poky/scripts:/home/petalinux20251/xilinx-zcu102-2025.1/build/tmp/work/cortexa72-cortexa53-amd-linux/firefox/141.0/recipe-sysroot-native/usr/bin/aarch64-amd-linux:/home/petalinux20251/xilinx-zcu102-2025.1/build/tmp/work/cortexa72-cortexa53-amd-linux/firefox/141.0/recipe-sysroot/usr/bin/crossscripts:/home/petalinux20251/xilinx-zcu102-2025.1/build/tmp/work/cortexa72-cortexa53-amd-linux/firefox/141.0/recipe-sysroot-native/usr/sbin:/home/petalinux20251/xilinx-zcu102-2025.1/build/tmp/work/cortexa72-cortexa53-amd-linux/firefox/141.0/recipe-sysroot-native/usr/bin:/home/petalinux20251/xilinx-zcu102-2025.1/build/tmp/work/cortexa72-cortexa53-amd-linux/firefox/141.0/recipe-sysroot-native/sbin:/home/petalinux20251/xilinx-zcu102-2025.1/build/tmp/work/cortexa72-cortexa53-amd-linux/firefox/141.0/recipe-sysroot-native/bin:/home/petalinux20251/xilinx-zcu102-2025.1/components/yocto/layers/poky/bitbake/bin:/home/petalinux20251/xilinx-zcu102-2025.1/build/tmp/hosttools"; export HOME="/home/petalinux20251"; LANG=C git -c gc.autoDetach=false -c core.pager=cat -c safe.bareRepository=all clone --bare --mirror https://github.com/mozilla/application-services.git /home/petalinux20251/xilinx-zcu102-2025.1/build/downloads/git2/github.com.mozilla.application-services.git --progress failed with exit code 128, see logfile for output
ERROR: firefox-141.0-r0 do_fetch: Bitbake Fetcher Error: FetchError('Unable to fetch URL from any source.', 'git://github.com/mozilla/application-services.git;protocol=https;branch=main;name=application-services;destsuffix=application-services')
ERROR: Logfile of failure stored in: /home/petalinux20251/xilinx-zcu102-2025.1/build/tmp/work/cortexa72-cortexa53-amd-linux/firefox/141.0/temp/log.do_fetch.1423949
ERROR: Task (/home/petalinux20251/meta-browser/meta-firefox/recipes-browser/firefox/firefox_141.0.bb:do_fetch) failed with exit code '1'
NOTE: Tasks Summary: Attempted 7422 tasks of which 7420 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/petalinux20251/meta-browser/meta-firefox/recipes-browser/firefox/firefox_141.0.bb:do_fetch
    log: /home/petalinux20251/xilinx-zcu102-2025.1/build/tmp/work/cortexa72-cortexa53-amd-linux/firefox/141.0/temp/log.do_fetch.1423949
Summary: There was 1 WARNING message.
Summary: There were 2 ERROR messages, returning a non-zero exit code.
[ERROR] Command bitbake petalinux-image-minimal -c do_image_complete failed

Looks like "git://github.com/mozilla/application-services.git;protocol=https;branch=main;name=application-services;destsuffix=application-services" in SRC_URI inside firefox.inc is failing to fetch during build.. But when i am able to git clone separately from https://github.com/mozilla/application-services.git.. Any thoughts to get the fetch working during build to get the build going further?

OldManYellsAtCloud, 2025-08-08:
Ahhh, git error 128 - the worst to troubleshoot, as it can mean anything :S

What happens if you delete `/home/petalinux20251/xilinx-zcu102-2025.1/build/downloads/git2/github.com.mozilla.application-services.git` folder and try again? Also, you can clone other git repositories from other recipes, right?

OldManYellsAtCloud, 2025-08-08:
Huhhh, not sure, personally I would follow this stackoverflow link: https://askubuntu.com/questions/26498/how-to-choose-the-default-gcc-and-g-version - I'd guess that you also have gcc12 installed beside libstdc++-12, and when gcc12 is used, I'd expect it would use this folder automatically.

shirasp-bh, 2025-08-08:
I did a build clean after i got the last build error in an intention to do a clean build after the gcc update... So i am not able to check the existence of cstddef now as the build folder is removed now.. I will check this when i do the firefox recipe build next time..

I am trying to build the chromium recipe in the scarthgap branch now... I wanted atleast one browser to work in the scarthgap branch as the petalinux 2025.1 version (from amd) i use is based on scarthgap yocto version.. I initially tried building the firefox recipe in the scarthgap branch but it was too old and was throwing build errors after which you advised me to use the meta browser master branch to build the latest 141 version of firefox.. since thats also not building for me, i thought of checking if the chromium builds in the scarthgap branch.. 

There are a bunch of yocto layers within the petalinux which may be based on scarthgap branch or atleast not based on thier master branch. So if we try to build firefox from its master with its dependent yocto layers not based on their master branch, will it cause build errors as we saw earlier?