OSSystems/meta-browser #909: Scarthgap: Chromium-ozone-wayland: libadler.rlib build failure

Antonios-C, 2025-08-01:
When trying to build chromium 132, latest Chromium version on the scarthgap branch, I get a build failure regarding libadler.rlib . Seems like the library is missing. 

LOGS: 
`[101/55791] ln -f obj/build/rust/std/lib/libadler.rlib prebuilt_rustc_sysroot/lib/rustlib/aarch64-oe-linux-gnu/lib/libadler.rlib 2>/dev/null || (rm -rf prebuilt_rustc_sysroot/lib/rustlib/aarch64-oe-linux-gnu/lib/libadler.rlib && cp -af obj/build/rust/std/lib/libadler.rlib prebuilt_rustc_sysroot/lib/rustlib/aarch64-oe-linux-gnu/lib/libadler.rlib)
| FAILED: prebuilt_rustc_sysroot/lib/rustlib/aarch64-oe-linux-gnu/lib/libadler.rlib
| ln -f obj/build/rust/std/lib/libadler.rlib prebuilt_rustc_sysroot/lib/rustlib/aarch64-oe-linux-gnu/lib/libadler.rlib 2>/dev/null || (rm -rf prebuilt_rustc_sysroot/lib/rustlib/aarch64-oe-linux-gnu/lib/libadler.rlib && cp -af obj/build/rust/std/lib/libadler.rlib prebuilt_rustc_sysroot/lib/rustlib/aarch64-oe-linux-gnu/lib/libadler.rlib)
| cp: cannot stat 'obj/build/rust/std/lib/libadler.rlib': No such file or directory
| [102/55791] ln -f obj/build/rust/std/lib/libgimli.rlib prebuilt_rustc_sysroot/lib/rustlib/aarch64-oe-linux-gnu/lib/libgimli.rlib 2>/dev/null || (rm -rf prebuilt_rustc_sysroot/lib/rustlib/aarch64-oe-linux-gnu/lib/libgimli.rlib && cp -af obj/build/rust/std/lib/libgimli.rlib prebuilt_rustc_sysroot/lib/rustlib/aarch64-oe-linux-gnu/lib/libgimli.rlib)
`

I was able to successfully build Chromium 132 earlier last week, and since then there have been no commits to the scarthgap branch. However, there have been changes in the meta-lts-mixins rust scarthgap branch.

OldManYellsAtCloud, 2025-08-02:
Is it possible that you wan into the same issue that's described in this comment? https://github.com/OSSystems/meta-browser/issues/901#issuecomment-3117375401 - meta-lts-mixins just got a brand new rust (1.87) a few days in the scarthgap branch, and that would fit

fwitk, 2025-08-04:
If somebody wonders, which commit works: commit 736bf97b2beb32c941588391ab2f54f8a96f0307 from the layer meta-lts-mixins worked for me building chromium.

PR #922 (2025-09-16): scarthgap: chromium: Fix build with Rust >= 1.86
Rust 1.86 ships adler2, so the library name has to be adjusted accordingly.

Fixes #909
closed 2025-10-31 by Antonios-C