OSSystems/meta-browser #929: chromium-x11 scarthgap build fails

shirasp-bh, 2025-09-26:
Here is my build configuration
Build Configuration:
BB_VERSION           = "2.8.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "x86_64-poky-linux"
MACHINE              = "intel-corei7-64"
DISTRO               = "poky"
DISTRO_VERSION       = "5.0.4"
TUNE_FEATURES        = "m64 corei7"
TARGET_FPU           = ""
meta                 
meta-poky            
meta-yocto-bsp       = "HEAD:cd44e6bd40b0c1f498b3feaeb5e9b72f8bf32d41"
meta-gnome           
meta-oe              
meta-filesystems     
meta-python          
meta-networking      
meta-xfce            
meta-multimedia      = "HEAD:72018ca1b1a471226917e8246e8bbf9a374ccf97"
meta-virtualization  = "HEAD:6f3c1d8f90947408a6587be222fec575a1ca5195"
meta-intel           = "HEAD:c2bc8e27e9cc83654a23e8d89525fd9b1e781eb6"
meta-perl            = "HEAD:72018ca1b1a471226917e8246e8bbf9a374ccf97"
meta-security        = "HEAD:459d837338ca230254baa2994f870bf6eb9d0139"
meta-clang           = "scarthgap:057ee563305e9484b29d02347aeafdadc5ea28ed"
meta-lts-mixins      = "scarthgap/rust:a0705b87e7c1a257dafb089063f3302ed49c87d3"
meta-chromium        = "scarthgap:1240b0a0b68ccac096e37b5b619c55a5b439c128"
meta-adlink-intel    = "HEAD:d5bf6c40f0c3e7ce02263807fe9339438da520c6"
meta-adlink-sema     = "HEAD:9e511f1546db812ae76a903f140b85b75daedc0e"
meta-tpm             = "HEAD:459d837338ca230254baa2994f870bf6eb9d0139"

The build is successful until i add meta-chromium & its two dependent layers: meta-clang & meta-lts-mixins.
But i have to add these layers to get chromium browser support (chromium-x11) on my intel target.
Attached the build logs.. Any help appreciated..

[buildlog.txt](https://github.com/user-attachments/files/22562745/buildlog.txt)

shirasp-bh, 2025-09-30:
You are right.. Above commit in oe-core resolved my initial build failure.. But as you mentioned, i ran it to the next build issue due to meta-lts-mixins version.. I have downgraded the meta-lts-mixins version to 1.86 now and re-initiated the build. Hope a clean build is not required.. Will get back with the results once the build is completed..

OldManYellsAtCloud, 2025-09-30:
Huhhh... I am on unknown territories now, this post is mostly guessing, what I would try first. Google says that this happens when artifacts from two different rust toolchains are mixed.

You could do something like

```
bitbake -c clean chromium-x11
bitbake -c cleanall rust rust-native librstd librstd-native cargo cargo-native
```

and then try to build again.

shirasp-bh, 2025-10-07:
I am able to get the chromium working now.. Thanks..